esp8266/Arduino #6564: Exception (0) in SoftAP when SPIslave used

saloid, 2019-09-28:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP8285 device]
- Core Version:			
[SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8285 Module]
- Flash Mode:			[other]
- Flash Size: 		[1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description
I am using SPISlave functionality and WiFiManager lib to config wifi connection.  External MCU makes some SPI requests to ESP. All works ok, except that moment, when I am trying to do any SPI request while ESP in STA+AP mode (while it switches to this mode, I believe). I found same issue, but with UART, several years ago: #1073. This "protection from UART interrupts" I found here: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp#L429
Soon I will try to add `ETS_SPI_INTR_DISABLE()` and enable too and write here about results, but this issue is about another possible problem:
**Maybe we need to block all other unsafe interrupts in same critical sections, not only UART?** There also a plenty of another interrupts, which may cause same issues: https://github.com/esp8266/Arduino/blob/master/tools/sdk/include/ets_sys.h#L128

devyte, 2019-10-14:
> while ESP in STA+AP mode (while it switches to this mode, I believe)

What is the code sequence you use for controlling the WiFi? Please provide an MCVE.

saloid, 2019-10-24:
Sorry, I don't have hardware to reproduce this now, but this part should be enough. 

```cpp
#include <SPISlave.h>

void setup()
{
SPISlave.setStatus(statusByte);
SPISlave.begin();
//Now start read status register from esp every 100mSec 
WiFi.begin (ssid, pass); //exception here, when spi interrupt happens in "unsafe" section
}
```

Exceptions happens not only in begin, but in other methods and libs too: in places, which are protected from uart interrupts with `ETS_UART_INTR_DISABLE()`, but not from SPI interrupts. Adding around this parts `ETS_SPI_INTR_DISABLE()` and `ETS_SPI_INTR_ENABLE()` solves this problem.