esp8266/Arduino #6567: Not compiling my OTA code

hazee007, 2019-09-29:
Hello guys i got an OTA code from [https://www.bakke.online/index.php/2017/06/02/self-updating-ota-firmware-for-esp8266/](url)

Here's the code 
```
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include<ESP8266WiFi.h>
#include <WiFiClient.h>
WiFiClient client;
//#include <Arduino.h>

//#include <Arduino.h> /// FOR ESP32 HTTP FOTA UPDATE ///
//#include <WiFi.h> /// FOR ESP32 
//#include <HTTPClient.h> /// FOR ESP32 HTTP FOTA UPDATE ///
//#include <HTTPUpdate.h> /// FOR ESP32 HTTP FOTA UPDATE ///
//#include <WiFiClient.h> /// FOR ESP32 HTTP FOTA UPDATE ///
//WiFiClient client;  //

const int FW_VERSION = 1244;

const char* fwUrlBase = "http://192.168.1.1/fota/";

String getMAC()
{
  uint8_t mac[6];
  char result[14];

 printf( result, sizeof( result ), "%02x%02x%02x%02x%02x%02x", mac[ 0 ], mac[ 1 ], mac[ 2 ], mac[ 3 ], mac[ 4 ], mac[ 5 ] );

  return String( result );
}

void checkForUpdates() {
  String mac = getMAC();
  String fwURL = String( fwUrlBase );
  fwURL.concat( mac );
  String fwVersionURL = fwURL;
  fwVersionURL.concat( ".version" );

  Serial.println( "Checking for firmware updates." );
  Serial.print( "MAC address: " );
  Serial.println( mac );
  Serial.print( "Firmware version URL: " );
  Serial.println( fwVersionURL );

//Dowinload_New_Update
  HTTPClient httpClient;
  httpClient.begin( fwVersionURL );
  int httpCode = httpClient.GET();
  if( httpCode == 200 ) {
    String newFWVersion = httpClient.getString();
  
    Serial.print( "Current firmware version: " );
    Serial.println( FW_VERSION );
    Serial.print( "Available firmware version: " );
    Serial.println( newFWVersion );
  
    int newVersion = newFWVersion.toInt();

    if( newVersion > FW_VERSION ) {
    Serial.println( "Preparing to update." );
  
    String fwImageURL = fwURL;
    fwImageURL.concat( ".bin" );
  
    t_httpUpdate_return ret = ESPhttpUpdate.update( fwImageURL );

//Faliure handler 
  switch(ret) {
        case HTTP_UPDATE_FAILED:
          Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
          break;

        case HTTP_UPDATE_NO_UPDATES:
          Serial.println("HTTP_UPDATE_NO_UPDATES");
          break;
      }
    }
    else {
      Serial.println( "Already on latest version" );
    }
  }
  else {
    Serial.print( "Firmware version check failed, got HTTP response code " );
    Serial.println( httpCode );
  }
  httpClient.end();
}

void setup(){
  Serial.begin(115200);
}

void loop(){
  
}
  
```

when i try compile this code with nodemcu 1.0, i got the following error
```

In file included from C:\Users\Dell\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/WiFiClientSecure.h:41:0,

                 from C:\Users\Dell\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/WiFiServerSecure.h:20,

                 from C:\Users\Dell\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/ESP8266WiFi.h:41,

                 from C:\Users\Dell\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266httpUpdate\src/ESP8266httpUpdate.h:32,

                 from C:\Users\Dell\Documents\Arduino\Multiple_OTA\Multiple_OTA.ino:2:

C:\Users\Dell\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/WiFiClientSecureBearSSL.h:38:5: error: 'BearSSL::WiFiClientSecure::~WiFiClientSecure()' marked override, but does not override

     ~WiFiClientSecure() override;

     ^

C:\Users\Dell\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/WiFiClientSecureBearSSL.h:41:9: error: 'int BearSSL::WiFiClientSecure::connect(const String&, uint16_t)' marked override, but does not override

     int connect(const String& host, uint16_t port) override;

         ^

C:\Users\Dell\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/WiFiClientSecureBearSSL.h:46:12: error: 'size_t BearSSL::WiFiClientSecure::write_P(const char*, size_t)' marked override, but does not override

     size_t write_P(PGM_P buf, size_t size) override;

            ^

C:\Users\Dell\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/WiFiClientSecureBearSSL.h:58:12: error: 'size_t BearSSL::WiFiClientSecure::peekBytes(uint8_t*, size_t)' marked override, but does not override

     size_t peekBytes(uint8_t *buffer, size_t length) override;

            ^

Multiple libraries were found for "WiFiClient.h"
 Used: C:\Program Files (x86)\Arduino\libraries\WiFi
 Not used: C:\Users\Dell\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```
can anyone hlep me understanding the error and how to solve it.
thanks...

Pablo2048, 2019-09-29:
Did you read the compiler output?

> Multiple libraries were found for "WiFiClient.h"
 Used: C:\Program Files (x86)\Arduino\libraries\WiFi
 Not used: C:\Users\Dell\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi

It seems like you have installed another WiFi library... The problem would be probably solved removing the first library. It has nothing to do with ESP8266 Arduino Core.

jandrassy, 2019-09-29:
move `#include<ESP8266WiFi.h>` to the top of the file

devyte, 2019-09-29:
Closing due to user error

closed 2019-09-29 by devyte