esp8266/Arduino #6574: Crashes since #6565 flashstring dedup

TD-er, 2019-10-01:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[1aeea12]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Update to last feature/stage state does reduce bin size, but also causes reboots with exception 28
See #6565 https://github.com/esp8266/Arduino/pull/6565#issuecomment-536913581

earlephilhower, 2019-10-01:
The one change apps will see is that all PSTRs may no longer be aligned to 4-byte boundaries anymore due to coalescing (i.e. `auto a = PSTR("The end"); auto b = PSTR("end");` will make b point to (a+3) ).

So code which assumed a by-4 alignment for PSTRs may have trouble (but by-4 was never _guaranteed_ and is a relatively new addition).

TD-er, 2019-10-01:
I assume it does also apply to F() macro strings, since I don't have that many PSTR defined strings in the code and the significant binary size reduction suggests a lot of deduped strings.

earlephilhower, 2019-10-01:
F() uses PSTR() under the hood.

@mhightower83, I have a niggling sensation that you had a by-4 alignment assumption for PSTRs somewhere in your code.  Alignment should be by-4 unless the PSTR in question is made to point into another string (i.e. another PSTR of longer length ends with your full PSTR).

TD-er, 2019-10-01:
Added a decoded stack trace to the start post.

earlephilhower, 2019-10-01:
So the me21 script is not showing line numbers?  That stinks, there's no reason it can't find lines, the exe used dumps them...

Anyway, looks like you have a NULL pointer problem.  Could be related to the F("on ") or the line.substring variable (in RAM).  Can you give the ELF and MAP for the file that the crash comes from to check the address used for F("on "). 

Also, what would be most useful, can you get a MCVE from this so we can run ourselves and see the crash and peek around?

TD-er, 2019-10-01:
How can I get the address of `F("on ")` from the elf file?

earlephilhower, 2019-10-01:
Just attach the ELF and MAP, I can open them up and see...

TD-er, 2019-10-01:
Since I did make a new build (with NDEBUG removed as build flag) the stack trace may also be changed.
Stack trace is included in the [rar file](https://www.dropbox.com/s/509rtxu6kpv1a5c/custom_ESP8266_4M1M_issue_6574.rar?dl=0)

earlephilhower, 2019-10-01:
@mhightower83, my comment above looks to be in error.  GCC and LD still obey the __align directive even when coalescing strings.  A simple test shows it only merging strings which match on a by-4 address:

````
int i=0;

void call1() {
  const char *str2 = PSTR("ello, world!");
  Serial.println(str2);
  Serial.printf("%p\n", &(str2[0]));
  i++;
}
void call2() {
  const char *str3 = PSTR(", world!");
  Serial.println(str3);
  Serial.printf("%p\n", &(str3[0]));
  i++;
}
void setup() {
  Serial.begin(115200);
}

void loop() {
  const char *str1 = PSTR("Hello, world!");
  call1();
  call2();
  Serial.println(str1);
  Serial.printf("%p\n", &(str1[0]));
  delay(1000);
}
````

Shows
````
ello, world!
0x40239d60
, world!
0x40239d64
Hello, world!
0x40239d50
````
(i.e. str2 of (===str1 starting at 2nd char) was not merged w/str1, but str3, which could merge with str1 or str2 ended up merging w/str3 to ensure alignment.)

TD-er, 2019-10-01:
Just looking at the code of String.

For example: `inline void setLen(int len) { if (isSSO()) sso.len = len; else ptr.len = len; }`
There is no check to see if the capacity of the buffers is sufficient.

What if the new string objects are always a multiple of 4 in size, but the strlen like functions only return upto the first occurrence of a 0?
The value of `len()` in String is being used as index for the buf in `subString`, like here:

```c++
String String::substring(unsigned int left, unsigned int right) const {
    if(left > right) {
        unsigned int temp = right;
        right = left;
        left = temp;
    }
    String out;
    if(left >= len())
        return out;
    if(right > len())
        right = len();
    char temp = buffer()[right];  // save the replaced character
    wbuffer()[right] = '\0';
    out = wbuffer() + left;  // pointer arithmetic
    wbuffer()[right] = temp;  //restore character
    return out;
}
```

And the function `len()` does not check if it is within the capacity of the buffers.

earlephilhower, 2019-10-01:
It's not a String problem, that bit has been stable for months as-is here and in the ESP32.  I wouldn't waste time there.  And strings are not a multiple of 4 in length, they're whatever size is allocated (or SSO'd).

TD-er, 2019-10-01:
But you stated that a PSTR may no longer be bound to address multiples of 4.
When running `unsigned char String::concat(const __FlashStringHelper * str)` , it calls `strlen_P` which was (if my memory serves me well) optimized to only check if the end was on a byte address and not a word address, but it does start with checking per word.

earlephilhower, 2019-10-01:
>It does look like both strings str2 & str3 were merged, right?
> So if you somehow change either string to have an odd offset (e.g. "o, world!") then one of then is not at a multiple of 4 bytes for the first byte.

Not following you there, sorry.

It did merge str 2 and 3, but only because the alignment matched.  If the strings were "ello, world!" and "yellow, world" then the linker would not combine them.  PSTRs don't have any reason other than speed of access to need a 4-byte alignment (they didn't in prior releases), so in fact I might want to consider dropping the alignment for even better savings.  Now it's probably getting mostly duplicated strings as opposed to any substring matches.

Anyway, the crash in your code is in memmove.  TRying to find if it;s source or destination.  Could be an OOM condition in the code and not related to the change, but need more debug to see.

If you can find a small app snippet that I can compile as opposed to the full app (which I can't because PIO and me never worked right for the 8266) that would speed things.

earlephilhower, 2019-10-01:
Try the change in #6577.  It looks like your substring was returning NULL (string empty) and so the this pointer for the equalsIgnoreCase was NULL (I think, it's hard w/o a MCVE to be 100% certain), possibly due to the the issue in #6575 .

dok-net, 2019-10-01:
@earlephilhower In the meantime, could you please just:
 Revert "Speed up empty String creation slightly (#6573)"
This reverts commit 1aeea124b358c3dc63e1c6fb96376695cf24840b.

That's HEAD of master at the time of writing this, and breaks everything (like simplest WebServer demo).

earlephilhower, 2019-10-01:
@dok-net, do you have a sample of code that breaks because of the `String(const char *p=nullptr)` PR?  This is taking something that was tested in optimizing for #6569 and applying generally.

Not saying it's not a problem, but code inspection doesn't seem to show anything glaring, so maybe there's a corner case we're missing.

earlephilhower, 2019-10-01:
Ah, I see it does kill HelloServer.  Let me do some debug...

earlephilhower, 2019-10-01:
@dok-net try the latest push to #6577, another 1-line fix to avoid String getting into an invalid state.

TD-er, 2019-10-01:
I just changed the changes you made in WString.cpp for #6577 and it does now boot fine.
Not sure if we run into other issues, but at least it is now not crashing.

earlephilhower, 2019-10-01:
Thanks for the updates.  The concatenated strings assembling w/o warnings or errors really bothers me.  GAS just ignored the remainder of the line after the first ending quote instead of giving a syntax error ("unexpected junk after .string" or something I'd have hoped for).

As I now have 2 reports of success, I'll merge that PR.  When you have more runtime on the chip, report back, please.

dok-net, 2019-10-01:
@earlephilhower Master works for me now, again.

earlephilhower, 2019-10-01:
Great, thanks for the update @dok-net .

devyte, 2019-10-04:
#6577 is merged, and no new reports are in. If there are further problems, please open a new issue.
Closing.

closed 2019-10-04 by devyte