esp8266/Arduino #6575: PSTR no longer supports concatenated string literals

stevebrophy58, 2019-10-01:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[1/10/19]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Only the first string literal of a concatenated pair is printed in the sketch below.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {
// put your setup code here, to run once:
  Serial.begin(115200);
  
  Serial.printf("\n");
  Serial.printf("test1" ", test2");
  Serial.printf("\n");

  Serial.printf_P(PSTR("test1" ", test2"));
  Serial.printf("\n");
}

void loop() {

}

```

### Debug Messages

```
15:51:27.175 -> test1, test2
15:51:27.175 -> test1

```

earlephilhower, 2019-10-01:
#6577 (a 1-line change you can manually add to your git repo, or do the full pull request) fixes this.  Basically, when PSTR uses assembly, tricks like this string concat (commonly used in debug messages, too) don't work, GAS doesn't understand them.

Your MCVE works fine with this change.

stevebrophy58, 2019-10-01:
Thanks, I can confirm that it fixes the problem reported.

earlephilhower, 2019-10-01:
Just reopening to track that it's no fixed until the final merge of that PR...

PR #6577 (2019-10-01): Replace ASM block w/C macro for PSTR
GAS doesn't support the C language idiom of catenating two strings
together with quotes (i.e. "x" "y" === "xy").

Specify the section attribute fully in the section attribute, instead,
to allow this.

Fixes #6575 and probably #6574
closed 2019-10-01 by earlephilhower