esp8266/Arduino #6601: reconfiguration of ap causes exception

Suraj151, 2019-10-05:
#### Platform

- Hardware:				ESP-12E
- Core Version:			dont know following doc [https://arduino-esp8266.readthedocs.io/en/latest/installing.html#instructions-other-os](url)
- Development Env:		Arduino IDE 1.8.7
- Operating System:		Ubuntu

### Settings in IDE

- Module:  			Nodemcu 1.0
- Flash Mode:			not listed
- Flash Size: 		        4MB
- lwip Variant:       v1.4 compile from source
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description
i am doing some dynamic reconfiguration tests with esp8266 access point. but after two or three attampts it rises exception(28)

```cpp

#include <ESP8266WiFi.h>

#define STASSID "yourssid"
#define STAPSK  "yourpassword"

int cycle = 0;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  configure_station();
  configure_ap();
}

void loop() {
  // put your main code here, to run repeatedly:
  if(cycle++ > 5){
    reconfigure_ap();
    cycle = 0;
  }
  delay(1000);
}

void configure_station(){
  
  WiFi.persistent(false);
  WiFi.setAutoReconnect(false);

  IPAddress local_IP(0,0,0,0);
  IPAddress gateway(0,0,0,0);
  IPAddress subnet(0,0,0,0);

  WiFi.enableSTA(true);
  WiFi.config( local_IP, gateway, subnet );
  WiFi.begin(STASSID, STAPSK);

  Serial.print(F("connecting to "));
  Serial.print(STASSID);
  uint8_t _wait = 1;
  while ( ! WiFi.isConnected() ) {

    delay(1000);
    if( _wait%10 == 0 ){
      Serial.print(F("\ntrying reconnect"));
      WiFi.reconnect();
    }
    if( _wait++ > 18 ){
      break;
[... 360 lines cut ...]
0x401001a8: pvPortMalloc(size_t, char const*, int) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/heap.cpp line 155
0x402015bb: ESP8266WiFiAPClass::softAP(char const*, char const*, int, int, int) at /home/ewings/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp line 169
0x40201513: ESP8266WiFiAPClass::softAP(char const*, char const*, int, int, int) at /home/ewings/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp line 133
0x40203326: uart_write(uart_t*, char const*, size_t) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/uart.cpp line 498
0x40203326: uart_write(uart_t*, char const*, size_t) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/uart.cpp line 498
0x40201d44: HardwareSerial::write(unsigned char const*, unsigned int) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/HardwareSerial.h line 158
0x40201fdd: Print::write(char const*) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.h line 60
0x40202020: Print::println() at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 178
0x4020210d: Print::printNumber(unsigned long, unsigned char) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 260
0x40100100: millis() at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring.cpp line 180
0x40203c38: malloc(size_t) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.cpp line 1677
0x40224336: udp_new at core/udp.c line 953
0x402032dd: uart_write_char(uart_t*, char) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/uart.cpp line 481
0x40203326: uart_write(uart_t*, char const*, size_t) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/uart.cpp line 498
0x40201d44: HardwareSerial::write(unsigned char const*, unsigned int) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/HardwareSerial.h line 158
0x40201fdd: Print::write(char const*) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.h line 60
0x40202020: Print::println() at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 178
0x40202094: Print::println(Printable const&) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 238
0x402012c0: configure_ap() at /tmp/arduino_modified_sketch_152134/sketch_oct05a.ino line 88
0x4020256c: esp_yield() at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 91
0x40202ac7: delay(unsigned long) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring.cpp line 54
0x40201367: loop() at /tmp/arduino_modified_sketch_152134/sketch_oct05a.ino line 22
0x4020261c: loop_wrapper() at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 125

```

TD-er, 2019-10-05:
The decoded stack does not appear to be based on the latest code.
See 
https://github.com/esp8266/Arduino/blob/5d20137339442982172dd6d84488f0c2bb639930/libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp#L169

That's not something doing heap allocation.
Can you see in your files what is located at line 169?

Suraj151, 2019-10-05:
i checked with latest but it seems all same

```
line 167 ==>     // check IP config
line 168 ==>    struct ip_info ip;
line 169 ==>    if(wifi_get_ip_info(SOFTAP_IF, &ip)) {
line 170 ==>        if(ip.ip.addr == 0x00000000) {
line 171 ==>            // Invalid config
line 172 ==>            DEBUG_WIFI("[AP] IP config Invalid resetting...\n");
```
is there anything wrong with umm_malloc.cpp ? i see many times eception(3) where exception decoder shows trace like 
```
0x40215984: _umm_free(void*) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.cpp line 1304
0x40215da8: free(void*) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.cpp line 1764
0x40100498: vPortFree(void*, char const*, int) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/heap.cpp line 61
0x4020be96: ESP8266WiFiAPClass::softAPdisconnect(bool) at /home/ewings/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp line 297
```
it is something like if i try to use ``softAPdisconnect`` or ``enableAP`` like functions in between application, exception loop starts from begin to line where i use those functions

Suraj151, 2019-10-09:
i updated all things with latest master branch and used latest lwip2 precompiled stack, now error has gone.
if i changed lwip Variant to v1.4 compile from source (which include napt custom feature) then got same issue again. so i think its issue with lwip1.4 or customs for that. but after change variant to lwip2 lower memory it works as expected !

devyte, 2019-10-14:
What happens with lwip2 higher bandwidth?
What happens if you add delay(100) after enableAP() and enableSTA()?

Suraj151, 2019-10-14:
### using lwip variant : v1.4 compile from source

if i add delay(100) after enableAP() and enableSTA() i got endless loop of exception(3)

```
connecting to yourssid.........
trying reconnect.........
WiFi Not Connecting. Will try later soon..
Configuring access point yourssid ..
AP IP address: 192.168.13.1
Reconfiguring access point 
Configuring access point yourssid ..
AP IP address: 192.168.9.0
Reconfiguring access point 
Configuring access point yourssid ..
AP IP address: 192.168.15.17
Reconfiguring access point 
Configuring access point yourssid ..

Exception (3):
epc1=0x4010041f epc2=0x00000000 epc3=0x00000000 excvaddr=0x4006eccc depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffb90 end: 3fffffc0 offset: 01a0
3ffffd30:  00000397 00000020 00000397 40100595  
3ffffd40:  3ffee43c 4000b872 0000000c 40224efc  
3ffffd50:  3ffeea30 00000014 0000000c 3ffecdec  
3ffffd60:  00000000 00000001 3fff057c 40100afc  

```

```
Decoding stack results
0x40100595: _umm_free(void*) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.cpp line 1380
0x40224efc: igmp_stop at core/ipv4/igmp.c line 255
0x40100afc: free(void*) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.cpp line 1930
0x40100204: vPortFree(void*, char const*, int) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/heap.cpp line 61
0x40100814: malloc(size_t) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.cpp line 1843
0x402015bf: ESP8266WiFiAPClass::softAP(char const*, char const*, int, int, int) at /home/ewings/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp line 146
0x40201533: ESP8266WiFiAPClass::softAP(char const*, char const*, int, int, int) at /home/ewings/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp line 133
0x40203825: uart_write(uart_t*, char const*, size_t) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/uart.cpp line 498
0x40202434: Print::println(Printable const&) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 238
0x402020b8: HardwareSerial::write(unsigned char const*, unsigned int) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/HardwareSerial.h line 159
0x402020c4: HardwareSerial::write(unsigned char const*, unsigned int) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/HardwareSerial.h line 160
0x402020b8: HardwareSerial::write(unsigned char const*, unsigned int) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/HardwareSerial.h line 159
0x40202379: Print::write(char const*) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.h line 60
0x402023c0: Print::println() at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 178
0x40100200: vPortFree(void*, char const*, int) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/heap.cpp line 57
0x40100814: malloc(size_t) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.cpp line 1843
0x40222b8e: udp_new at core/udp.c line 953
0x40223a20: dhcps_start at app/dhcpserver.c line 732
0x40220108: dhcp_rebind at core/dhcp.c line 1172
0x4020142b: ESP8266WiFiAPClass::softAPConfig(IPAddress, IPAddress, IPAddress) at /home/ewings/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp line 259
0x402029aa: esp_yield() at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 104
0x40202f6e: __delay(unsigned long) at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring.cpp line 57
0x402012b6: configure_ap() at /tmp/arduino_modified_sketch_504570/RangeExtender-NAPT.ino line 86
0x40201af4: ESP8266WiFiGenericClass::enableAP(bool) at /home/ewings/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp line 445
0x40201346: reconfigure_ap() at /tmp/arduino_modified_sketch_504570/RangeExtender-NAPT.ino line 102
0x4020136d: loop() at /tmp/arduino_modified_sketch_504570/RangeExtender-NAPT.ino line 19
0x40202af0: loop_wrapper() at /home/ewings/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 176

```
### using lwip variant : v2 lower memory/ higher bandwidth

this works fine. next i want to go with napt feature introduced with lwip2. dont know what will happen if i use this dynamic reconfiguration with napt enabled network. 
but once it done as expected, then definitely will enable dynamic random sub-netting. will let you know if there is interesting happens.

devyte, 2019-11-06:
From previous comment, I understand that lwip2 works correctly without any code changes or the delay(100) added. 
BTW, lwip1.4 is deprecated, you shouldn't be using it unless you have hit some bug with lwip2 that hasn't been reported or something, and you definitely shouldn't be building it from source unless you know what you're doing, including glue code. The only reason it's still there is as plan B in some unforeseen case when building with lwip2.
Closing because report is for deprecated lib.

closed 2019-11-06 by devyte