esp8266/Arduino #6612: [esptool.py]: BUG! no data upload possible on ESP8266-01

hasenradball, 2019-10-06:
### Basic Infos

- [ x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ x] I have tested that the issue is present in current master branch (aka latest git).
- [ x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ x] I have filled out all fields below.

#### Platform

- Hardware:			ESP8266-01
- Core Version:			2.5.2
- Development Env:		Arduino IDE
- Operating System:		Windows & Ubuntu

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			all tested
- Flash Size: 		        1MB
- lwip Variant:                   
- Reset Method:		
- Flash Frequency:	         
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:               115200

### Problem Description

If you want upload data with the arduino core 2.5.2 on an ESP8266-01 I got an failure. no upload possible with core 2.5.2.
Tested on Windows & ubuntu with the ESP8266 Sketch Data Upload Tool (v0.4).

Going back to core 2.5.0 data upload on the ESP8266-01 it is possible.

Testing both cores with the ESP8266 there is no problem.

### Debug Messages

```
[SPIFFS] data    : D:\06-1_Arduino\ESP8266\ESP8266_WiFi_Relais_Temp_Hum_v2_1\data
[SPIFFS] size    : 256
[SPIFFS] page    : 256
[SPIFFS] block   : 4096
/button.css

/index.html

/script.js

/wificonfig.css

/wificonfig.html

/wificonfig.js

[SPIFFS] upload  : C:\Users\Frank\AppData\Local\Temp\arduino_build_531003/ESP8266_WiFi_Relais_Temp_Hum_v2_1.spiffs.bin
[SPIFFS] address  : 0xBB000
[SPIFFS] reset    : ck
[SPIFFS] port     : COM9
[SPIFFS] speed    : 9600
[SPIFFS] python   : C:\Users\Frank\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1\python.exe
[SPIFFS] uploader : C:\Users\Frank\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\tools\upload.py

esptool.py v2.6

Serial port COM9

Connecting....

Chip is ESP8266EX

Features: WiFi

MAC: 84:f3:eb:94:c6:c4

Uploading stub...

Running stub...

Stub running...

Configuring flash size...

Auto-detected Flash size: 1MB

Compressed 262144 bytes to 5074...

Writing at 0x000bb000... (100 %)
Wrote 262144 bytes (5074 compressed) at 0x000bb000 in 5.4 seconds (effective 390.2 kbit/s)...

Traceback (most recent call last):

  File "C:\Users\Frank\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\tools\upload.py", line 25, in <module>

    esptool.main(fakeargs)

  File "C:/Users/Frank/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 2698, in main

    operation_func(esp, args)

  File "C:/Users/Frank/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 2177, in write_flash

    res = esp.flash_md5sum(address, uncsize)

  File "C:/Users/Frank/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 104, in inner

    return func(*args, **kwargs)

  File "C:/Users/Frank/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 651, in flash_md5sum

    timeout=timeout)

  File "C:/Users/Frank/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 354, in check_command

    val, data = self.command(op, data, chk, timeout=timeout)

  File "C:/Users/Frank/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 332, in command

    p = self.read()

  File "C:/Users/Frank/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 277, in read

    return next(self._slip_reader)

  File "C:/Users/Frank/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 1877, in slip_reader

    raise FatalError("Timed out waiting for packet %s" % waiting_for)

esptool.FatalError: Timed out waiting for packet header

SPIFFS Upload failed!
```

hasenradball, 2019-10-17:
After some testing I recognized that with some chip where the error occurs, that the data upload was ok.
The error seems to raise after the upload wenn checking the files.
But restarting the board after receiving the error, everything seems to be fine.

projectgus, 2019-11-10:
Hi @hasenradball ,

What kind of USB/serial adapter do you use to connect to the ESP8266-01 module?

hasenradball, 2019-11-10:
I have many options to upload the ESP01

jandrassy, 2019-11-10:
did you try higher baud rate?

hasenradball, 2019-11-10:
Yes I tried higher baudrate.
The latest status is that the upload works but maybe the check after the upload fail and raises the error.
But after restart the files are in thr Spiffs available.

This issue occours not at all chips, only at some ones.

devyte, 2019-11-24:
Is this issue still valid with latest git?

hasenradball, 2019-11-24:
You mean with 2.6.1?
I haven't tested yet. I will check it the next days.

hasenradball, 2019-12-02:
Actually I tried this issue with 2.5.2 and it is not reproducable with the same chip.
I would recommend to clode this point.

hasenradball, 2019-12-02:
No issue anymor detected!.

hasenradball, 2019-12-02:
Bug actually Available yet!

Tested with 2.6.2:
```
esptool.py v2.8
Serial port COM8
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 84:f3:eb:94:c6:c4
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 0.0s
Compressed 350592 bytes to 251120...

Writing at 0x00000000... (6 %)
Writing at 0x00004000... (12 %)
Writing at 0x00008000... (18 %)
Writing at 0x0000c000... (25 %)
Writing at 0x00010000... (31 %)
Writing at 0x00014000... (37 %)
Writing at 0x00018000... (43 %)
Writing at 0x0001c000... (50 %)
Writing at 0x00020000... (56 %)
Writing at 0x00024000... (62 %)
Writing at 0x00028000... (68 %)
Writing at 0x0002c000... (75 %)
Writing at 0x00030000... (81 %)
Writing at 0x00034000... (87 %)
Writing at 0x00038000... (93 %)
Writing at 0x0003c000... (100 %)
Wrote 350592 bytes (251120 compressed) at 0x00000000 in 27.9 seconds (effective 100.5 kbit/s)...
Hash of data verified.

Leaving...
Soft resetting...
[SPIFFS] data    : X:\ESP8266-01\ESP8266-01_WiFi_Relais_BME280_v2_4\data
[SPIFFS] size    : 256
[SPIFFS] page    : 256
[SPIFFS] block   : 4096
/button.css

/index.html

/jquery-3.4.1.min.js

/script.js

/wificonfig.css

/wificonfig.html

/wificonfig.js

[SPIFFS] upload  : C:\Users\Frank\AppData\Local\Temp\arduino_build_934613/ESP8266-01_WiFi_Relais_BME280_v2_4.spiffs.bin
[SPIFFS] address  : 0xBB000
[SPIFFS] reset    : --before no_reset --after soft_reset
[SPIFFS] port     : COM8
[SPIFFS] speed    : 115200
[SPIFFS] python   : python.exe
[SPIFFS] uploader : C:\Users\Frank\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\tools\upload.py

usage: esptool [-h] [--chip {auto,esp8266,esp32}] [--port PORT] [--baud BAUD]

               [--before {default_reset,no_reset,no_reset_no_sync}]

               [--after {hard_reset,soft_reset,no_reset}] [--no-stub]

               [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]

               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,version}

               ...

esptool: error: unrecognized arguments: --end

SPIFFS Upload failed!
[SPIFFS] data    : X:\ESP8266-01\ESP8266-01_WiFi_Relais_BME280_v2_4\data
[SPIFFS] size    : 256
[SPIFFS] page    : 256
[SPIFFS] block   : 4096
/button.css

/index.html

/jquery-3.4.1.min.js

/script.js

/wificonfig.css

/wificonfig.html

/wificonfig.js

[SPIFFS] upload  : C:\Users\Frank\AppData\Local\Temp\arduino_build_934613/ESP8266-01_WiFi_Relais_BME280_v2_4.spiffs.bin
[SPIFFS] address  : 0xBB000
[SPIFFS] reset    : --before no_reset --after soft_reset
[SPIFFS] port     : COM8
[SPIFFS] speed    : 115200
[SPIFFS] python   : python.exe
[SPIFFS] uploader : C:\Users\Frank\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\tools\upload.py

usage: esptool [-h] [--chip {auto,esp8266,esp32}] [--port PORT] [--baud BAUD]

               [--before {default_reset,no_reset,no_reset_no_sync}]

               [--after {hard_reset,soft_reset,no_reset}] [--no-stub]

               [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]

               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,version}

               ...

esptool: error: unrecognized arguments: --end

SPIFFS Upload failed!

```

hasenradball, 2019-12-02:
With 2.6.2 no data will be uploaded instead with 2.5.2 data was uploaded but error occurs.

projectgus, 2019-12-02:
@hasenradball 

This error:

> esptool: error: unrecognized arguments: --end

Is to do with the SPIFFS uploader plugin, if you update the plugin it should go away. See also #6864

hasenradball, 2019-12-03:
What is the latest plugin version?

hasenradball, 2019-12-03:
Ok I see I have to go up to the new version! Thank you I will test it.

hasenradball, 2019-12-03:
Ok with Arduino core 2.6.2 and ESP8266FS-0.5.0.zip the following Meassage occurs:
```
Der Sketch verwendet 346432 Bytes (45%) des Programmspeicherplatzes. Das Maximum sind 761840 Bytes.
Globale Variablen verwenden 31024 Bytes (37%) des dynamischen Speichers, 50896 Bytes für lokale Variablen verbleiben. Das Maximum sind 81920 Bytes.
esptool.py v2.8
Serial port COM8
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 84:f3:eb:94:c6:c4
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 0.0s
Compressed 350592 bytes to 251120...

Writing at 0x00000000... (6 %)
Writing at 0x00004000... (12 %)
Writing at 0x00008000... (18 %)
Writing at 0x0000c000... (25 %)
Writing at 0x00010000... (31 %)
Writing at 0x00014000... (37 %)
Writing at 0x00018000... (43 %)
Writing at 0x0001c000... (50 %)
Writing at 0x00020000... (56 %)
Writing at 0x00024000... (62 %)
Writing at 0x00028000... (68 %)
Writing at 0x0002c000... (75 %)
Writing at 0x00030000... (81 %)
Writing at 0x00034000... (87 %)
Writing at 0x00038000... (93 %)
Writing at 0x0003c000... (100 %)
Wrote 350592 bytes (251120 compressed) at 0x00000000 in 28.6 seconds (effective 98.1 kbit/s)...
Hash of data verified.

Leaving...
Soft resetting...
[SPIFFS] data    : X:\ESP8266-01\ESP8266-01_WiFi_Relais_BME280_v2_4\data
[SPIFFS] size    : 256
[SPIFFS] page    : 256
[SPIFFS] block   : 4096
/button.css

/index.html

/jquery-3.4.1.min.js

/script.js

/wificonfig.css

/wificonfig.html

/wificonfig.js

[SPIFFS] upload  : C:\Users\Frank\AppData\Local\Temp\arduino_build_912280/ESP8266-01_WiFi_Relais_BME280_v2_4.spiffs.bin
[SPIFFS] address  : 0xBB000
[SPIFFS] reset    : --before no_reset --after soft_reset
[SPIFFS] port     : COM8
[SPIFFS] speed    : 115200
[SPIFFS] python   : C:\Users\Frank\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1\python3.exe
[SPIFFS] uploader : C:\Users\Frank\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\tools\upload.py

esptool.py v2.8

Serial port COM8

Connecting....

Chip is ESP8266EX

Features: WiFi

Crystal is 26MHz

MAC: 84:f3:eb:94:c6:c4

Uploading stub...

Running stub...

Stub running...

Configuring flash size...

Auto-detected Flash size: 1MB

Compressed 262144 bytes to 45281...

Writing at 0x000bb000... (33 %)
Writing at 0x000bf000... (66 %)
Writing at 0x000c3000... (100 %)
Wrote 262144 bytes (45281 compressed) at 0x000bb000 in 11.3 seconds (effective 185.4 kbit/s)...

Traceback (most recent call last):

  File "C:\Users\Frank\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\tools\upload.py", line 65, in <module>

    esptool.main(cmdline)

  File "C:/Users/Frank/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/tools/esptool\esptool.py", line 2938, in main

    operation_func(esp, args)

  File "C:/Users/Frank/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/tools/esptool\esptool.py", line 2398, in write_flash

    res = esp.flash_md5sum(address, uncsize)

  File "C:/Users/Frank/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/tools/esptool\esptool.py", line 104, in inner

    return func(*args, **kwargs)

  File "C:/Users/Frank/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/tools/esptool\esptool.py", line 691, in flash_md5sum

    timeout=timeout)

  File "C:/Users/Frank/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/tools/esptool\esptool.py", line 369, in check_command

    val, data = self.command(op, data, chk, timeout=timeout)

  File "C:/Users/Frank/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/tools/esptool\esptool.py", line 347, in command

    p = self.read()

  File "C:/Users/Frank/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/tools/esptool\esptool.py", line 292, in read

    return next(self._slip_reader)

  File "C:/Users/Frank/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/tools/esptool\esptool.py", line 2045, in slip_reader

    raise FatalError("Timed out waiting for packet %s" % waiting_for)

esptool.FatalError: Timed out waiting for packet header

SPIFFS Upload failed!
```
Same issue than with 2.5.2 and ESP8266FS-0.4.0.zip

hasenradball, 2019-12-03:
But files seems to be on the Spiffs folder....

spigotx, 2020-02-26:
Is there any update on this issue? I've same problem. I reinstalled the pc and the issue still persists.
Arduino IDE: 1.8.12
ESP8266FS: 0.5.0
Latest working Core Version: 2.4.2
Thanks.

hasenradball, 2020-02-26:
Tested with actual core and arduino 1.8.12.
No problem occured.

closed 2020-02-26 by hasenradball

spigotx, 2020-02-27:
Then could you please help me find out what am I doing wrong? Because I'm able to upload code with v2.4.2, but I get error with anything else. Right now I have v2.6.3 and I get this errors:

```
Executable segment sizes:

IROM   : 392108          - code in flash         (default or ICACHE_FLASH_ATTR) 

IRAM   : 28728   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 

DATA   : 8788  )         - initialized variables (global, static) in RAM/HEAP 

RODATA : 13732 ) / 81920 - constants             (global, static) in RAM/HEAP 

BSS    : 31344 )         - zeroed variables      (global, static) in RAM/HEAP 

Sketch uses 443356 bytes (42%) of program storage space. Maximum is 1044464 bytes.
Global variables use 53864 bytes (65%) of dynamic memory, leaving 28056 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.8
Serial port COM6
Connecting........
Traceback (most recent call last):
  File "C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3/tools/upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "C:/Users/user/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 2890, in main
    esp.connect(args.before)
  File "C:/Users/user/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 478, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=True)
  File "C:/Users/user/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 455, in _connect_attempt
    self.sync()
  File "C:/Users/user/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 394, in sync
    timeout=SYNC_TIMEOUT)
  File "C:/Users/user/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 347, in command
    p = self.read()
  File "C:/Users/user/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 292, in read
    return next(self._slip_reader)
  File "C:/Users/user/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 2041, in slip_reader
    read_bytes = port.read(1 if waiting == 0 else waiting)
  File "C:/Users/user/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/pyserial\serial\serialwin32.py", line 293, in read
    raise SerialException("GetOverlappedResult failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: GetOverlappedResult failed (OSError(22, 'Reached the end of the file.', None, 38))
serial.serialutil.SerialException: GetOverlappedResult failed (OSError(22, 'Reached the end of the file.', None, 38))
```

Thank you.

hasenradball, 2020-02-27:
Can you please make a screenshot from your settings which you use for upload?

projectgus, 2020-02-27:
Hi @spigotx ,

What kind of dev board or USB/serial adapter do you have? This `serial.serialutil.SerialException: GetOverlappedResult failed (OSError(22, 'Reached the end of the file.', None, 38))` error basically means that esptool is asking the OS to read some data and the OS (Windows) is returning a low level error saying "Nope, I can't read from this serial port!".

Usually this means there is some low level problem like a Windows driver issue, a conflict with some other software running in the background (or a virus scanner, etc), or some hardware problem like the USB cable is faulty or something like that. Although that doesn't quite explain why it works in the older Arduino.

spigotx, 2020-02-28:
> 
> 
> Can you please make a screenshot from your settings which you use for upload?

[image]

spigotx, 2020-02-28:
> 
> 
> Hi @spigotx ,
> 
> What kind of dev board or USB/serial adapter do you have? This `serial.serialutil.SerialException: GetOverlappedResult failed (OSError(22, 'Reached the end of the file.', None, 38))` error basically means that esptool is asking the OS to read some data from the serial port (COM port) and the OS (Windows) is returning a low level error saying "Nope, I can't read from this serial port!".
> 
> Usually this means there is some low level problem like a Windows driver issue, a conflict with some other software running in the background (or a virus scanner, etc), or some hardware problem like the USB cable is faulty or something like that. Although that doesn't quite explain why it works in the older Arduino.
> 
> EDIT: silly question but also please double check in Windows Device Manager that this is the correct COM port, and double check for any "X" or "?" or similar on the port in Device Manager to indicate Windows is detecting some problem.

Devboard: Wemos D1 mini / NodeMCU Dev board

I always double check if there isn't any HW problem. And if there would, I wouldn't be able to upload sketch with v2.4.2.
I don't use any AVIR SW on this machine. I have fresh install of Win 10. Only Arduino IDE and MS VS.

[image]

Thanks for help.

hasenradball, 2020-02-28:
Which chip you want to flash???
ESP8266-01?

hasenradball, 2020-02-28:
With the ESP8266 there was no problem!
With the ESP8266-01 use speed 115200 baud!

hasenradball, 2020-02-28:
**Please try it with this settings on the ESP8266-01!**

[image]

spigotx, 2020-02-29:
Thanks for helping me and for pointing me in the right direction. 
Problem is in serial communication as projectgus suggested. 
For Arduino IDE I'm using virtual machine (hosted on ESX server). Connection between VM and Arduino is made over remote desktop.
The serial communication over RD in v2.4.2 is working OK, but since v2.5 it stopped working. When I connect dev board directly to the VM, the upload is working even in v2.6.3. 
So my only hope is that someone will make it work again.

Thanks.

unmateria, 2020-10-04:
try enabling --trace :)