esp8266/Arduino #6615: memory slowly running out when a lot of multicasting in network

mariusstaicu, 2019-10-07:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:			        esp-12 (nodemcu v2 or wemos d1 mini  - no difference)
- Core Version:			2.5.2
- Development Env:		Platformio 4.0.3
- Operating System:	        Ubuntu 19.04

### Settings in IDE

- Module:  		board esp12e
- Flash Mode:		dio
- Flash Size: 		4MB
- lwip Variant:           v2 Lower Memory
- Reset Method:	nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:	80Mhz
- Upload Using:	OTA and Serial, the same
- Upload Speed:      115200

### Problem Description

Noticed devices in our network started to reset once every 6-24h because they run out of memory. Slowly but steady, the freed heap starts to drop and not recover anymore.
After a long investigation about the cause I concluded the following:

1. devices were being affected by this memory leak only on our main wireless (where there are ~20-30 devices using mdns multicast - including other esp8266 devices using ArduinoOTA,  MDNS broadcast and SSDP discovery . Once I moved the esp on a dedicated wifi, the symptoms were gone.
2. started making a minimal MVCE, removed all down to minimum and traced the problem coming from using ArduinoOTA usage, which I think it goes down to mdns. Whenever I stopped using ArduinoOTA in my code, the memory would stay steady for long term.
3. lwip 1.4 - problem does not show, memory is steady over long term; lwip2 low memory - problem appears
4. tried to use tcpCleanup from FAQ, it seems only to slow down the heap decrease, but not solve the problem. From 6-10h max uptime on a device, it went to 2+ days until the restart.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <ArduinoOTA.h>

struct tcp_pcb;
extern struct tcp_pcb* tcp_tw_pcbs;
extern "C" void tcp_abort (struct tcp_pcb* pcb);

#include <ESP8266WiFi.h>
#include <PolledTimeout.h>

int initialHeap;

void setup() {
    Serial.begin(115200);

    if (WiFi.getMode() == WIFI_STA) {
        WiFi.begin("ssid", "password");
    }

    printf("connecting to WiFi.. \n");
    int retry = 0;
    while (WiFi.status() != WL_CONNECTED && retry < 20) {
        printf(".");
        delay(500);
        retry++;
    }
    printf("\n");

[... 2318 lines cut ...]
free heap: 35472, difference from startup: -13760
free heap: 37824, difference from startup: -11408
free heap: 38384, difference from startup: -10848
free heap: 39688, difference from startup: -9544
free heap: 40280, difference from startup: -8952
free heap: 39800, difference from startup: -9432
free heap: 39776, difference from startup: -9456
free heap: 39592, difference from startup: -9640
free heap: 42312, difference from startup: -6920
free heap: 42312, difference from startup: -6920
free heap: 41440, difference from startup: -7792
free heap: 41544, difference from startup: -7688
free heap: 42616, difference from startup: -6616
free heap: 42960, difference from startup: -6272
free heap: 43000, difference from startup: -6232
free heap: 41240, difference from startup: -7992
free heap: 38984, difference from startup: -10248
free heap: 40344, difference from startup: -8888
free heap: 41984, difference from startup: -7248
free heap: 42240, difference from startup: -6992
free heap: 42888, difference from startup: -6344
free heap: 42432, difference from startup: -6800
```

I have also a ~45 MB logfile in which I enabled also debug logs from MDNSResponder if anyone is interested in that.

d-a-v, 2019-10-08:
Thanks for the MCVE, I will try to reproduce

vborcea, 2020-10-23:
@d-a-v hello,
are there any updates on this issue?
Do you need more info?

TD-er, 2020-10-23:
For a lot of protocols (e.g. UDP, OTA, running webserver, DNS, etc.) we have a function that should be called on regular intervals to serve the received data.
Is there something similar for these packages, without the need to run their service like mDNS?

If not, then I guess it would be a nice idea to have some bitwise flag signalling what protocols should be dealt with by a service (to be set when you start a service and reset when you stop it).
All packets not marked for processing should then be dropped.

The other way to fix it is to keep track of timestamps, timeouts and all sorts of things, which may give quite some extra administration overhead.
But maybe that last one is a preferred one?

hreintke, 2020-10-23:
@vborcea @TD-er ( @d-a-v )
There has been an update to the mDNS implementation lately. 
Would be good to repeat your tests with the current git master and verify the behavior.

d-a-v, 2020-10-23:
Latest git or [installable beta release v0.0.1](https://d-a-v.github.io/esp8266/Arduino/index.html) include these changes.
There were also changes in UDP to limit the number of buffered packets, so it is worth re-testing like @hreintke suggested.

d-a-v, 2020-11-09:
Does memory keep more steady now ?

devyte, 2020-11-25:
Closing due to lack of feedback.

closed 2020-11-25 by devyte

mariusstaicu, 2021-01-08:
Sorry, I was just able to find time to test it now. It seems there is little to no memory drop over time with this alpha implementation. 
After ~16h the memory is well over 48k, and seems stable.

d-a-v, 2021-01-08:
Thanks for the feedback !