esp8266/Arduino #6618: HTTPS and MQTTS at same time with different endpoints on ESP8266

WBorsato, 2019-10-07:
Hello,

I'm trying to establish a connection to a MQTT broked using TLS v1.2 and use the firebase cloud message https API at same time. The two endpoints are different.

When I connect to the second "server", during the handshake when device and server trade certificates, I get the issue:

=== CERTIFICATE ISSUED TO ===
Common Name (CN):      edgecert.googleapis.com
Organization (O):      Google LLC
Location (L):         Mountain View
Country (C):         US
State (ST):         California
Basic Constraints:      critical, CA:FALSE, pathlen:10000
Key Usage:         critical, Digital Signature, Key Encipherment
Subject Alt Name:      edgecert.googleapis.com android.googleapis.com bigtable.googleapis.com clouderrorreporting.googleapis.com cloudtrace.googleapis.com container.googleapis.com dataflow.googleapis.com datastore.googleapis.com fcm.googleapis.com firestore.googleapis.com logging.googleapis.com monitoring.googleapis.com pubsub.googleapis.com runtimeconfig.googleapis.com servicecontrol.googleapis.com spanner.googleapis.com stackdriver.googleapis.com
=== CERTIFICATE ISSUED BY ===
Common Name (CN):      GTS CA 1O1
Organization (O):      Google Trust Services
Country (C):         US
Not Before:         Tue Sep 17 13:30:13 2019
Not After:         Tue Dec 10 13:30:13 2019
RSA bitsize:         2048
Sig Type:         SHA256
=== CERTIFICATE ISSUED TO ===
Common Name (CN):      GTS CA 1O1
Organization (O):      Google Trust Services
Country (C):         US
Basic Constraints:      critical, CA:TRUE, pathlen:0
Key Usage:         critical, Digital Signature, Key Cert Sign, CRL Sign
=== CERTIFICATE ISSUED BY ===
Common Name (CN):      GlobalSign
Organization (O):      GlobalSign
Organizational Unit (OU):   GlobalSign Root CA - R2
Not Before:         Thu Jun 15 00:00:42 2017
Not After:         Wed Dec 15 00:00:42 2021
RSA bitsize:         2048
Sig Type:         SHA256
State:   receiving Server Hello Done (14)
Fatal exception 29(StoreProhibitedCause):
epc1=0x40213cf2, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

Exception (29):
epc1=0x40213cf2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont
sp: 3ffff9a0 end: 3fffffd0 offset: 01a0

Is it possible to handle 2 certificates at same time?

devyte, 2019-10-07:
Release 2.4.2 is over a year old, and there have been 589 commits since then, including a change in the tls library, which was prone to odd crashes when faced with oom. I suggest trying with a newer version, or better yet, latest git.

As a side note, a secure server requires a 16KB buffer, plus other mem totaling about 22KB. Running 2 secure servers likely leaves you oom, unless your app is aggressively optimized for heap.
It is possible to reduce the 16KB requirement if you control the client side of the connection.

Closing for now. If you encounter the problem with latest git, please open a new issue and fill out the required info, including MCVE.

closed 2019-10-07 by devyte

WBorsato, 2019-10-07:
> Release 2.4.2 is over a year old, and there have been 589 commits since then, including a change in the tls library, which was prone to odd crashes when faced with oom. I suggest trying with a newer [...]

I have tested right now on the 2.5.2 and I'm also having the same issue.

d-a-v, 2019-10-07:
You can also enable debug options, and try/read the `BearSSL_MaxFragmentLength.ino` example.