esp8266/Arduino #6622: [HELP] Where can I find the definition of the WiFI autoReconnect interval?

nextguyover, 2019-10-08:
When ESP is connected to a WiFi network, and it disconnects, it automatically attempts to reconnect to the network once every second. This severely affects loading time for a SoftAP web server, if active at that time. 

My intention was to just disable this feature using "WiFi.setAutoReconnect(false);". This, unfortunately, has a bug where, when disabled, the WiFi.status() does not change from WL_CONNECTED, and WiFi.isConnected() does not change to false when a WiFi network disconnects.

Next up I tried to just increase the delay from 1 second to 10 seconds, which is where I'm at now.

I tried searching for parts of the debug message printed ("no WIFI_NETWORK_SSID found, reconnect after 1s"): "found, reconnect" and "reconnect after", in all the files in the library folder (C:\Users\current_user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries) using the "Find in Files" function of Notepad++. I did not get any results from this.

My question is, where is the 1-second auto-reconnect interval defined? And, is there already some sort of function already in place for changing this interval value?

TD-er, 2019-10-08:
> My intention was to just disable this feature using "WiFi.setAutoReconnect(false);". This, unfortunately, has a bug where, when disabled, the WiFi.status() does not change from WL_CONNECTED, and WiFi.isConnected() does not change to false when a WiFi network disconnects.

I think this remark is worth an issue on its own and is a big eye opener for me.

nextguyover, 2019-10-08:
> I think this remark is worth an issue on its own and is a big eye opener for me.

After spending a few hours searching for the 1 second delay definition, and failing, I attempted to fix this issue of not recognising the disconnect. When WIFI debugging is enabled, I saw that "STA disconnect: 3" gets printed to the Serial Monitor. I used Notepad++ to search for this, and found the function "void ESP8266WiFiGenericClass::_eventCallback(void* arg)" in ESP8266WiFiGeneric.cpp.

In the same if statement that calls the Debug output, there was this line "WiFiClient::stopAll();". I performed a search for "stopAll(", and found function void WiFiClient::stopAll() in WiFiClient.cpp. I then just added "wifi_station_disconnect();" to this function. "wifi_station_disconnect();" I found looking at the WiFi.disconnect() function.

It recognises the disconnect now, but I'm not sure whether this is a proper solution. I'm running tests at the moment, I'll see how it goes. I'm really new to this, so if I'm doing something wrong, or if there's something I can do better, please let me know.

TD-er, 2019-10-08:
> Did anyone try the wifi events ?
> 
> There's `(WiFiEvent)WIFI_EVENT_STAMODE_DISCONNECTED`
> and `(WiFiDisconnectReason)WIFI_DISCONNECT_REASON_...`
> 

I'm using it: https://github.com/letscontrolit/ESPEasy/blob/ca86fd2f19fb203bae3d1c1b2ab134d92ecbbd32/src/ESPEasyWiFiEvent.h#L44-L60

But I've also seen sometimes that the wifistatus is not reflecting the true status.
That's why my WiFi code is so much more complex and elaborate than I would like it to be.
Just to keep track of WiFi inconsistencies.

Misiu, 2019-11-06:
> But I've also seen sometimes that the wifistatus is not reflecting the true status.
> That's why my WiFi code is so much more complex and elaborate than I would like it to be.
> Just to keep track of WiFi inconsistencies.

This should be addressed as a separate issue. Currently, we must use a complex way to track real status.
With a proper fix, that extra complex code can be removed, so we get less wifi related problems and smaller build size.