esp8266/Arduino #6623: HTTPClient setInsecure() does not return expected stream

ChrisBeeson, 2019-10-08:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below (that I know)

#### Platform

- Hardware:					[ESP8266EX]
- Core Version:			[ 2.2.3 ]
- Development Env:		[Platformio]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Wemos D1 mini r2]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

When I use BearSSL .setInsecure() and getStream() the bytes downloaded are not as expected.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

  HTTPClient https;
  BearSSL::WiFiClientSecure secureClient;
  secureClient.setInsecure();
  https.begin(secureClient, STORAGE_BASE_URL, 443, url, true);
  WiFiClient stream = https.getStream();

// Dump Hex
char readBuf[32] = {0};
stream.readBytes(readBuf, sizeof(readBuf));
hexDump("read buf", readBuf, sizeof(readBuf));

```
I used hexDump from [here](https://stackoverflow.com/questions/7775991/how-to-get-hexdump-of-a-structure-data/7776146#7776146)

### Debug Messages

The first 32 bytes downloaded from a firebase bucket are:

`0000 17 03 03 03 e7 89 f2 6a cc dd 73 1c 86 46 6b 4c`

But if I download the same file in my browser the bytes are:

`0000 e9 01 02 40 9c f2 10 40 00 f0 10 40 68 05 00 00`

File sizes are identical.

ChrisBeeson, 2019-10-08:
However!  If I read directly from `https.getStream()` the bytes are correct.

earlephilhower, 2019-10-08:
WiFiClient is an unencrypted socket, and you're reading an encrypted stream (`setInsecure` just means it doesn't authenticate the cert, not that it doesn't encrypt).

Try `WifiClientSecure stream = https.getStream()`

ChrisBeeson, 2019-10-08:
`no suitable user-defined conversion from "WiFiClient" to "BearSSL::WiFiClientSecure" exists`

earlephilhower, 2019-10-08:
Well, then you can't do what you're trying to do.  You could put in a PR.  There probably should be an overload to return a secure client, returning an error when connected HTTP, and the original one should return an error when using a HTTPS connection.

Even simpler, since you're passing in the WiFiClientSecure, why not just use the WiFiClientSecure you already have?  That's all the `getStream()` seems to be doing on first inspection.

devyte, 2019-10-09:
@ChrisBeeson I think this
`WiFiClient stream = https.getStream();`
should be
`WiFiClient & stream = https.getStream();`