esp8266/Arduino #6626: WiFiClient cannot connect to IPv6 link local address

Jamesits, 2019-10-10:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP8266MOD
- Core Version:			2.5.2
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Wemos D1 R2 & mini
- Flash Size: 		4MB
- lwip Variant:   v2 IPv6 Higher Bandwidth
- Reset Method:		ck
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   921600

### Problem Description

As per https://github.com/esp8266/Arduino/pull/5136, IPv6 is supported on the recent versions of ESP8266. However, I observed that `WiFiClient.connect()` is not able to connect to an IPv6 link-local IP address (`fe80::/10`) on the same network with the ESP8266.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch
```C++
#include <ESP8266WiFi.h>

#define STA_SSID "your-ssid"
#define STA_PSWD "your-wifi-password"
#define SERVER_IP "fe80::aaaa:bbff:fecc:dddd"

IPAddress server;
WiFiClient client;

void setup() {
  Serial.begin(115200);

  WiFi.mode(WIFI_STA);
  WiFi.setAutoConnect(true);
  WiFi.setAutoReconnect(true);
  WiFi.begin(STA_SSID, STA_PSWD);
}

void loop() {
    Serial.println("\nStarting connection...");
    // if you get a connection, report back via serial:
    server.fromString(SERVER_IP);
    if (client.connect(server, 8000)) {
      Serial.println("connected");
      // Make a HTTP request:
      client.println("GET / HTTP/1.0");
      client.println();
    } else {
      Serial.println("Failed to connect");
    }
    delay(500);
}
```

### Debug Messages
Sketch output:

```
Starting connection...
Failed to connect

```

I've done a packet capture on the server and can see NO packets from ESP8266.

theGeekyLad, 2022-01-02:
Is there any update on this?

d-a-v, 2022-01-04:
I am still not an IPv6 specialist.

My linux laptop cannot reach my server through `fe80:` and I need to add the interface name,
like:  `$ ping fe80:...:852c%wlp2s0`.
However `$ ping fd6f:...:852c` works.

Can you give it a try ?

Jamesits, 2022-01-05:
@d-a-v Normally you'd need to bind to an interface when using `fe80` link-local IPs since they are bound to an interface. There is no way to do this on the ESP8266 SDK...

d-a-v, 2022-01-05:
lwIP has it and we use it:
https://github.com/esp8266/Arduino/blob/a05a71fa9d2e6b143cb34f01b47e22c4b66b80a1/libraries/ESP8266WiFi/src/include/UdpContext.h#L99-L104
https://github.com/esp8266/Arduino/blob/c312a2eaf1356ceaafad7c4935fa850e087c84fe/libraries/ESP8266WiFi/src/include/ClientContext.h#L134-L139

`netif_default` should be [set to the STA interface](https://github.com/d-a-v/esp82xx-nonos-linklayer/blob/450bb63c1bc8b35770ca7f246945cf383569f52f/glue-lwip/lwip-git.c#L325-L328) (<= this is next version and this is [current version](https://github.com/d-a-v/esp82xx-nonos-linklayer/blob/2c9b64c9ea08115a2410a5ecdb1f7f8c96229523/glue-lwip/lwip-git.c#L321-L325)).

d-a-v, 2022-01-05:
I was asking to try because `fd6f:` is not public and is a reserved address dedicated to local area, as `fe80:` is reserved for the link only. I don't remember and I have no ipv6 environment right now to check if the ipv6 implementation of lwIP binds to this domain too.
Please correct me if I'm mistaken.