esp8266/Arduino #6633: Strange DHCP behaviour

olepalm2630, 2019-10-11:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ x] I have filled out all fields below.

#### Platform

- Hardware:			Adafruit Feather Huzzah 8266
- Core Version:			2.5.2
- Development Env:		[Arduino IDE and |Platformio
- Operating System:		[Windows

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			dio
- Flash Size: 		        4MB
- lwip Variant:                v2 Lower Memory
- Reset Method:		ck
- Flash Frequency:	        [?]
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:              115200

**Problem Description:**
Insisting on DHCP address if configured DNS is valid

**Detailed problem description**
When requesting a static IP a DHCP request is issued if (and only if) a valid DNS address is configured. 
If DNS is set to 0.0.0.0 or gateway IP, a static IP is obtained.

**Code**
#include <Arduino.h>
#include <ESP8266WiFi.h>
#define STASSID "SamsScam"
#define STAPSK  "********"
const char* ssid     = STASSID;
const char* password = STAPSK;

//Static IP address configuration//
IPAddress staticIP(192, 168, 0, 145); //ESP static ip
IPAddress gateway(192, 168, 0, 1);   //IP Address of your WiFi Router (Gateway)
IPAddress subnet(255, 255, 255, 0);  //Subnet mask
IPAddress dns(194, 239, 134, 83);  //DNS

//WiFiClient espClient;

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.mode(WIFI_STA);
  WiFi.setSleepMode(WIFI_NONE_SLEEP);
  WiFi.disconnect();  //Prevent connecting to wifi based on previous configuration
  WiFi.hostname("logger");      // DHCP Hostname (useful for finding device for static lease)
  WiFi.config(staticIP, subnet, gateway, dns);
  delay(100);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
 }

void loop() {
  //while(1);// put your main code here, to run repeatedly:
}

jandrassy, 2019-10-11:
bool config(IPAddress local_ip, IPAddress gateway, IPAddress subnet, IPAddress dns1 = (uint32_t)0x00000000, IPAddress dns2 = (uint32_t)0x00000000);

olepalm2630, 2019-10-11:
OOPS  how embarrasing….
It is actually directly out of an example.
I'll try this out ASAP.
Thank you very much.

olepalm2630, 2019-10-11:
Now it's fine.
Thanks again.

closed 2019-10-11 by olepalm2630