esp8266/Arduino #6637: String::replace (growing) drops null terminator in sso mode

new8266user, 2019-10-13:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[2.5.2]
- Development Env:		[Emacs]
- Operating System:		[Debian Gnu/Linux]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[searched quite a while, couldn't find it]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

in void String::replace(const String& find, const String& replace)
WString.cpp:761  memmove drops null terminator in sso mode

    // // maybe CAN'T copy '\0' terminator unless in sso?
    // // but need to in sso mode.
    // // This does NOT copy terminator.
    // memmove(readFrom + diff, readFrom, len() - (readFrom - buffer()));
    
    // This fixes it for my issue:  (probably needs more testing)
    memmove(readFrom + diff, readFrom, len() - (readFrom - buffer())+1);

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

    void setup() {
      Serial.begin(115200);
      String victim = "0123456bad!";
      Serial.println(victim);
      victim = "i ? ";
      Serial.println(victim);
      victim.replace("i", "it's");
      Serial.println(victim);
    }

    void loop() {
    }

### Debug Messages

    0123456bad!
    i ? 
    it's ? bad!

devyte, 2019-10-13:
> I have tested that the issue is present in current master branch (aka latest git).

Is that true? You tested latest git and the issue is still present?

new8266user, 2019-10-13:
yes.  Thanks for questioning me.

My patch did not hold up to further testing...
looks like moving the null adds diff into len(), so it gets added twice.

A better fix might be to move the line:
`    int newLen = len() + diff;
`
up one line, as we need to avoid re-scanning the buffer once the '\0' has been clobbered.

This kinda makes sense, as len was a variable (not a function that could call strlen()) in the original (non-sso) avr version.

devyte, 2019-10-13:
There are host side tests that cover the code in String. Please review them, they run on Linux. From what you have said so far, my suspicion is that the case you're describing isn't covered. If that is true, please:
1. run the suite to get a green light baseline run
2. add a test that covers your case
3. run the suite to make sure your issue is reproduced (run should fail) 
4. implement your fix in String 
5. rerun the suite to make sure all tests pass
6. make a PR with both your fix and the new test

mbm60, 2019-10-13:
I tested.

**With 2.5.2 :**
```
0123456bad!
i ? 
it's ? bad!
```

**With latest git :**
```
0123456bad!
i ? 
it's ? 
```

d-a-v, 2019-10-13:
Can't reproduce, agreed with @mbm60,
This MCVE is working with latest git on host emulation.

earlephilhower, 2019-10-13:
There is no memmove on line 761 in the sources, and the string (wherever it is) is 0-terminated at line 768.

https://github.com/esp8266/Arduino/blob/122e87019f35ecb928c00631b6dbd4d581e12105/cores/esp8266/WString.cpp#L761-L768

Code inspection and multiple counter confirmations.  Closing as NTF.

closed 2019-10-13 by earlephilhower