esp8266/Arduino #6640: Error compiling library ESP8266WiFi.h

mayor89, 2019-10-14:
### Basic Infos

- [X ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X ] I have tested that the issue is present in current master branch (aka latest git).
- [X ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[DOUT]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [9600] (serial upload only)

### Problem Description
In file included from C:\Users\Mayor\Documents\Arduino\libraries\ESP8266WiFi\src/WiFiClientSecureBearSSL.h:29:0,
                 from C:\Users\Mayor\Documents\Arduino\libraries\ESP8266WiFi\src/WiFiClientSecure.h:41,
                 from C:\Users\Mayor\Documents\Arduino\libraries\ESP8266WiFi\src/WiFiServerSecure.h:20,
                 from C:\Users\Mayor\Documents\Arduino\libraries\ESP8266WiFi\src/ESP8266WiFi.h:41,
                 from C:\Users\Mayor\Documents\Arduino\test\test2\test2.ino:1:
C:\Users\Mayor\Documents\Arduino\libraries\ESP8266WiFi\src/BearSSLHelpers.h:148:34: error: 'virtual const unsigned char* BearSSL::HashSHA256::oid()' marked override, but does not override
     virtual const unsigned char *oid() override;
                                ^
exit status 1
Error compiling for board Generic ESP8266 Module.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

char ssid[] = ""; // replace with your ssid & pass
char pass[] = "";

void setup() {
}

void loop() {
}

```

### Debug Messages

```

jandrassy, 2019-10-14:
delete the library from libraries. it is part of the core package

mayor89, 2019-10-14:
Hello Juraj. I did what you said and assume that this will work:

`#include <WiFi.h>
#include <WiFiClient.h>
#include <WiFiServer.h>
#include <WiFiUdp.h>

char ssid[] = ""; // replace with your ssid & pass
char pass[] = "";

void setup() {
}

void loop() {
}`

but I get following error:

`C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries\WiFi\src\utility\spi_drv.cpp:21:17: fatal error: SPI.h: No such file or directory

 #include <SPI.h>`

and if I add the SPI library>

`test2:5:17: error: Please import the SPI library from the Sketch > Import Library menu.
As of Arduino 0019, the Ethernet library depends on the SPI library.
You appear to be using it or another library that depends on the SPI library.`

Now I am little confused as SPI should be present in the core package also right?
I also reinstalled the IDE but it is not helping at all.
The program I pasted in the problem description was working fine a year ago and it is working on my ESP8266 since then.. What am I missing or not understanding?

mayor89, 2019-10-14:
Update of board helped my to solve the issue. I am sorry for this useless thread.

closed 2019-10-14 by mayor89