esp8266/Arduino #6641: SPIFFS Data corruption with ESP-07 (1 MB flash size)

euphi, 2019-10-14:
Since some weeks I have the problem that writing to SPIFFS corrupts data, if the program is running on an ESP-07 Module (1 MB flash size).

I use platformio to build and upload application and SPIFFS. (So `mkspiffs` and `esptool` are used).

The generated and uploaded SPIFFS can be read from the application (but I only tried with some small config files), but as soons as I write to SPIFFS from the application, **the changed (or newly created) file is corrupted and reading from it results in reading only NULL-bytes.**

The same application runs correctly on ESP-12/D1 Mini (4 MB flash size).

I tried with the `eagle.flash.1m64.ld` linker script, but also with `eagle.flash.512k32.ld`.

With "512k 32k" data is also corrupted, but FSInfo stills shows a usuable SPIFFFS size of arround 56kb (same as für 64k). Why?

* Version of ESP Arduino Core is stable 2.5.2
* Example of source-code that does not work correctly on ESP-07: https://github.com/fablabnbg/ESP_HomieDoorControl/blob/master/src/HomieDoorOpener.cpp
_(but I also tried much simpler program, that just format SPIFFS, opens a file, writes to it, close the file, open it again and read daSPIFFS and read back data)_
* See also: https://community.platformio.org/t/spiffs-and-esp-07-data-corruption/9951

-----------------------------------

### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [na] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					ESP-07 - NOK // ESP-12 OK
- Core Version:			2.5.2
- Development Env:		Platformio
- Operating System:		Ubuntu

### Settings in ~~IDE~~ platformio

- Module:  			esp-07 (NOK) - d1_mini (OK)
- Flash Mode:			qio
- Flash Size: 		1MB - NOK // 4 MB - OK
- lwip Variant:   not relevant
- Reset Method:		ck, but not releavnt
- Flash Frequency:	not relevant
- CPU Frequency:		not relevant
- Upload Using:		not relevant (uploading SPIFFS via Serial is OK)
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Detailed problem description goes here.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch
```C++
	/* DEBUG-Code */
	File testfile = SPIFFS.open("/data/test.json", "w");
	size_t rc_test = testfile.print("Test!\nTest 2!");
	Serial.printf("Wrote %d byte to JSON-Configfile.\n", rc_test);
	testfile.close();
	/* end DEBUG-Code */

	/* DEBUG-Code */
	String buffer;
	if(SPIFFS.exists("/data/test.json")) {
		File readfile = SPIFFS.open("/data/test.json", "r");
		while (readfile.available()) {
			buffer = readfile.readStringUntil('\n');
			Serial.println(buffer); //Printing for debugging purpose
		}
	}
```
* Only 0-Bytes are read, so the above code runs extremely slow (filling `String buffer` until it is full and then prints an empty line takes > 1 sec per line). (This is plausible behaviour when reading 0-Bytes).

* see also link above

### Debug Messages
* none

TD-er, 2019-10-14:
Can you see what flash brand is used? (using esptool.py for example)

earlephilhower, 2019-10-14:
@TD-er, didn't you also see SPIFFS corrupting files on a smaller FS after some time (the GC() call was added for this, I think)...

TD-er, 2019-10-14:
Yep, the main problem was that writing may return success while the data could not be written.
I noticed it when appending to a file, since you then can check the success by looking at the file size.

So it could be the corruption also happens here due to caching which may result in false positive write return values.

euphi, 2019-10-15:
Yes, in my case, writing returns the correct number of bytes written, so everything seems successful.

I will try to get more information about flash size this evening (+7 hours from now).

TD-er, 2019-10-15:
The thing is, when calling the write on a file, you don't know whether you need to erase a block.

Very short intro in how flash/SPIFFS works and where the write error may come from:

The flash can only change a 1 into a 0. To reset to 1 again, you need to erase the block which can only be done on a large block (typical 8k)
That's 32 pages of 256 bytes.
If you overwrite data in a file, you must for sure write to a new page. If you append to a file, it depends on where in the page you are whether you must write to a new page or not.
As long as there are unused and erased pages, there is no problem.
When there are no free pages, the garbage collector must try to erase a block, so it must look for 32 pages marked as "no longer used" and all belonging to the same block.

In order not to block the write too long, this writing + calling the garbage collector may be done in a later stadium, but the write function must return a value on how many bytes it promises to write.
If in a later stage the garbage collector appears to be unsuccessful, then it cannot write all data.
This means the data did not end up in the file where it was supposed to end up.

Now the really nasty part.
SPIFFS does use up-to 2 blocks (16kByte) for its internal administration. On top of that, on a fragmented filesystem it may appear there is enough free space. But if (worst case scenario) the "blocks marked as unused" are all scattered over blocks which have each 1 page still used by another file, then you cannot erase a block.

The garbage collector in SPIFFS does have a parameter which can allow a number of pages to be moved in order to erase a block, but that value is not exposed to the Arduino level. So calling the garbage collector only erases, and does not move any page.

When handling a file which is relatively large compared to the filesystem, then it will be very likely you will hit this issue.
You could overcome this by opening the whole file, read it in memory, do the changes, delete the file, call the garbage collector and write it again.
But then you must make sure there are no open handles to the file and the unit does not crash/reboot in between.

euphi, 2019-10-15:
I forgot to mention something important: When uploading a SPIFFS "image" with esptool, there is an error message when uploading to ESP-07:

```
[...]
Configuring flash size...
Auto-detected Flash size: 1MB
Compressed 65536 bytes to 185...
Wrote 65536 bytes (185 compressed) at 0x000eb000 in 0.0 seconds (effective 25971.4 kbit/s)...

A fatal error occurred: Timed out waiting for packet header
*** [uploadfs] Error 2
```

However, as already said, reading the (small config) files from SPIFFS is ok.

devyte, 2019-10-15:
@euphi please retest with latest git. It's possible that this is a symptom of a stability issue that was fixed after 2.5.2.
Instructions for installing are on readthedocs.

euphi, 2019-10-15:
Well. It seems that latest git resolves this issue.

The test sketch opens a uploaded file and prints it, the creates a new file, writes to it, closes it and opens it again to print it. Interestingly, this time also the uploaded file cannot be read from 2.5.2?!

Debug output with git-version:
```
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v40f456ac
~ld

SDK:2.2.1(cfd48f3)/Core:2.5.2-195-g40f456ac=20502195/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-16-ge23a07e/BearSSL:89454af
SPIFFSImpl: allocating 512+240+1400=2152 bytes
SPIFFSImpl: mounting fs @eb000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=0

{ "test": 1 }

SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loading configuration...
File system (SPIFFS): 
 Total KB: 51.72 KB
 Used KB: 0.49 KB
Formatting SPIFFS
SPIFFSImpl: mounting fs @eb000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=0
SPIFFSImpl: mounting fs @eb000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=0
File system (SPIFFS): 
 Total KB: 51.72 KB
 Used KB: 0.00 KB
Saving configuration...
SPIFFS_remove: rc=-10002 path=`/config.txt`
SPIFFS_close: fd=1
SPIFFS_close: fd=1
File system (SPIFFS): 
 Total KB: 51.72 KB
 Used KB: 0.49 KB
Print config file...
Test!
2,3,4

SPIFFS_close: fd=1
SPIFFS_close: fd=1

```

Debug output with 2.5.2:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
SPIFFSImpl: allocating 512+240+1400=2152 bytes
SPIFFSImpl: mounting fs @eb000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=0
SPIFFSImpl::open: fd=-10002 path=`/data/config.json` openMode=0 accessMode=1 err=-10002
Failed to read file
Loading configuration...
File system (SPIFFS): 
 Total KB: 51.72 KB
 Used KB: 0.25 KB
Formatting SPIFFS
SPIFFSImpl: mounting fs @eb000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=0
SPIFFSImpl: mounting fs @eb000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=0
File system (SPIFFS): 
 Total KB: 51.72 KB
 Used KB: 0.00 KB
Saving configuration...
SPIFFS_remove: rc=-10002 path=`/config.txt`
SPIFFS_close: fd=1
SPIFFS_close: fd=1
File system (SPIFFS): 
 Total KB: 51.72 KB
 Used KB: 0.49 KB
Print config file...
SPIFFS_read rc=-10012
�SPIFFS_read rc=-10012
�SPIFFS_read rc=-10012
�SPIFFS_read rc=-10012
�SPIFFS_read rc=-10012
�SPIFFS_read rc=-10012
�SPIFFS_read rc=-10012
�SPIFFS_read rc=-10012
�SPIFFS_read rc=-10012

```

devyte, 2019-10-23:
Glad to hear it. I'm closing as already fixed.

closed 2019-10-23 by devyte