esp8266/Arduino #6642: Light Sleep Mode not working

downsider7, 2019-10-14:
I've followed the steps in the ESP8266 Non-OS SDK API Reference manual (v2.0.0,2016) page 97

Running on SDK 2.2.1, Core 2.5.2
Board ESP-12E

`Serial.println("going to sleep")
wifi_station_disconnect();
wifi_set_opmode(NULL_MODE); // set WiFi mode to null mode.
wifi_fpm_set_sleep_type(LIGHT_SLEEP_T); // light sleep
wifi_fpm_open(); // enable force sleep
wifi_fpm_set_wakeup_cb(fpm_wakup_cb_func1); // Set wakeup callback
wifi_fpm_do_sleep(50*1000000);
delay(1000);
Serial.println("Awake")`

What I see is the awake message 1 second after the going to sleep message. (I've tried with delays of 1,2,5 or 10s just incase it was taking time to sleep)

The current consumed is 14mA.
The callback function is never triggered.
The wifi_fpm_do_sleep() returns a status of 0, indicating success.

Deep sleep works, the current drops to almost 0.
Modem sleep works.

I can't make Light sleep work at all. The RF should be off, the CPU unclocked.

Does light sleep work? am I doing something daft?

Tech-TX, 2019-12-30:
@downsider7  try this
```cpp
pinMode(WAKE_UP_PIN, INPUT_PULLUP);
// ...
WiFi.mode(WIFI_OFF);
delay(100);
Serial.println("Going to sleep, pull Dxx low to wake me");
delay(3);  // needs a brief delay or the whole message above doesn't print
wifi_fpm_set_sleep_type(LIGHT_SLEEP_T);
gpio_pin_wakeup_enable(GPIO_ID_PIN(WAKE_UP_PIN), GPIO_PIN_INTR_LOLEVEL);
// or HILEVEL/ANYLEVEL, no change allowed, sorry
wifi_fpm_open();
wifi_fpm_do_sleep(0xFFFFFFF);
delay(100);
Serial.write("Woke up!");
```
I'm not using the callback, but the code above puts it in Forced Light Sleep for me.  After the GPIO interrupt it falls through to the 'Woke up!' message.

Forced Light Sleep only works with wifi_fpm_do_sleep(0xFFFFFFF); any other value and it doesn't sleep.

Tech-TX, 2020-01-02:
Here's an expanded set of low power tests with every power-saving mode I could think of or read about.  If I've missed any, please let me know and I'll expand it a bit further. :smile:  This should work with any build from 2.5 onward, unknown with builds before that but probably not.
```cpp
// This is an example of the different low-power modes of the ESP8266
//
// My setup was a WeMos D1 Mini with 3.3V connected to the 3V3 pin through an ammeter
// so that it bypassed the on-board voltage regulator and USB chip.  There's still about
// 0.7mA worth of leakage current due to the unpowered chips, so an ESP-01 will show lower
// current readings than what I could achieve.  These tests should work with any module.
// While the modem is connected the current is 67mA or jumping around with a recorded minimum.
//
// Since I'm bypassing the USB chip, I've included OTA upload.  You'll need to upload
// from USB or a USB-to-TTL converter the first time, then you can disconnect and use OTA
// afterwards during any test if the WiFi is connected.  Some tests sleep WiFi so OTA won't go.
//
// This test assumes you have a pushbutton switch connected between D3 and GND.  This is used
// to advance the tests.  You'll also need to connect D0/GPIO16 to RST for the Deep Sleep tests.
// Connect an LED from any free pin through a 330 ohm resistor to the 3.3V supply, NOT the 3V3
// pin on the module or it adds to the measured current.  When it blinks you can go to the next test.
//
// This example code is in the public domain.

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

#define WAKE_UP_PIN D3  // GPIO0, can be used to force a flash upload with RESET if you're stuck
#define LED D1  // blinks when waiting for a button press
//#define LED D4  // GPIO2 activity indicator for ESP-01 modules as D4 is LED_BUILTIN on others

ADC_MODE(ADC_VCC);  // allows us to monitor the internal VCC level, and it varies with WiFi load

// Replace with your network credentials
const char* AP_SSID = "SSID here";  // your router's SSID
const char* AP_PASS = "password here";  // your router's password

// CRC function used to ensure data validity of RTC User Memory
uint32_t calculateCRC32(const uint8_t *data, size_t length);

// Structure which will be stored in RTC memory to remember the reset loop count.
// First field is CRC32, which is calculated based on the rest of the structure contents.
// Any fields can go after the CRC32.  Structure must be 4-byte aligned.
struct {
  uint32_t crc32;
  byte data[4];  // the last byte stores the reset count
} rtcData;

byte resetLoop = 0;  // keeps track of the number of Deep Sleep tests / resets

void setup() {
  pinMode(LED, OUTPUT);  // LED wired to 3.3V through a 330 ohm resistor (no-load activity indicator)
  pinMode(WAKE_UP_PIN, INPUT_PULLUP);  // polled to advance tests, INTR for the Forced Light Sleep test
  Serial.begin(74880);
  if (ESP.getResetReason() == "External System") {
    resetLoop = 5;
    updateRTC();  // if external reset (RESET button), wipe the RTC CRC and start all over
  }

  // Read struct from RTC memory
  if (ESP.rtcUserMemoryRead(64, (uint32_t*) &rtcData, sizeof(rtcData))) {
    uint32_t crcOfData = calculateCRC32((uint8_t*) &rtcData.data[0], sizeof(rtcData.data));
    if (crcOfData != rtcData.crc32) {
      Serial.println(F("CRC32 in RTC memory doesn't match, first loop after power up?"));
      resetLoop = 0;  // first test loop since power on or external reset
    } else {
      resetLoop = rtcData.data[3];  // read the previous reset count
    }
  }
}

[... 261 lines cut ...]
> sixth test - Forced Light Sleep using NONOS SDK calls
> press the button to continue
> CPU going to sleep, pull WAKE_UP_PIN low to wake it up
> The internal VCC reads 2945, current = 0.4mA
> Woke from Forced Light Sleep - this is the callback
> Woke up!
> seventh test - Deep Sleep for 10 seconds, wake with RF_DEFAULT
> The internal VCC reads 2945, current = 0.4mA during Deep Sleep
> going into Deep Sleep now
> <boot message>
> eighth test - currently in RF_DEFAULT, Deep Sleep for 10 seconds, wake with RFCAL
> The internal VCC reads 2943, current = 67mA
> press the button to continue
> <boot message>
> ninth test - currently in RFCAL, Deep Sleep for 10 seconds, wake with NO_RFCAL
> The internal VCC reads 2945, current = 67mA
> press the button to continue
> <boot message>
> tenth test - currently in NO_RFCAL, Deep Sleep for 10 seconds, wake with RF_DISABLED
> The internal VCC reads 2945, current = 67mA
> press the button to continue
> <boot message>
> Tests completed, currently in RF_DISABLED, press the button to do an ESP.restart()
> The internal VCC reads 3226, current = 15mA
> <end of tests, it restarts from here>

downsider7, 2020-01-02:
Thank you for this - I will start having a look at it as soon as I get home from work.

:-)

downsider7, 2020-01-02:
@Tech-TX Looking through your comments, it seems to confirm what I have found - you can only wake from light forced sleep with an I/O pin - you can't wake from that state with a sleep time OR a pin wiggle wake.

I need to think through my application and see if I can work with that, or whether its possible to get a pin wiggle from the RTC and simply diode-OR that with my external wake-up signal.

Thank you for confirming "Forced Light Sleep only works with wifi_fpm_do_sleep(0xFFFFFFF); any other value and it doesn't sleep." - that saves me a lot of heartache and frustration thinking I was doing something wrong.

Tech-TX, 2020-01-03:
I edited the file a little to make it more user friendly, but it still works the same.  I removed everything that wasn't absolutely needed to put the chip into the various power modes.  There's a LOT of examples of people with different solutions, many now out of date.

The table in the Espressif 'Low Power Solutions' PDF is both out of date and incomplete or confusing  on the various low-power modes.  Here's my revision to it:

[image]

I haven't re-verified the average current in the 3 DTIM modes as I don't yet have a good way to do that.  It's hard to calculate average power by visually reading a Fluke meter. :stuck_out_tongue_closed_eyes:

d-a-v, 2020-01-06:
@Tech-TX 

>  There's a LOT of examples of people with different solutions, many now out of date

This is true. There is also #6356 which tries to address this issue, with its [example](https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/WiFiShutdown/WiFiShutdown.ino).

Would you have a chance to test it with your hardware tools, and possibly update the core functions with your findings, in #6989 ?

Tech-TX, 2020-01-11:
@d-a-v I'm sorry David, I didn't notice your request until late last night.  It looks like your example runs OK and does what you're expecting, although I'm having a hard time following what your state machine is doing on the scope as it's frequently changing states.  Was there anything in particular you wanted me to look at?

I get slightly different current readings on an unmodified clone D1 Mini:
 ~80mA during normal operations
 ~25mA during wifi shutdown
  ~6mA during deepsleep

compared to your readings
 ~85mA during normal operations
 ~30mA during wifi shutdown
  ~5mA during deepsleep

On the board I've cut the chips off, it's even lower:
 ~71mA during normal operations
 ~18mA during wifi shutdown (for some reason I'm not getting the 15mA I got last weekend)
  ~16uA during deepsleep

I agree on updating the core functions to force the selected sleep mode independent of what the user has done.  That eliminates a couple of the extra steps I had to do in my example.  I see you're doing a delay() after WIFI_OFF in one place.  I've just verified that the minimum is either delay(5) or yield() to insure that it goes into shutdown, otherwise it's occasionally erratic.

Tech-TX, 2020-01-12:
David, the only thing I noted was that **WiFi.mode(WIFI_RESUME)** doesn't bring it out of Forced Modem Sleep.  It turns the modem on (current goes up to 67mA) but there aren't any spikes from the WiFi trying to reconnect.
The **WiFi.mode(WIFI_SHUTDOWN)** works but requires **WiFi.forceSleepWake()** to bring the WiFi back up.

d-a-v, 2020-01-17:
@Tech-TX 
Your findings are very valuable.
My initial attempt was to hide this complexity under an updated API for, like [you say in #6989](https://github.com/esp8266/Arduino/pull/6989#issue-359372651), helping others. This was an old idea made real by #6356 (see refs inside) with the same motivation as yours.
I knew SHUTDOWN and RESUME wouldn't be perfect (I am glad anyway that we have close numbers when regulator and serial chips are VCCed).
You claim my example and its state machine is not clear. And despite my efforts you are right. Yours is also quite hairy :) This highlights the fact that it ***needs*** to be hidden under an API for everyone's mind health.
You seem to have reviewed the API update from #6356 and have proposals to fix it.

What do you think of adding in #6989 : 
- polishing the API (#6356) with your findings
- adding another example using the API that would do what your #6989 example does
  so users reading the examples would know what the (fixed) API would do for them
- possibly removing my example (which only purpose is to show the API)

It may be too much to do at once but your feeling about that is interesting to know.

Tech-TX, 2020-01-18:
David, I think you're taking offense where none was meant.  I didn't say your example was unclear, only that I couldn't follow the state changes on the 'scope.  It was changing too frequently to take more than a brief measurement, and I'd have needed to bring out a bunch of test points to the pins to add into my mixed-signal oscilloscope so I could identify each state and the result.

That's why I stopped everything at each of the tests: it allows the user to take a breath and look at what it's doing in each sleep mode.  If they're trying to integrate average current over time then it's simpler when it's only doing one thing.  I don't yet have a way to monitor average current/amperage, but I have all the parts.  It just needs time and code.

My snarky comment in the README about '_poor examples_' was what I'd found on arduino.cc, Espressif's BBS and esp8266.com.  I hadn't seen your example until you mentioned it, as 'WiFiShutdown' doesn't immediately bring to mind 'sleep modes'; I knew how to turn WiFi off already.   :smiley: I'd read a lot of pages from _other_ people trying to get different types of sleep to work, and many were failing.  That's why I did a demonstrator of the minimum required to get into all of the sleep modes.

From what I understand of your demo, it's a stress-test trying to see if repeated WiFi connects/resumes are OK across Deep Sleep and/or Forced Modem Sleep, and it illustrates the new SHUTDOWN/RESUME class.

You don't need to do anything with https://github.com/esp8266/Arduino/pull/6356 as it works. I wasn't able to use the WiFi.mode(WIFI_RESUME) as I can't figure out _how_ it works.  I'm only barely functional at C and not competent at C++ so I don't understand much of ESP8266WiFiGeneric.cpp and the relevant parts of WiFiShutdown.ino.  I presume it's C++ as I don't recognize the symbols in those positions.

d-a-v, 2020-01-19:
No offense taken :) I take for granted that some of my examples are not clear enough.
Regarding the WiFiShutdown-to-be-renamed one, we can make the delay configurable between states.
Regarding `WIFI_RESUME`, `::resumeFromShutdown()` is called in which wifi is restarted with what was saved in the nv/rtc-ram-argument.

Anyway I look forward your example to be merged so I can try and improve the internal API then propose another example based on yours but using the API in the usual arduino way (simple example so one help me until it is simple enough).

Tech-TX, 2020-01-20:
Yes, I understand what you're doing and how... I got that quickly from simple inspection.  The SHUTDOWN routine stores the details from the connection in the WiFiState struct (if it exists), and the RESUME then uses that struct to restore the connection more quickly.  It's a nice arrangement, and essentially halves the reconnect time (or faster).  That would be handy as my router is VERY slow at reconnects at times, taking over 20 seconds.  The pointers are what's tripping me up, as I've only dealt with them from the _other_ direction (writing assembly drivers that our C folks used).  I never got into C until summer, so I'm going through the learning pains now that you sailed through years ago. ;-)

Tech-TX, 2020-01-22:
I'm almost done integrating your new modes into my tests,  just need to fix one pointer I goofed up last night.  I'm getting about the same resume times you are,  which is encouraging.  Twice as fast on reconnect is great!

Tech-TX, 2020-01-23:
@d-a-v am I correct in assuming that WiFi.mode(WIFI_RESUME, (struct)) invalidates the stored CRC / struct once it's done resuming?  That's the effect I'm seeing, and I'm not sure if it's intended.  I can resume once, but when trying to resume a second time the saved state is no longer valid.  It only works once after doing WiFi.mode(WIFI_SHUTDOWN, (struct)).  I could copy the whole struct to another area of RTC memory as it's only 152 words long, and then copy it back if the saved state isn't valid.

I've updated the demo above with my latest version, including pasting bits of your WiFiShutdown code in.  It's all working OK.

d-a-v, 2020-01-23:
Yes, nvram data is [invalidated on success](https://github.com/esp8266/Arduino/blob/158039e41447542f0945b5864d79647aa7e48545/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp#L790-L791) on purpose.

> It's all working OK.

Thanks for testing this api and putting it into a better example !

Tech-TX, 2020-01-24:
Not better, merely different.  You were looking at other problems than I was.  I was frustrated in summer trying to get the Sleep modes to work, and the documentation didn't help much.  When I saw downsider7 struggling with the same issues I'd had, I **needed** to correct the docs or make them more clear.

I was a teacher for a while (and still am occasionally) so I'm a bit more _wordy_ than most people.  I write for 14 year old kids, as that's about when I started microprocessors back in the Dark Ages with a CDP1802 as my first computer.  The ESP8266 with the IDE is a much nicer 'first computer' today.

I think I have a final copy of the demo, just going through it line-by-line to make sure everything is correct and works where I give them easy options to change things.  It's both a demo and an experimenter's kit.

Tech-TX, 2020-01-25:
@d-a-v maybe you can suggest something here.  I'm trying to get the **preinitWiFiOff();** to work, as it apparently got broken sometime in the last (year?).  I presume it worked at one point.  
The problem is two-fold:

1) Forced Modem Sleep won't reliably go to sleep without the CPU executing an '_idle task_'
[image]
I have 1 CPU that's 2 years old and several that are new, and I can verify that all 6 chips need at least **delay(1)** to sleep the modem.

2) **delay()** in **core_esp8266_wiring.cpp** has not been initialized before the **preinit()** stage of boot, so attempting to call **delay(10)** inside **ESP8266WiFiGeneric.cpp** returns immediately without calling **esp_schedule()** for 10ms (the '_idle task_').  

Here's the quick test I did, and what I attempted in **ESP8266WiFiGeneric.cpp** to fix it so that **preinitWiFiOff()** works again.  The circled numbers relate to the pin wiggles further below so you can see what is happening, and when:

[image]

Here's the result on the 'scope, first a wide view of the boot + setup(), then zooming in on the **preinit()**, zooming again, then scrolled over to **setup()** in the last screen.  The top trace is my pin wiggle, the lower trace is CPU amps/current.  The time per division for the 'scope is at the top of the display.  The last two screens are 100us per division.

[image]
[image]
[image]
[image]

**wifi_set_opmode_current(NULL_MODE);** is the same as **wifi_set_opmode_current(WIFI_OFF);** as both variables are zero.  I'd copied my code (cloned and fixed from the Espressif example) in place of what's in **ESP8266WiFiGeneric.cpp** for the **preinitWiFiOff()** function during this investigation.  The same sequence is used in setup() in the example.  The delay() in **preinitWiFiOff()** never executes, nor does the one at the bottom of **preinit()**.  Short of cloning the **delay()** code into  **preinitWiFiOff()** so that I can force it to idle, I don't see a way to make this work.

Here's the example I used above with GPIO4 used for the pin wiggles:

```c
#include <ESP8266WiFi.h>

void preinit() {
  pinMode(4, OUTPUT);
  digitalWrite(4, LOW);  // pin wiggle
  ESP8266WiFiClass::preinitWiFiOff();  // currently not working
  delay(10);  // never executes
  digitalWrite(4, LOW);  // pin wiggle
}

void setup() {
  pinMode(4, OUTPUT);
  digitalWrite(4, HIGH);  // pin wiggle
  digitalWrite(4, LOW);  // pin wiggle
  digitalWrite(4, HIGH);  // pin wiggle
  digitalWrite(4, LOW);  // pin wiggle
  wifi_set_opmode_current(NULL_MODE);  // set Wi-Fi working mode to unconfigured, don't save to flash
  wifi_fpm_set_sleep_type(MODEM_SLEEP_T);  // set the sleep type to modem sleep
  wifi_fpm_open();  // enable Forced Modem Sleep
  wifi_fpm_do_sleep(0xFFFFFFF);  // force modem to enter sleep mode
  delay(1);  // without a delay() here it doesn't reliably enter sleep
  digitalWrite(4, HIGH);  // pin wiggle
  digitalWrite(4, LOW);  // pin wiggle
  digitalWrite(4, HIGH);  // pin wiggle
}

void loop() {
}
```

Tech-TX, 2020-01-26:
Never mind, I just found something that accomplishes nearly the same end that igrr did back in 2016.  Tested, and it _somewhat_ works with the current git.  His old code from back then doesn't work any longer as-written.
https://github.com/esp8266/Arduino/issues/2111#issuecomment-224251391

First cut, turn off the RFCAL in boot, and put as much of the Forced Modem Sleep inside preinit() as possible, with the delay(1) the first thing in setup(),
```c
#include <ESP8266WiFi.h>

RF_PRE_INIT() {
  pinMode(4, OUTPUT);
  digitalWrite(4, HIGH);  // pin wiggle (1)
  system_phy_set_powerup_option(2);  // shut down the RFCAL at boot
  digitalWrite(4, LOW);  // pin wiggle (2)
}

void preinit() {
  pinMode(4, OUTPUT);
  digitalWrite(4, HIGH);  // pin wiggle (3)
  wifi_set_opmode_current(NULL_MODE);  // set Wi-Fi working mode to unconfigured, don't save to flash
  wifi_fpm_set_sleep_type(MODEM_SLEEP_T);  // set the sleep type to modem sleep
  wifi_fpm_open();  // enable Forced Modem Sleep, can't put this in RF_PRE_INIT or it doesn't sleep
  wifi_fpm_do_sleep(0xFFFFFFF);  // force modem to enter sleep mode, in RF_PRE_INIT this causes exceptions
  digitalWrite(4, LOW);  // pin wiggle (4)
  // won't go into Forced Modem Sleep until it sees the delay() in setup()
}

void setup() {
  digitalWrite(4, HIGH);  // pin wiggle (5)
  delay(1);  // G'night, John-Boy
  digitalWrite(4, LOW);  // pin wiggle (6)
}

void loop() {
}
```
and the result is:
[image]

Second cut, move as much inside **RF_PRE_INIT()** as will work, saving another 200+us of that modem power burst,
```c
#include <ESP8266WiFi.h>

RF_PRE_INIT() {
  pinMode(4, OUTPUT);
  digitalWrite(4, HIGH);  // pin wiggle (1)
  system_phy_set_powerup_option(2);  // shut the RFCAL at boot
  wifi_set_opmode_current(NULL_MODE);  // set Wi-Fi working mode to unconfigured, don't save to flash
  wifi_fpm_set_sleep_type(MODEM_SLEEP_T);  // set the sleep type to modem sleep
  digitalWrite(4, LOW);  // pin wiggle (2)
}

void preinit() {
  pinMode(4, OUTPUT);
  digitalWrite(4, HIGH);  // pin wiggle (3)
  wifi_fpm_open();  // enable Forced Modem Sleep, can't put this in RF_PRE_INIT or it doesn't sleep
  wifi_fpm_do_sleep(0xFFFFFFF);  // force modem to enter sleep mode, in RF_PRE_INIT this causes exceptions
  digitalWrite(4, LOW);  // pin wiggle (4)
  // won't go into Forced Modem Sleep until it sees the delay() in setup()
}

void setup() {
  digitalWrite(4, HIGH);  // pin wiggle (5)
  delay(1);  // G'night, John-Boy
  digitalWrite(4, LOW);  // pin wiggle (6)
}

void loop() {
}
```
Slightly better power management, but you still can't boot it with a 50mA power supply unless you have a supercap to hold the power up during the modem turn-on,  That's too long of a burst for a regular electrolytic cap to supply.
[image]

I don't think it's worth trying to move the delay() inside **preinit()** as that only saves another 150us of that 70mA modem burst,  The effort would be better spent moving the whole thing inside **early_init()** as that previously happened before the modem turns on, but I can't find **early_init()** in the source tree any longer.  Looks like the second cut above is the best we'll get today.

[image]

d-a-v, 2020-01-27:
Thanks for the very fine detailed analysis and clear explanation.
I understand your current issue is `delay()` not able to run at that step of the booting process: `delay()` is executed without error but no delay is really introduced.

- Did you try to call `delay(0)` or `esp_schedule()` multiple times for 1ms (using `esp_get_cycle_count()`) ?
  (side note: I don't really know the effect of `esp_schedule()` which is `ets_post(LOOP_TASK_PRIORITY, 0, 0);` before the main arduino user task is started)
- What happens when all `preinit()` code is moved to `RF_PRE_INIT()` ?
- If it is better to have both functions, you can try to save more usecs by moving `preinit()` earlier from inside its caller https://github.com/esp8266/Arduino/blob/0989932669f027c5e9d7a5c703a4811dc3106c65/cores/esp8266/core_esp8266_main.cpp#L323

Tech-TX, 2020-01-31:
@d-a-v Calling either delay(0) or esp_schedule() 1000 times inside preinit() adds about a millisecond to the preinit(), but doesn't turn off the modem.  I also tried both with a 2000 count, adding 2ms: still no joy.

When I move 
```c
  wifi_fpm_open();
  wifi_fpm_do_sleep(0xFFFFFFF);
```
The wifi_fpm_open() apparently doesn't execute in RF_PRE_INIT(), as it won't sleep.  When I also move  wifi_fpm_do_sleep(0xFFFFFFF)  inside RF_PRE_INIT() I get this in loop():

>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x4010f000, len 3456, room 16 
> tail 0
> chksum 0x84
> csum 0x84
> vfecacfd5
> ~ld

which _looks_ like a software WDT every 4 seconds, but without the WDT message and no stack trace.  

From what I can glean from the API reference, 'idle' (delay) does _something_ in the blob or CPU that allows the power management modes to execute.  It needs the OS_timer running, and I'd guess it's not initialized before preinit() happens.  I found delay() in core_esp8266_wiring.cpp but don't understand what happens there as I haven't played with the OS_timer.  It sets & arms the timer, executes a yield(), then disarms the timer and falls through.  I get lost in the yield() and esp_schedule() code, and don't see control passing to the blob anywhere there.

Tech-TX, 2020-02-01:
@downsider7 I don't know if you've been following my pull request for the demo, but I got Timed Light Sleep running several days ago.  It's solid.  On the boards that don't have voltage regs or USB chips, I read ~ 360uA when it's in Timed or Forced Light Sleep.  Since the Timed version is only using 7 nybbles for the time, the maximum timed sleep is 2^28-1 microseconds, ~ 268 seconds.  I *really* wish they'd used a millisecond timebase for the timed sleep modes instead of microsecond.  The microsecond 'accuracy' is a complete waste, due to the irregularity in restarting the CPU clock (baseband PLL).  Having a sleep that would last over a day would have been more handy.

TD-er, 2020-02-01:
> I read ~ 360uA

Just curious, how do you measure those low currents?
It is very easy to be off by a big factor with those low currents.
Still it will be very clear to see if a node is in sleep mode or not, but for truly deep sleep applications it is good to know the right order of current it draws.

Tech-TX, 2020-02-01:
@TD-er I'm running a Wavetek DM27XT DVM in-line with the power to the chip.  The DVM's minimum range is 200uA full-scale, and it goes up to a 10A range.  It's repeatable and at least as accurate as my Fluke, but has a lower current range than the Fluke does (2mA full-scale minimum range).  I'm also monitoring on the 'scope with a 1 ohm shunt, but the 'scope isn't nearly as accurate.

devyte, 2020-02-02:
Given that #6989 is merged, and the length and weight of this thread, I'm closing. For discussions on @d-a-v 's comments about the pseudo modes and a user-facing api, please see #7055. For other power mode issues, please open a new issue and fill out the required info.

closed 2020-02-02 by devyte