esp8266/Arduino #6644: Wemos d1 mini PRO 16MB SPIFFS.begin() fails

Buffalchill, 2019-10-15:
#### Platform

- Hardware:					[ESP-12|
- Core Version:			2.5.1
- Development Env:		[Arduino IDE 1.8.2
- Operating System:		[Windows|

### Settings in IDE

- Module:  			Wemos D1 mini PRO
- Flash Size: 		        16MB
- Flash Frequency:	        [40Mhz]
- CPU Frequency:		[80Mhz|
- Upload Using:		|SERIAL]

### Problem Description

Hello,
I used the SPIFFS / SD card example to test my WEMOS d1 mini PRO but it seems like it doesn't work with 16MB flash size.
When i choose Wemos d1 mini r2  it works with 4mb flash size.
Is there a trick how it will work with 16MB flash size ?

######### Sketch

#include "FS.h"
FS* filesystem = &SPIFFS;

File myFile;

void setup() {
  // Open serial communications and wait for port to open:
  Serial.begin(115200);

  Serial.println();
  Serial.println("Initializing SPIFFS ...");

  if (!SPIFFS.begin()) {
    Serial.println("initialization failed!");
    return;
  }
  Serial.println("initialization done.");

  // open the file. note that only one file can be open at a time,
  // so you have to close this one before opening another.
  myFile = SPIFFS.open("test.txt", "w");

  // if the file opened okay, write to it:
  if (myFile) {
    Serial.print("Writing to test.txt...");
    myFile.println("testing 1, 2, 3.");
    // close the file:
    myFile.close();
    Serial.println("done.");
  } else {
    // if the file didn't open, print an error:
    Serial.println("error opening test.txt");
  }

  // re-open the file for reading:
  myFile = SPIFFS.open("test.txt", "r");
  if (myFile) {
    Serial.println("test.txt:");

    // read from the file until there's nothing else in it:
    while (myFile.available()) {
      Serial.write(myFile.read());
    }
    // close the file:
    myFile.close();
  } else {
    // if the file didn't open, print an error:
    Serial.println("error opening test.txt");
  }
}

void loop() {
  // nothing happens after setup
}

devyte, 2019-10-15:
>  it doesn't work with 16MB flash size.

That doesn't help much, please explain. 

There is a known limitation with using SPIFFS with larger than 3MB size: access time increases too much and you run into wdts. There is an idea to test whether larger block/page size for those large FS size cases helps, but it hasn't really been explored. 

Other than that, update your core and try LittleFS instead of SPIFFS.

Buffalchill, 2019-10-15:
The explanation is:
I choose **Flash Size 16M (14M SPIFFS)**  or *Flash Size 16M (15M SPIFFS)**
And these are the only options i have for this Board.
It is implemented into the Arduino IDE, so i thought it would work.
But, SPIFFS.begin()  returns false and the SPIFFS are not mounted after calling the funtion.

MarkyAD, 2019-10-16:
Try manually formatting the filesystem in your sketch with SPIFFS.format(); Or create a directory inside your sketch directory called "data". In the arduino IDE choose "ESP8266 Sketch Data Upload" from the Tools menu. Formatting a 15M partition takes a little time, be patient. I have several of these boards and they all work with 15M SPIFFS.

devyte, 2019-10-31:
I just tested the above sketch built with today's latest git on a wemos d1 pro, wiped the entire flash, and I get the following output:
```
Initializing SPIFFS card...
initialization done.
Writing to test.txt...done.
test.txt:
testing 1, 2, 3.
```

The begin() call takes its sweet time, but it does work.

> Core Version: 2.5.1

A whole lot has been fixed since 2.5.1. Also, there is a reason that the issue template has a checkbox for testing with latest git.

Closing as can't reproduce.

smartgeocaching, 2019-11-06:
Hello I have 2.5.1 installed.
But i still see it failing... and now i found out following line showing auto detected Flash size of 4MB in the Compiler Window:

[image]

Buffalchill, 2019-11-06:
Hello I have 2.5.1 installed but i still see it failing... and now i found out following line showing auto detected Flash size of 4MB in the Compiler Window:

[image]

Buffalchill, 2019-11-09:
OK I unistalled my Core and then installed the latest Core with git.
But it still fails...

I also found in the Compiler Window:
[image]

devyte, 2019-11-09:
What is the brand of your flash chip?
Also, look at ESPClass and please print out info about the flash chip.

Buffalchill, 2019-11-09:
Should I blame the seller ?

[image]

[image]

devyte, 2019-11-09:
Your flash chip is cFeon Q32c-104hip. A quick google says it's a 32Mbit, which means 4MB.
In contrast, my wemos d1 pro has a winbond 25Q128.
32 vs. 128 is telling. 
Closing.

closed 2019-11-09 by devyte