esp8266/Arduino #6645: Static ip and dns problems

anITiot, 2019-10-15:
I must use the order as shown "WiFi.config(staticip, dns, gateway, subnet);" to have static ip and dns but I can't use dns2 in the WiFi.config.

IPAddress staticip(192, 168, 1, 61); //ESP static ip
IPAddress gateway(192, 168, 1, 1); //IP Address of your WiFi Router (Gateway)
IPAddress subnet(255, 255, 255, 0); //Subnet mask
IPAddress dns(8, 8, 8, 8); //DNS
IPAddress dns2(8, 8, 4, 4); //DNS2

And the manual says WiFi.config(staticip, gateway, subnet, dns, dns2) but this dosent work for me, it gives me the DHCPaddress but if I discard the DNSaddresses in WiFiconfig.h it takes the static ip but I will not have any dns.

ESP8266 latest and Arduino 1.8.10

Suggestion:
And instead of add subnet mask is it possible to enter 192.168.1.61/24?

jandrassy, 2019-10-15:
`bool config(IPAddress local_ip, IPAddress gateway, IPAddress subnet, IPAddress dns1 = (uint32_t)0x00000000, IPAddress dns2 = (uint32_t)0x00000000);`

anITiot, 2019-10-15:
tested and confirmed, it works for a newbie as me :)
I use the CAPP-method (Copy And Paste Programming) now and learning that way.

closed 2019-10-15 by anITiot

jandrassy, 2019-10-15:
the ESP8266WiFi library config function has different order of parameters then the Arduino WiFi libraries, but it attempts to detect if Arduino order is used. sometimes it works