esp8266/Arduino #6649: Data transfer fast on local network but slow on internet

Vbansal1996, 2019-10-17:
I am experiencing the issue of slowness when transferring to cloud. Example: 400kb file transfer takes approx 1 min 40 seconds on cloud, where-as if I transfer the same data to 192.168.0.5 (local network), it takes 8 seconds. I am using bare code to achieve this. Please give a hint.

d-a-v, 2019-10-17:
Maybe by using lwip2 - "higher bandwidth" ?
Mode details on your setup, what is transferred, and duration when sending the same data from a PC to the same location.

closed 2019-10-17 by devyte

Vbansal1996, 2019-10-18:
Any guidance for Iwip2 ?
300KB of image is transferred
on local server it takes 6-7 sec for full.
But it takes 1 min 40 sec on our live server.

From PC it takes 3 sec to sent full 300KB image data

jandrassy, 2019-10-18:
> Any guidance for Iwip2 ?

it is an option in Tools menu

Vbansal1996, 2019-10-18:
Hi 

It didn't change anything still the response is same
set IwIp2 to HighBandwidth

d-a-v, 2019-10-18:
Can you make a tcpdump/wireshark of the traffic ?

liebman, 2019-10-18:
Could this be HTTP on local connections and HTTPS on "cloud"?

jandrassy, 2019-10-18:
how do you send the file? you didn't attach any code