esp8266/Arduino #6656: ESP-12F crash when upgrading from 2.4.2 to 2.5.0 or later

OmarAlkassab, 2019-10-21:
My code is running well, but when I use the ESP8266WebServer, and the following function:

> MyWebServer.on("/", handleHome);

the esp restart when calling the function:

> MyWebServer.send(200, "text/html", "Hello World");

the following is the debug after crash:

> Exception (3):
> epc1=0x402205a4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x402598e7 depc=0x00000000
> 
> >>>stack>>>
> 
> ctx: cont
> sp: 3ffffc70 end: 3fffffc0 offset: 01a0
> 3ffffe10:  402598e7 00000004 3fff03d8 4020df84
> 3ffffe20:  3ffffe48 ffffffff 3fff03d8 40100191
> 3ffffe30:  00000000 3ffef02c 3ffffe60 00000000
> 3ffffe40:  00000000 fffef02c 3ffef02c 4020ebf4
> 3ffffe50:  ff00746e 00000000 3ffef02c 40100927
> 3ffffe60:  00000000 00000000 ff000054 00000000
> 3ffffe70:  00000000 3ffef02c 3ffeefe8 4021caca
> 3ffffe80:  ff000039 00000000 00000000 401000a9
> 3ffffe90:  00000000 00000000 ff395810 4020bc72
> 3ffffea0:  00000000 ff000000 00000000 00000000
> 3ffffeb0:  ff63656e 640f0f0a 40100b48 00031499
> 3ffffec0:  3ffef010 3ffef09c 00000001 3ffef010
> 3ffffed0:  00000001 00000000 00000001 00000004
> 3ffffee0:  00000000 3fff28bc 3ffeefe8 00000000
> 3ffffef0:  00000001 3ffef010 3ffeefe8 4020be24
> 3fffff00:  4021db60 00000000 3fff2b34 402073ff
> 3fffff10:  3ffef0d0 3ffe8f65 3ffeefe8 40207bb8
> 3fffff20:  3fff0186 3ffe8f65 3ffeefe8 402045c8
> 3fffff30:  40100248 00000000 4021cad4 4021cac0
> 3fffff40:  00000000 00000000 ff00746c 00020035
> 3fffff50:  00000031 3fff01b8 3ffe853c 00000000
> 3fffff60:  03610f50 3fff04f8 402100bc 3fff04f8
> 3fffff70:  3ffeedf8 00000000 3ffeef30 40204871
> 3fffff80:  3ffeedf8 00000000 3ffeef30 4020177c
> 3fffff90:  3fffdad0 00000000 3fff04c8 40202818
> 3fffffa0:  3fffdad0 00000000 3fff04c8 4021016c
> 3fffffb0:  feefeffe feefeffe 3ffe853c 40101125
> <<<stack<<<

closed 2019-10-21 by devyte

OmarAlkassab, 2019-10-21:
@devyte any suggestion?

jandrassy, 2019-10-21:
> 
> 
> @devyte any suggestion?

fill the issue template

OmarAlkassab, 2019-10-21:
is that enough?
@devyte 
@jandrassy

Pablo2048, 2019-10-21:
You are kidding right? You have checked ```If there is a stack dump, I have decoded it.``` but decoded stack is missing, MCVE wont compile, so guess yourself - is that enough?

OmarAlkassab, 2019-10-21:
Sorry I have missed to copy and paste it! because I return back to core 2.4.2
I will update the core then reproduce the issue, then edit it.

OmarAlkassab, 2019-10-21:
I debugged my code, and I discovered that the problem in use of PROGMEM

OmarAlkassab, 2019-10-21:
I solved the issue by using the `String s = FPSTR(HOME_page);`  instead of `String s = HOME_page;`

Pablo2048, 2019-10-21:
Is there any reason why you can not use ```send_P()``` method? Copying data rom progmem and then sending the same data to the client doesn't look very smart to me...

OmarAlkassab, 2019-10-21:
@Pablo2048, You are right, I edit my code and use send_P() instead.
Many thanks.