esp8266/Arduino #6657: Cannot connect to 802.11n access point

MaxMaeder, 2019-10-21:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [c] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware: ESP-12
- Core Version: latest git as of 10/20/2019
- Development Env: Arduino IDE
- Operating System: MacOS

### Settings in IDE

- Module: Generic ESP8266 Module
- Flash Mode:	qio
- Flash Size: 4MB/1MB
- lwip Variant: v1.4
- Reset Method: ck
- Flash Frequency: 40Mhz
- CPU Frequency:	80Mhz
- Upload Using: SERIAL
- Upload Speed: 115200

### Problem Description

I’m unable to connect to a 802.11n access point using an example sketch. It looks like other people have had problems similar to this (#2795), but supposedly these issues were fixed with #6484. 

I have tried both the latest git and release 2.5.2. My router is a Belkin AC1200 using factory settings. The access point is 2.4Ghz.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

/*
    This sketch establishes a TCP connection to a "quote of the day" service.
    It sends a "hello" message, and then prints received data.
*/

#include <ESP8266WiFi.h>

#ifndef STASSID
#define STASSID "test-ssid"
#define STAPSK  "test1234"
#endif

const char* ssid     = STASSID;
const char* password = STAPSK;

const char* host = "djxmmx.net";
const uint16_t port = 17;

void setup() {
  Serial.begin(115200);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  /* Explicitly set the ESP8266 to be a WiFi-client, otherwise, it by default,
     would try to act as both a client and an access-point and could cause
     network-issues with your other WiFi-devices on your WiFi-network. */
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  Serial.print("connecting to ");
  Serial.print(host);
  Serial.print(':');
  Serial.println(port);

  // Use WiFiClient class to create TCP connections
  WiFiClient client;
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    delay(5000);
    return;
  }

  // This will send a string to the server
  Serial.println("sending data to server");
  if (client.connected()) {
    client.println("hello from ESP8266");
  }

  // wait for data to be available
  unsigned long timeout = millis();
  while (client.available() == 0) {
    if (millis() - timeout > 5000) {
      Serial.println(">>> Client Timeout !");
      client.stop();
      delay(60000);
      return;
    }
  }

  // Read all the lines of the reply from server and print them to Serial
  Serial.println("receiving from remote server");
  // not testing 'client.connected()' since we do not need to send data here
  while (client.available()) {
    char ch = static_cast<char>(client.read());
    Serial.print(ch);
  }

  // Close the connection
  Serial.println();
  Serial.println("closing connection");
  client.stop();

  delay(300000); // execute once every 5 minutes, don't flood remote service
}

```

### Debug Messages

```
Connecting to test-ssid
............................................. [goes on forever]
```

devyte, 2019-10-23:
On the router: Did you try disabling wmm? Setting channel width to 20MHz? Changing other settings?

terpenoid, 2019-10-26:
same error. subscribe.

MaxMaeder, 2019-10-26:
I think this connects to issue #1354. Previously, I ran a sketch that used WiFi.setPhyMode() while the esp8266 was in station mode. This seemingly prevented me from joining any access points until I erased the WiFi config data from the esp8266. 

This seems like a bug to me.

devyte, 2019-10-29:
Wifi.setPhyMode() has to be set before starting wifi.
I suggest also setting max power to 80% for n mode, this has helped in some cases.
You're calling WiFi.mode() but you're not checking return value.
You're calling WiFi.begin() but you're not checking return value.
What happens if you add a delay(100) between the two?
How old is your board? How often do you change wifi settings?

MaxMaeder, 2019-11-24:
Sorry for not replying sooner. 

> Wifi.setPhyMode() has to be set before starting wifi.

I no longer believe the issue has to do with the PHY mode, since I wiped the WiFi settings, and the issue still persisted. I ran into this issue on a client's device, and I was not able to reproduce this issue with a newer version of their router that I purchased. I still will be able to test any solutions by pushing a firmware update to their device.

> I suggest also setting max power to 80% for n mode, this has helped in some cases.

I will try this: 
`WiFi.setOutputPower(16.4);`

> You're calling WiFi.mode() but you're not checking return value.
> You're calling WiFi.begin() but you're not checking return value.

What would you recommend I do with the return values? Something like this:
`while(WiFi.mode() != WIFI_STA) {
  WiFi.mode(WIFI_STA);
}`
 and this: 
`while(true) {
  if (WiFi.begin(connectSSID.c_str(), connectPass.c_str())) break;
}`

> What happens if you add a delay(100) between the two?

In my implementation, I have a `delay(1000);` between the two

> How old is your board? How often do you change wifi settings?

My board is about one year old. I have changed the WiFi settings on it about ten times.

My current implementation: https://drive.google.com/file/d/1xvzRWBSG7JIdK0K50d13tYR8YgbKT3FP/view?usp=sharing

Thank you!

TimelessNL, 2020-01-11:
I've been using the latest stable ArduinoCore release (2.6.3) on a NodeMCU and noticed that WMM is not advertised by the ESP anymore. When I both connect a 2.5.1 and 2.6.3 device only the 2.5.1 devices is running with WMM enabled. And both are running the same user code (_Tasmota 8.1.0_)

Therefor this could lead to issues where WMM is enforced, and also limits other 802.11n capabilities.

**ArduinoCore 2.5.1**
```
Station xx:xx:xx:xx:xx:xx (on wlan1)
        inactive time:  320 ms
        rx bytes:       7410
        rx packets:     73
        tx bytes:       3010
        tx packets:     24
        tx retries:     8
        tx failed:      0
        rx drop misc:   0
        signal:         -78 [-80, -83, -86] dBm
        signal avg:     -77 [-79, -84, -85] dBm
        tx bitrate:     6.5 MBit/s MCS 0
        rx bitrate:     6.0 MBit/s
        rx duration:    0 us
        expected throughput:    4.394Mbps
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       short
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        short preamble: yes
        connected time: 30 seconds
```
**ArduinoCore 2.6.3**
```
Station xx:xx:xx:xx:xx:xx (on wlan1)
        inactive time:  120 ms
        rx bytes:       868445
        rx packets:     25370
        tx bytes:       1964542
        tx packets:     3920
        tx retries:     6998
        tx failed:      156
        rx drop misc:   840
        signal:         -86 [-92, -91, -90] dBm
        signal avg:     -85 [-91, -91, -88] dBm
        tx bitrate:     1.0 MBit/s
        rx bitrate:     6.0 MBit/s
        rx duration:    0 us
        last ack signal:7 dBm
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       short
        WMM/WME:        no
        MFP:            no
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        short preamble: yes
        connected time: 5050 seconds
```

devyte, 2020-08-15:
The lack of wmm is an Espressif thing, we have no control over it.
About other connection issues, I did several tests with my TP-Link Archer C7 ac1750 running dd-wrt, and I couldn't reproduce any connection issues for any of b/g/n modes.
There have been some compatibility problems reported on Espressif's bbs forum before. Such compatibility issues are specific to the Espressif's closed source libs, we have no control over it. 
If none of your ESPs can connect, then it's likely a compatibility problem.
If only one board can't connect, bit others are ok, then it's likely a board-specific problem.
The way to figure out whether it really is a compatibility problem in Espressif's libs would be to implement a raw sdk application (no Arduino core involvement), but you would have to do that yourself. 
If the raw app can connect and the core with the same sdk version can't, it would point to a core problem. I can't think of how that could be possible, but it would be a hint. 
If neither can connect, it's a compatibility problem.
If a raw app with a newer sdk (e. g. 3.x) can connect, but a raw app with our core's sdk (2.x) can't, then the problem has already been fixed on Espressif's side, and we would need to migrate to sdk 3.x (not a minor thing).

I'm closing for now. If new info is added here that points to a core issue, this can be reopened.

closed 2020-08-15 by devyte