esp8266/Arduino #6659: missing: code examples to read string inputs via html GET and POST

dsyleixa, 2019-10-22:
missing: code examples to read string inputs via html GET and POST
can you please provide some in your standard repo examples?

closed 2019-10-23 by devyte

dsyleixa, 2019-10-23:
why has this topic been closed? The issue is not resolved AFAIK.

jandrassy, 2019-10-23:
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/SimpleAuthentication/SimpleAuthentication.ino

dsyleixa, 2019-10-23:
SimpleAutentification is not a basic example how to read  a string input from keyboard (until Enter is pressed) from a text input field on a most simple html website by GET or POST.

jandrassy, 2019-10-23:
> SimpleAutentification is not a basic example how to read a string input from keyboard (until Enter is pressed) from a text input field on a most simple html website by GET and POST.
> Using Cookies or password autentification are not subject of my topic.

it is
the login and password are entered in a html form and send with a submit button which can be tied with Enter key. the esp8266 web server processes the data sent by the form into parameters with names based on the names of the inputs on the form. study it

dsyleixa, 2019-10-23:
it is not a SIMPLE example but overburdened by additional features like cookies and apart from this overburdened POST method  it does also not show a GET method.

jandrassy, 2019-10-23:
> it is not a SIMPLE MINIMAL example but overburdened by additional features like cookies and apart from this overburdened POST method it does also not show a SIMPLE GET method with evaluation in the Serial monitor to check it.
> I also didn't say anything about autentification, btw.

write one. contribute

dsyleixa, 2019-10-23:
if I knew one or could do so then I had not requested it.

jandrassy, 2019-10-23:
the ESP8266 library handles GET and POST parameters for you. So the handling on the Arduino side is the same for GET and POST paraeters. On HTML form part the difference is only in the FORM element attribute`method`. it can be POST or GET like `method='POST'` or `method='GET'`. but this is not Arduino knowledge, this is HTML and web development

devyte, 2019-10-23:
@dsyleixa I closed this because:
- you didn't bother to follow the instructions in the issue template
- it is not an issue in the core
- it is a request, and a "how do I" type at that

For general help, please refer to user forums like esp8266.com or stackoverflow. If you figure it out and would like to see such examples in our core, please make a PR with the proposed examples, and we'll review.

me-no-dev, 2019-10-24:
Things went very easy here :) could not say the same for his thread on esp32...