esp8266/Arduino #6661: the WiFiServer Example is unfunctional: https://github.com/esp8266/Arduino/blob/master/doc/esp8266wifi/server-examples.rst

dsyleixa, 2019-10-22:
the WiFiServer Example is unfunctional: https://github.com/esp8266/Arduino/blob/master/doc/esp8266wifi/server-examples.rst

Arduino 1.8.9
nodeMCU core 2.5.2
Windows 7/64 pro
Firefox 69.0.3

(with core 2.2.0 it still works, but longer for cores since 2.4.x - 2.5.x)

after entering the IP ( http://192.168.2.107/ ) there is the
error:

> Fehler: Verbindung unterbrochen
> 
> Die Verbindung zum Server wurde zurückgesetzt, während die Seite geladen wurde.
> 
>     Die Website könnte vorübergehend nicht erreichbar sein, versuchen Sie es bitte später nochmals.
>     Wenn Sie auch keine andere Website aufrufen können, überprüfen Sie bitte die Netzwerk-/Internetverbindung.
>     Wenn Ihr Computer oder Netzwerk von einer Firewall oder einem Proxy geschützt wird, stellen Sie bitte sicher, dass Firefox auf das Internet zugreifen darf.
> 

```

#include <ESP8266WiFi.h>

const char* ssid = "********";
const char* password = "********";

WiFiServer server(80);

void setup()
{
  Serial.begin(115200);
  Serial.println();

  Serial.printf("Connecting to %s ", ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println(" connected");

  server.begin();
  Serial.printf("Web server started, open %s in a web browser\n", WiFi.localIP().toString().c_str());
}

// prepare a web page to be send to a client (web browser)
String prepareHtmlPage()
{
  String htmlPage =
     String("HTTP/1.1 200 OK\r\n") +
            "Content-Type: text/html\r\n" +
            "Connection: close\r\n" +  // the connection will be closed after completion of the response
            "Refresh: 5\r\n" +  // refresh the page automatically every 5 sec
            "\r\n" +
            "<!DOCTYPE HTML>" +
            "<html>" +
            "Analog input:  " + String(analogRead(A0)) +
            "</html>" +
            "\r\n";
  return htmlPage;
}

void loop()
{
  WiFiClient client = server.available();
  // wait for a client (web browser) to connect
  if (client)
  {
    Serial.println("\n[Client connected]");
    while (client.connected())
    {
      // read line by line what the client (web browser) is requesting
      if (client.available())
      {
        String line = client.readStringUntil('\r');
        Serial.print(line);
        // wait for end of client's request, that is marked with an empty line
        if (line.length() == 1 && line[0] == '\n')
        {
          client.println(prepareHtmlPage());
          break;
        }
      }
    }
    delay(1); // give the web browser time to receive the data

    // close the connection:
    client.stop();
    Serial.println("[Client disonnected]");
  }
}

```

Serial-output:

> 
> Connecting to WLAN-3YA7LD .......... connected
> Web server started, open 192.168.2.107 in a web browser
> 
> [Client connected]
> [Client disonnected]
> 
> [Client connected]
> GET / HTTP/1.1
> Host: 192.168.2.107
> User-Agent: Mozilla/5.0 (Windows NT 6.1; Win64; x64; rv:69.0) Gecko/20100101 Firefox/69.0
> Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
> Accept-Language: de,en-US;q=0.7,en;q=0.3
> Accept-Encoding: gzip, deflate
> DNT: 1
> Connection: keep-alive
> Upgrade-Insecure-Requests: 1
> [Client disonnected]
> 
> [Client connected]
> GET /favicon.ico HTTP/1.1
> Host: 192.168.2.107
> User-Agent: Mozilla/5.0 (Windows NT 6.1; Win64; x64; rv:69.0) Gecko/20100101 Firefox/69.0
> Accept: image/webp,*/*
> Accept-Language: de,en-US;q=0.7,en;q=0.3
> Accept-Encoding: gzip, deflate
> DNT: 1
> Connection: keep-alive
> [Client disonnected]
> 
> [Client connected]
> GET / HTTP/1.1
> Host: 192.168.2.107
> User-Agent: Mozilla/5.0 (Windows NT 6.1; Win64; x64; rv:69.0) Gecko/20100101 Firefox/69.0
> Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
> Accept-Language: de,en-US;q=0.7,en;q=0.3
> Accept-Encoding: gzip, deflate
> DNT: 1
> Connection: keep-alive
> Upgrade-Insecure-Requests: 1
> [Client disonnected]
> 
> [Client connected]
> GET /favicon.ico HTTP/1.1
> Host: 192.168.2.107
> User-Agent: Mozilla/5.0 (Windows NT 6.1; Win64; x64; rv:69.0) Gecko/20100101 Firefox/69.0
> Accept: image/webp,*/*
> Accept-Language: de,en-US;q=0.7,en;q=0.3
> Accept-Encoding: gzip, deflate
> DNT: 1
> Connection: keep-alive
> [Client disonnected]
> 
> [Client connected]
> GET / HTTP/1.1
> Host: 192.168.2.107
> User-Agent: Mozilla/5.0 (Windows NT 6.1; Win64; x64; rv:69.0) Gecko/20100101 Firefox/69.0
> Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
> Accept-Language: de,en-US;q=0.7,en;q=0.3
> Accept-Encoding: gzip, deflate
> DNT: 1
> Connection: keep-alive
> Upgrade-Insecure-Requests: 1
> Pragma: no-cache
> Cache-Control: no-cache
> [Client disonnected]
> 
> [Client connected]
> GET /favicon.ico HTTP/1.1
> Host: 192.168.2.107
> User-Agent: Mozilla/5.0 (Windows NT 6.1; Win64; x64; rv:69.0) Gecko/20100101 Firefox/69.0
> Accept: image/webp,*/*
> Accept-Language: de,en-US;q=0.7,en;q=0.3
> Accept-Encoding: gzip, deflate
> DNT: 1
> Connection: keep-alive
> Pragma: no-cache
> Cache-Control: no-cache
> [Client disonnected]
>

dsyleixa, 2019-10-22:
instead, with the old core 2.2.0 the example works, 
the website shows the line

> ### Analog input: 2 

and the Serial out put is:

> 
> Connecting to WLAN-3YA7LD ........ connected
> Web server started, open 192.168.2.107 in a web browser
> 
> [Client connected]
> GET / HTTP/1.1
> Host: 192.168.2.107
> User-Agent: Mozilla/5.0 (Windows NT 6.1; Win64; x64; rv:69.0) Gecko/20100101 Firefox/69.0
> Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
> Accept-Language: de,en-US;q=0.7,en;q=0.3
> Accept-Encoding: gzip, deflate
> DNT: 1
> Connection: keep-alive
> Upgrade-Insecure-Requests: 1
> [Client disonnected]
> 
> [Client connected]
> GET /favicon.ico HTTP/1.1
> Host: 192.168.2.107
> User-Agent: Mozilla/5.0 (Windows NT 6.1; Win64; x64; rv:69.0) Gecko/20100101 Firefox/69.0
> Accept: image/webp,*/*
> Accept-Language: de,en-US;q=0.7,en;q=0.3
> Accept-Encoding: gzip, deflate
> DNT: 1
> Connection: keep-alive
> [Client disonnected]
> 
> [Client connected]
> GET / HTTP/1.1
> Host: 192.168.2.107
> User-Agent: Mozilla/5.0 (Windows NT 6.1; Win64; x64; rv:69.0) Gecko/20100101 Firefox/69.0
> Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
> Accept-Language: de,en-US;q=0.7,en;q=0.3
> Accept-Encoding: gzip, deflate
> DNT: 1
> Connection: keep-alive
> Upgrade-Insecure-Requests: 1
> Cache-Control: max-age=0
> [Client disonnected]
>

devyte, 2019-10-30:
That example is beyond awful and semantically wrong. I suppose it needs updating (I just tested it indeed it doesn't work).

closed 2020-11-10 by d-a-v

dsyleixa, 2020-11-11:
hello,
I see some messed-up code in green in red but where is the cleaned-up example, ready for upload and use?

d-a-v, 2020-11-11:
https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/server-examples.html#put-it-together

PR #7697 (2020-11-10): update documentation example
fixes #6661