esp8266/Arduino #6663: Link problem with 2.50/2.5.1/2.5.2 xtensa-lx106-elf-gcc*: error: esp8266\esp8266.a: No such file or directory

timmost, 2019-10-23:
- [x] This issue complies with the [issue POLICY doc]
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
Issue was closed without solution, 
https://github.com/esp8266/Arduino/issues/5524
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[other - WEMOS d1mini pro]
- Core Version:			[l1e17ddd]
- Development Env:		[Visual studio 16.3.5  /Visual Micro 19.0]
- Operating System:		[Windows 10]

### Settings in IDE

- Module:  			[Wemos D1 mini r2]

### Problem Description

Compilation does not finish, stops with error message below.
Problem does not appear with 2.4.2, but appears consistently with 2.5.0 - 2.5.2. 
Cleaned installation completely including regex, tested with Arduino IDE and Visual Studio.

Latest trial with git based installation without IDE installed along https://arduino-esp8266.readthedocs.io/en/latest/installing.html#using-git-version.

No trace of requested file anywhere in file system esp8266\esp8266.a.
File name length restriction in Win 10 eliminated (was a suspicion of another user)

My guess is that this is a wrong linking instruction, or simply that the file is missing in the libraries.

```
Error linking for board LOLIN(WEMOS) D1 mini Pro
Build failed for project 'Sketch1'
 
xtensa-lx106-elf-gcc*: error: esp8266\esp8266.a: No such file or directory
```
### [Blink sketch ]
```
/*
 Name:		Sketch1.ino
*/
#include <dummy.h>
void setup() {
	// initialize digital pin LED_BUILTIN as an output.
	pinMode(LED_BUILTIN, OUTPUT);
}
void loop() {
	digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
	delay(2000);                       // wait for a second
	digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
	delay(2000);                       // wait for a second
}
```

### Compiler Messages

```
Board Properties
name=LOLIN(WEMOS) D1 mini Pro
build.board=ESP8266_WEMOS_D1MINIPRO
build.variant=d1_mini
upload.tool=esptool
upload.maximum_data_size=81920
upload.wait_for_upload_port=true
upload.erase_cmd=version
serial.disableDTR=true
serial.disableRTS=true
build.mcu=esp8266
[... 558 lines cut ...]
"E:\OneDrive\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar" cru "C:\Users\Timmo Sturm\AppData\Local\Temp\VMBCore\visualmicro\748d1396789af0e5cfa730dcd5f3516c\core.a" "C:\Users\Timmo Sturm\AppData\Local\Temp\VMBCore\visualmicro\748d1396789af0e5cfa730dcd5f3516c\core\spiffs_nucleus.cpp.o"
"E:\OneDrive\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar" cru "C:\Users\Timmo Sturm\AppData\Local\Temp\VMBCore\visualmicro\748d1396789af0e5cfa730dcd5f3516c\core.a" "C:\Users\Timmo Sturm\AppData\Local\Temp\VMBCore\visualmicro\748d1396789af0e5cfa730dcd5f3516c\core\umm_malloc.cpp.o"
"E:\OneDrive\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar" cru "C:\Users\Timmo Sturm\AppData\Local\Temp\VMBCore\visualmicro\748d1396789af0e5cfa730dcd5f3516c\core.a" "C:\Users\Timmo Sturm\AppData\Local\Temp\VMBCore\visualmicro\748d1396789af0e5cfa730dcd5f3516c\core\umm_performance.cpp.o"
 
Building libraries ...

Using library esp8266 version 1.0 in folder "file:///E:/OneDrive/Arduino/hardware/esp8266com/esp8266/libraries/esp8266"
recipe.hooks.sketch.prebuild.pattern
"E:\OneDrive\Arduino\hardware\esp8266com\esp8266/tools/python3/python3" "E:\OneDrive\Arduino\hardware\esp8266com\esp8266/tools/signing.py" --mode header --publickey "E:\OneDrive\Arduino\Projects\Sketch1\Sketch1/public.key" --out "C:\Users\Timmo Sturm\AppData\Local\Temp\VMBuilds\Sketch1\esp8266com_d1_mini_pro\Release/core/Updater_Signing.h"
 
Building project code ...
"E:\OneDrive\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IE:\OneDrive\Arduino\hardware\esp8266com\esp8266/tools/sdk/include" "-IE:\OneDrive\Arduino\hardware\esp8266com\esp8266/tools/sdk/lwip2/include" "-IE:\OneDrive\Arduino\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Timmo Sturm\AppData\Local\Temp\VMBuilds\Sketch1\esp8266com_d1_mini_pro\Release/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -MMD -ffunction-sections -fdata-sections -fno-exceptions -DNONOSDK22y=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10807 -DARDUINO_ESP8266_WEMOS_D1MINIPRO -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINIPRO\" -DFLASHMODE_DIO -DESP8266 -I"E:\OneDrive\Arduino\Projects\Sketch1\Sketch1" -I"E:\OneDrive\Arduino\hardware\esp8266com\esp8266\cores\esp8266" -I"E:\OneDrive\Arduino\hardware\esp8266com\esp8266\variants\d1_mini" -I"E:\OneDrive\Arduino\hardware\esp8266com\esp8266\libraries\esp8266\src" "C:\Users\Timmo Sturm\AppData\Local\Temp\VMBuilds\Sketch1\esp8266com_d1_mini_pro\Release\Sketch1.cpp" -o "C:\Users\Timmo Sturm\AppData\Local\Temp\VMBuilds\Sketch1\esp8266com_d1_mini_pro\Release\Sketch1.cpp.o"
 
Linking it all together ...
# Coping cached core C:\Users\Timmo Sturm\AppData\Local\Temp\VMBCore\visualmicro\748d1396789af0e5cfa730dcd5f3516c\core.a to C:\Users\Timmo Sturm\AppData\Local\Temp\VMBuilds\Sketch1\esp8266com_d1_mini_pro\Release\core.a

recipe.hooks.linking.prelink.1.pattern
"E:\OneDrive\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc" -CC -E -P -DVTABLES_IN_FLASH "E:\OneDrive\Arduino\hardware\esp8266com\esp8266/tools/sdk/ld/eagle.app.v6.common.ld.h" -o "C:\Users\Timmo Sturm\AppData\Local\Temp\VMBuilds\Sketch1\esp8266com_d1_mini_pro\Release/local.eagle.app.v6.common.ld"
"E:\OneDrive\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc" -fno-exceptions -Wl,-Map "-Wl,C:\Users\Timmo Sturm\AppData\Local\Temp\VMBuilds\Sketch1\esp8266com_d1_mini_pro\Release/Sketch1.ino.map" -g -w -Os -nostdlib -Wl,--no-check-sections -u app_entry -u _printf_float -u _scanf_float -Wl,-static "-LE:\OneDrive\Arduino\hardware\esp8266com\esp8266/tools/sdk/lib" "-LE:\OneDrive\Arduino\hardware\esp8266com\esp8266/tools/sdk/lib/NONOSDK22y" "-LE:\OneDrive\Arduino\hardware\esp8266com\esp8266/tools/sdk/ld" "-LE:\OneDrive\Arduino\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/lib" "-Teagle.flash.16m14m.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read -o "C:\Users\Timmo Sturm\AppData\Local\Temp\VMBuilds\Sketch1\esp8266com_d1_mini_pro\Release/Sketch1.ino.elf" -Wl,--start-group "Sketch1.cpp.o" "esp8266\esp8266.a" "C:\Users\Timmo Sturm\AppData\Local\Temp\VMBuilds\Sketch1\esp8266com_d1_mini_pro\Release\core.a" -lhal -lphy -lpp -lnet80211 -llwip2-536-feat -lwpa -lcrypto -lmain -lwps -lbearssl -laxtls -lespnow -lsmartconfig -lairkiss -lwpa2 -lstdc++ -lm -lc -lgcc -Wl,--end-group "-LC:\Users\Timmo Sturm\AppData\Local\Temp\VMBuilds\Sketch1\esp8266com_d1_mini_pro\Release"

Error linking for board LOLIN(WEMOS) D1 mini Pro
Build failed for project 'Sketch1'
 
xtensa-lx106-elf-gcc*: error: esp8266\esp8266.a: No such file or directory
```

d-a-v, 2019-10-23:
It looks that this issues comes from your installation.
Can you try to move/rename your "Arduino15" directory so the IDE create a clean one ?
(have you followed [this proposal](https://github.com/esp8266/Arduino/issues/5524#issuecomment-456014389) ?)
Also try with a recent version of Arduino IDE.

timmost, 2019-10-24:
Did a reinstall of Arduino IDE on the computer, pristine - including regex, no Arduino files anymore on computer, no visual studio etc. 

```
Arduino: 1.8.10 (Windows 10), Board: "LOLIN(WEMOS) D1 mini Pro, 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 16M (14M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Timmo Sturm\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini_pro:xtal=80,vt=flash,exception=disabled,ssl=all,eesz=16M14M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -vid-pid=10C4_EA60 -ide-version=10810 -build-path C:\Users\TIMMOS~1\AppData\Local\Temp\arduino_build_938077 -warnings=none -build-cache C:\Users\TIMMOS~1\AppData\Local\Temp\arduino_cache_79671 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.python.path=C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.python-3.7.2-post1.path=C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1 -verbose C:\Users\Timmo Sturm\Documents\Arduino\sketch_oct24a\sketch_oct24a.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Timmo Sturm\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini_pro:xtal=80,vt=flash,exception=disabled,ssl=all,eesz=16M14M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -vid-pid=10C4_EA60 -ide-version=10810 -build-path C:\Users\TIMMOS~1\AppData\Local\Temp\arduino_build_938077 -warnings=none -build-cache C:\Users\TIMMOS~1\AppData\Local\Temp\arduino_cache_79671 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.python.path=C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.python-3.7.2-post1.path=C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1 -verbose C:\Users\Timmo Sturm\Documents\Arduino\sketch_oct24a\sketch_oct24a.ino
Using board 'd1_mini_pro' from platform in folder: C:\Users\Timmo
Using core 'esp8266' from platform in folder: C:\Users\Timmo
Detecting libraries used...
"C:\\Users\\Timmo Sturm\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\Timmo Sturm\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/include" "-IC:\\Users\\Timmo Sturm\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/lwip2/include" "-IC:\\Users\\Timmo Sturm\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\TIMMOS~1\\AppData\\Local\\Temp\\arduino_build_938077/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10810 -DARDUINO_ESP8266_WEMOS_D1MINIPRO -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINIPRO\"" -DFLASHMODE_DIO -DESP8266 "-IC:\\Users\\Timmo Sturm\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\cores\\esp8266" "-IC:\\Users\\Timmo Sturm\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\variants\\d1_mini" "C:\\Users\\TIMMOS~1\\AppData\\Local\\Temp\\arduino_build_938077\\sketch\\sketch_oct24a.ino.cpp" -o nul
Alternatives for dummy.h: [esp8266@1.0]
ResolveLibrary(dummy.h)
[... cut ...] up "-LC:\\Users\\TIMMOS~1\\AppData\\Local\\Temp\\arduino_build_938077"
xtensa-lx106-elf-gcc: error: C:\Users\TIMMOS~1\AppData\Local\Temp\arduino_build_938077\libraries\esp8266\esp8266.a: No such file or directory

Multiple libraries were found for "dummy.h"
 Used: C:\Users\Timmo
Using library esp8266 at version 1.0 in folder: C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\esp8266
exit status 1
Error compiling for board LOLIN(WEMOS) D1 mini Pro.
```

timmost, 2019-10-24:
reverted back to 2.4.2, compiled without problems.

```
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Timmo Sturm\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini_pro:CpuFrequency=80,VTable=flash,FlashSize=16M15M,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=921600 -vid-pid=10C4_EA60 -ide-version=10810 -build-path C:\Users\TIMMOS~1\AppData\Local\Temp\arduino_build_938077 -warnings=none -build-cache C:\Users\TIMMOS~1\AppData\Local\Temp\arduino_cache_79671 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.xtensa-lx106-elf-gcc-1.20.0-26-gb404fb9-2.path=C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.13 -prefs=runtime.tools.esptool-0.4.13.path=C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.13 -prefs=runtime.tools.mkspiffs.path=C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.2.0 -prefs=runtime.tools.mkspiffs-0.2.0.path=C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.2.0 -verbose C:\Users\Timmo Sturm\Documents\Arduino\sketch_oct24a\sketch_oct24a.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Timmo Sturm\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini_pro:CpuFrequency=80,VTable=flash,FlashSize=16M15M,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=921600 -vid-pid=10C4_EA60 -ide-version=10810 -build-path C:\Users\TIMMOS~1\AppData\Local\Temp\arduino_build_938077 -warnings=none -build-cache C:\Users\TIMMOS~1\AppData\Local\Temp\arduino_cache_79671 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.xtensa-lx106-elf-gcc-1.20.0-26-gb404fb9-2.path=C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.13 -prefs=runtime.tools.esptool-0.4.13.path=C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.13 -prefs=runtime.tools.mkspiffs.path=C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.2.0 -prefs=runtime.tools.mkspiffs-0.2.0.path=C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.2.0 -verbose C:\Users\Timmo Sturm\Documents\Arduino\sketch_oct24a\sketch_oct24a.ino
Using board 'd1_mini_pro' from platform in folder: C:\Users\Timmo
Using core 'esp8266' from platform in folder: C:\Users\Timmo
Detecting libraries used...
"C:\\Users\\Timmo Sturm\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\Timmo Sturm\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2/tools/sdk/include" "-IC:\\Users\\Timmo Sturm\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2/tools/sdk/lwip2/include" "-IC:\\Users\\Timmo Sturm\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\TIMMOS~1\\AppData\\Local\\Temp\\arduino_build_938077/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DARDUINO=10810 -DARDUINO_ESP8266_WEMOS_D1MINIPRO -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINIPRO\"" -DESP8266 "-IC:\\Users\\Timmo Sturm\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2\\cores\\esp8266" "-IC:\\Users\\Timmo Sturm\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2\\variants\\d1_mini" "C:\\Users\\TIMMOS~1\\AppData\\Local\\Temp\\arduino_build_938077\\sketch\\sketch_oct24a.ino.cpp" -o nul
Alternatives for dummy.h: [esp8266@1.0]
ResolveLibrary(dummy.h)
  -> candidates: [esp8266@1.0]
"C:\\Users\\Timmo Sturm\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\Timmo Sturm\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2/tools/sdk/include" "-IC:\\Users\\Tim [... cut ...]
Using library esp8266 at version 1.0 in folder: C:\Users\Timmo Sturm\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\esp8266
"C:\\Users\\Timmo Sturm\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-size" -A "C:\\Users\\TIMMOS~1\\AppData\\Local\\Temp\\arduino_build_938077/sketch_oct24a.ino.elf"
Sketch uses 247952 bytes (23%) of program storage space. Maximum is 1044464 bytes.
Global variables use 28032 bytes (34%) of dynamic memory, leaving 53888 bytes for local variables. Maximum is 81920 bytes.
```
>

timmost, 2019-10-24:
Sorry, pressed wrong button, issue is still open.

d-a-v, 2019-10-24:
Can you please, with 2.5.2, edit the file `libraries/esp8266/library.properties` and remove the line `dot_a_linkage=true` ?

devyte, 2019-10-25:
To me this looks like the windows filename length limit.

earlephilhower, 2019-10-25:
This is obviously a VisualMicro issue, not a core one.  I suggest you file an issue with them.  There's nothing for folks here to do, unfortunately.  You can also try https://esp8266.com or https://gitter.im/esp8266/Arduino to see if someone else is using VM.

Symptoms indicate that VM is not properly parsing the platform.txt recipes, or the library.properties files, hence not building the required .a files to link in.

timmost, 2019-10-25:
Hi there, the last two trials shown above were on a pristine Arduino Installation (did a complete purge including Reegex cleanting o all Arduino and VS).
Installed only Arduino and the repository.

> Did a reinstall of Arduino IDE on the computer, pristine - including regex, no Arduino files anymore on computer, no visual studio etc.
>

earlephilhower, 2019-10-25:
Fair enough.  I looked at the checkboxes and saw VS/VM, but I do see Arduino below.

FWIW we do build using Arduino on Windows as part of CI and are not seeing any similar problems.

It's going out on a limb, but I've seen overzealous antivirus tools end up quarantining .a files.  Might be worth a check, if there are no other leads.

d-a-v, 2019-10-25:
I believe it could be this: [above](https://github.com/esp8266/Arduino/issues/6663#issuecomment-546113957)
because there is no source file in `library/esp8266`

d-a-v, 2019-10-25:
@timmost Do you think you can try the proposed test for your issue ?

timmost, 2019-10-25:
Hi Guys, thx a lot for your efforts, I am traveling , but will test tomorrow morning. I also tested on a different laptop, where things worked. My aim is to compare compiler outputs step by step. 

On the backlashes.. seemed to be all over the place, backwards, forwards, doubles.

So somehow previous structures might influence the install of the new library, even with reg ex cleaning etc.. 
Thx timmo

timmost, 2019-10-26:
> Can you please, with 2.5.2, edit the file `libraries/esp8266/library.properties` and remove the line `dot_a_linkage=true` ?

Hi DAV, seems that this did the trick!! 
Works now with both latest Arduino (1.8.10) and Visual  Studio (16.3.6) Visual Micro (19.0) and the ESP library (2.5.2) on Win 10.
Also works in my Onedrive folder as project directory.

Great work... I just wonder what this setting for the linkage process does.... 

Best Timmo

d-a-v, 2019-10-26:
Thanks! This has been found by comparing your detailed and exhaustive logs above.
A fix will come soon for next release.

timmost, 2019-10-26:
It always amazes me how great open source communities work.. happy to help and contribute!
Best Timmo

timmost, 2019-10-26:
(I leave it to you to close along the fixing process)

PR #6671 (2019-10-26): fake library directory: prevent building an archive
In some cases, "esp8266" library archive is used by arduino builder but is non existent because there is no source file.
Disabling building this archive fixes #6663.
closed 2019-10-26 by earlephilhower