esp8266/Arduino #6665: Docs error

saloid, 2019-10-24:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.

### Problem Description

In [this doc](https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/station-class.html#setautoreconnect) page there is a note: 

> Note: running setAutoReconnect(true) when module is already disconnected will not make it reconnect to the access point. Instead reconnect() should be used.

So I should use `reconnect()` when ESP is disconnected and autoReconnect is false, **BUT** in `reconnect()` description I found another note:

> Notes: 1. Station should be already connected to an access point. If this is not the case, then function will return false not performing any action. 

It’s about useless `reconnect()` method when ESP is disconnected.

saloid, 2019-10-24:
Example:
```C++
WiFi.setAutoReconnect(false);
WiFi.begin(ssid, password);
//here wifi connected
//but then connection lost
WiFi.setAutoReconnect(true); //won`t help because first Note, but suggested next line
WiFi.reconnect(); //but this won`t help too because second Note.
```

saloid, 2019-10-24:
I don't know how it really will work in this case, but in doc it's logical mismatch.

jandrassy, 2019-10-24:
> 
> 
> I don't know how it really will work in this case, but in doc it's logical mismatch.

What WiFi.begin() does, the AT firmware calls "Join AP" . So the esp8266 must first join an AP to be able to reconnect to it in case of disconnection. But is "join" a better word then "connect"?

closed 2019-10-25 by devyte

devyte, 2019-10-25:
AutoReconnect() defines the behavior when a connection is dropped  It's like a flag, and doesn't trigger an action. Think of it like calling the following code when a connection is dropped:
```cpp
if(autoReconnectIsSet) 
  WiFi.reconnect() ;
```
The doc says that calling WiFi.autoReconnect(true) just sets the autoReconnectIsSet flag and doesn't itself call WiFi.reconnect(),so you have to call it yourself. 

Reconnect() is the manual way to trigger an action. It requires that a connection has already been established. In other words, if a connection was established, and then was dropped, you can try to get it back by calling this. It won't work if no prior connection was established. 

I don't see a problem here.