esp8266/Arduino #6675: change static variable time_zone?

osmtools, 2019-10-27:
I'm using https://github.com/esp8266/Arduino/blob/36c369bb8fc8c76945203018e97db70387da6df4/cores/esp8266/time.cpp#L101
with `timezone_sec=0` but wondering about that the time after this call is 8 hours in front.

I'm not a c(++) expert, but could it be, that the following as no affect?
https://github.com/esp8266/Arduino/blob/36c369bb8fc8c76945203018e97db70387da6df4/cores/esp8266/sntp-lwip2.cpp#L440
because of the static declaration of `time_zone` in
https://github.com/esp8266/Arduino/blob/36c369bb8fc8c76945203018e97db70387da6df4/cores/esp8266/sntp-lwip2.cpp#L103

devyte, 2019-10-27:
line 440 is the declaration of the global, line 103 is making use of it. It's correct. BTW, static means that the variable is private to the file and isn't visible/accesible from other files. 
You're supposed to call configTime with the arg timezone_sec as your timezone offset in seconds. E.g. I'm timezone -3, so in my case I'd call with -3*60*60.
The default timezone is Espressif's +8. Calling with 0 makes the timezone 0.
I don't see an issue here, so closing.

closed 2019-10-27 by devyte

osmtools, 2019-10-27:
@devyte but why the print out of 
`  time_t now = time(nullptr);
  Serial.println(ctime(&now));` with `configTime(0, 0, ntpServer);` is then UTC +8?

devyte, 2019-10-27:
You said 8 hours ahead. That's not the same as utc +8, unless your timezone is 0. Which of the two is it? Is your timezone 0?

osmtools, 2019-10-27:
I'm in GMT (=UTC), so i'm using timezone_sec=0, right?
Just right now it's 15:17:00, but the print out is 23:17:00. So 8 hours ahead and i would interpret this as UTC +8. But i expect the print out of 15:17:00. If i change the static variable to 0 then i get 15:17:00.
It's a little bit confusing to me.

osmtools, 2019-10-27:
And btw. if i'm using timezone_sec=-10800 (UTC -3 hours) the print out is 23:17:00, too...

devyte, 2019-10-27:
Alright, please open a new issue and follow the template instructions, including MCVE, and let's discuss there. Add the above explanations in the description.