esp8266/Arduino #6678: sntp_set_timezone_in_seconds doesn't seem to have any effect

osmtools, 2019-10-28:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12E (LoLin  NodeMCU V3)
- Core Version:			ba50bd57b81cf4106fe44ab831d5a061facd4be9
- Development Env:		Arduino IDE 1.8.10
- Operating System:		Ubuntu LTS 18.04.2

### Settings in IDE

- Module:  			NodeMCU 1.0 (ESP-12E Module)
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

No matter what value I assign to the `timezone_sec` arg in
https://github.com/esp8266/Arduino/blob/36c369bb8fc8c76945203018e97db70387da6df4/cores/esp8266/time.cpp#L101
the output of `time_t now = time(nullptr); Serial.println(time(&now)); Serial.println(ctime(&now));` will be UTC+8.

So if i set `timezone_sec=0` i would expect the output of GMT/UTC but it won't be.

If i set
https://github.com/esp8266/Arduino/blob/36c369bb8fc8c76945203018e97db70387da6df4/cores/esp8266/sntp-lwip2.cpp#L103
to 0 instead, then you will get the time in GMT/UTC.

It seems that the method
https://github.com/esp8266/Arduino/blob/36c369bb8fc8c76945203018e97db70387da6df4/cores/esp8266/sntp-lwip2.cpp#L437
hasn't any effect to the global variable `time_zone`

Also have a look at #6675 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include <time.h>

const char* ssid       = "...";
const char* password   = "...";

const int timezone_sec = 0;
const int daylightOffset_sec = 0;
const char* ntpServer = "pool.ntp.org";

void setup()
{
  Serial.begin(115200);
  
  Serial.printf("Connecting to %s ", ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
  }
  Serial.println(" CONNECTED");
  
  Serial.println("Call configTime...");
  configTime(timezone_sec, daylightOffset_sec, ntpServer);

  Serial.println("Now it's:");
  time_t now = time(nullptr);
  Serial.println(time(&now));
  Serial.println(ctime(&now));

  WiFi.disconnect(true);
  WiFi.mode(WIFI_OFF);
}

void loop()
{
}

```

### Debug Messages

Run at Mon Oct 28 08:03:51 2019 (GMT/UTC)

```
Connecting to xxx ....... CONNECTED
Call configTime...
Now it's:
1572275031
Mon Oct 28 15:03:51 2019
```

devyte, 2019-10-30:
I suspect that your usage is wrong: you're supposed to wait for the time callback to fire, which happens once the time on board has actually been set from the sntp server.
My own run of your MCVE shows that the time certainly has not been set, I get this:
```
Call configTime...
Now it's:
28804
Thu Jan  1 08:00:04 1970
```
If I modify the sketch to set a callback and not disconnect right after, I can see the cb firing something like 5 seconds later.
Still, something is fishy, because the above printout does show +8h despite being passed 0 for timezone, so I think it should be Thu Jan  1 00:00:04 1970.

CC @d-a-v .

osmtools, 2019-10-31:
@devyte if i remove
```cpp
  WiFi.disconnect(true);
  WiFi.mode(WIFI_OFF);
```
in `setup()` and add
```cpp
  delay(1000);
  Serial.println("Now it's:");
  time_t now = time(nullptr);
  Serial.println(time(&now));
  Serial.println(ctime(&now));
```
to `loop()` i get indeed the following debug output:
```
Connecting to FRITZ!Box WLAN 3170 ....... CONNECTED
Call configTime...
Now it's:
1572536653
Thu Oct 31 15:44:13 2019

Now it's:
1572507853
Thu Oct 31 07:44:13 2019

Now it's:
1572507854
Thu Oct 31 07:44:14 2019
```
So the first call in `loop()` is +8hours, the second call in `setup()` is just fine.

d-a-v, 2019-10-31:
That's because two reasons:
- you configure time *after* internals send a first sntp request
  this explains why you get two different results
- you don't let time for SNTP answer to be received
  you can use the callback to know exactly when time has been configured

I updated your MCVE as follow:
```cpp
const int timezone_sec = 0;
const int daylightOffset_sec = 0;
const char* ntpServer = "pool.ntp.org";

void time_is_set ()
{
  Serial.println("Now it's:");
  time_t now = time(nullptr);
  Serial.println(time(&now));
  Serial.println(ctime(&now));

  //WiFi.disconnect(true);
  //WiFi.mode(WIFI_OFF);
}

void setup()
{
  Serial.begin(115200);
  settimeofday_cb(time_is_set);

  Serial.println("Call configTime...");
  configTime(timezone_sec, daylightOffset_sec, ntpServer);
  // move the two above lines after(*) "connected" below and re-run

  Serial.printf("Connecting to %s ", ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println(" CONNECTED");

  // here(*)
}

void loop()
{
}
```

But I think you'd get better results with the [provided example](https://github.com/esp8266/Arduino/blob/b4c28e74d66c5386f0b540379050d62bfbd0d4ca/libraries/esp8266/examples/NTP-TZ-DST/NTP-TZ-DST.ino#L217)

If the example is too complex, or doesn't match the general use-case,
I'd be pleased if you proposed another one.

devyte, 2019-11-05:
@d-a-v my one question here, per my comment anf experiment above, is: assuming
-  a fresh boot
- no sntp response
- setting of timezone to 0 via configTime

then shouldn't ctime(time(nullptr)) print 00:00 instead lf 08:00?

thorsten-l, 2019-12-19:
#include <time.h>
#include <sys/time.h>
...

void setup() {
  // initialize internal timer and timezone to UTC (1970-01-01 00:00:00)
  struct timeval tv;
  tv.tv_usec = 0;
  tv.tv_sec = 0;
  struct timezone tz;
  tz.tz_dsttime = 0;
  tz.tz_minuteswest = 0;
  settimeofday( &tv, &tz );

d-a-v, 2019-12-19:
@thorsten-l 

The second argument [is ignored](https://www.unix.com/man-page/v7/3c/settimeofday/).

Have you seen the [provided example](https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/NTP-TZ-DST/NTP-TZ-DST.ino) accessible from the arduino IDE ?

d-a-v, 2020-01-06:
Is this issue still relevant ?

PR #6993 (2020-02-18): configTime(tzsec,dstsec,): fix UTC/local management
This PR also removes dead code since probably newlib updates.
The NTP-TZ-DST example is updated.

fixes #6921
fixes #6678
commit f066ed24 (2020-02-18): configTime(tzsec,dstsec,): fix UTC/local management (#6993)

* configTime(tzsec,dstsec,): fix UTC/local management
This PR also remove dead code since probably newlib updates
The NTP-TZ-DST example is also updated
* restore sntp_set_timezone_in_seconds()
fixes #6678
* +configTzTime()
closed 2020-02-18 by d-a-v