esp8266/Arduino #6682: Lolin D1 Mini won't connect to 802.11g router

gbhug5a, 2019-10-28:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				Lolin D1 Mini v3.1
- Core Version:			2.5.2 and 2.4.2 and 2.3.0
- Development Env:		Arduino IDE
- Operating System:		Windows 7

### Settings in IDE

- Module:  			Lolin(Wemos) D1 R2 & mini
- Flash Mode:			?
- Flash Size: 		        4MB no SPIFFs
- lwip Variant:                  v2 Lower Memory
- Reset Method:		?
- Flash Frequency:	        ?
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:              921600

### Problem Description

A basic client example sketch flashes fine, but is not able to connect to an 802.11g router.  Other devices connect to the router with no problem.  SSID and password are confirmed correct.  The problem happens with Arduino IDE v1.8.8, and with ESP8266 cores 2.5.2, 2.4.2 and 2.3.0.  This is a genuine, new Lolin D1 Mini v3.1.0 from their store.  When the sketch runs, it never gets out of the connect While loop.

It appears the Mini finds the SSID, but something goes wrong and it disconnects.  See debug output below.  The router's log shows the attempted connections by the Mini's MAC address, but the log entry is a repeating:

2019/10/28 12:18:00 | WIRELESS | wl0: 11g : DeAuthentication (rcvd. station leaving) User - 2c:f4:32:4a:5f:4

It appears the router is kicking the Mini off the network.  Not sure about that.  The router is an old Buffalo WHR-HP-G54 802.11g model.  Security is shown in the router setup as "AES   WPA-PSK".  Other options not selected are "TKIP" and "WEP".  The "SSL Support" board option is set to "All SSL ciphers (most compitible)".

I don't have another Mini or another router to test.

Does "state: 3 -> 0 (19)" actually mean a "pairwise_cypher" problem?  If so, what is going wrong.  Emphasize that many other devices connect successfully to this router.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```#include <ESP8266WiFi.h>

#ifndef STASSID
#define STASSID "DATACENTER"
#define STAPSK  "bullwinkle"
#endif

const char* ssid     = STASSID;
const char* password = STAPSK;

const char* host = "djxmmx.net";
const uint16_t port = 17;

void setup() {
  delay(10000);
  Serial.begin(115200);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  /* Explicitly set the ESP8266 to be a WiFi-client, otherwise, it by default,
     would try to act as both a client and an access-point and could cause
     network-issues with your other WiFi-devices on your WiFi-network. */
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.println(WiFi.status());
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

```

### Debug Messages

```
SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020

Connecting to DATACENTER
scandone
6
6
6
6
6
scandone
state: 0 -> 2 (b0)
6
state: 2 -> 3 (0)
state: 3 -> 0 (19)
wifi evt: 1
STA disconnect: 203
6
6
reconnect
6
6
6
6
6
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (19)
wifi evt: 1
STA disconnect: 203
6
6
reconnect
6
6

```

devyte, 2019-10-28:
You said:
> [x] I have tested that the issue is present in current master branch (aka latest git).

and

> the problem happens ... with ESP8266 cores 2.5.2, 2.4.2 and 2.3.0

so which is it? Please make sure to test with latest git before reporting an issue. Install instructions are in readthedocs.

gbhug5a, 2019-10-28:
Sorry, I thought v2.5.2 was the latest.  I'm not an engineer, and have no experience with Git or Python, and based on the instructions, installing the latest Git version would be well beyond me.  If that means the problem can't be address, I will certainly understand.

gbhug5a, 2019-10-28:
I tried additional changes to the router's setup, and finally found one that works - enabling 802.11b.  It's basically a "g" router, and what I've read about the ESP8266 is that it will handle B, G and N.  But for some reason, for this router the Mini only works on B.  So this solves my problem.  Sorry to have bothered you.

devyte, 2019-10-29:
Not a bother, just trying to clarify. 
Several points here:
- 2.5.2 is the latest release, but there have been a whole lot of issues resolved since then, including one that shows symptoms similar to what you say. That is why I asked about latest git: reporting an issue that is already fixed in current master branch doesn't help much. No worries.
- One case with connection issues was resolved by lowering the Tx power on the ESP. It was for 802.11n, though, not g. 
- Are you sure your ESP's phy is set correctly?

Reference (methods for WiFi):
```cpp
void setOutputPower(float dBm);
bool setPhyMode(WiFiPhyMode_t mode);
WiFiPhyMode_t getPhyMode();

typedef enum WiFiPhyMode
{
    WIFI_PHY_MODE_11B = 1, WIFI_PHY_MODE_11G = 2, WIFI_PHY_MODE_11N = 3
} WiFiPhyMode_t;
```

I'd try something like this:
```cpp
if(!WiFi.setPhyMode(WIFI_PHY_MODE_11G))
  Serial.println("Wifi phy mode change failed");
WiFi.setOutputPower(15.0);

//now start wifi normally
if(!WiFi.mode(WIFI_STA))
  Serial.println("Wifi mode failed");
WiFi.begin(ssid, password);
...
```

devyte, 2019-10-29:
I just read:
> Lolin(Wemos) D1 R2 & mini

I use these boards a lot, have something like 100 in action, and I don't have a problem with any of B/G/N. That means:
1. the problem has been fixed since 2.5.2, so nothing to do
2. a mismatch between your router config and the ESP config, so a user error
3. a compatibility problem between your router and the ESP, in which case we can't help
Closing for now.

closed 2019-10-29 by devyte