esp8266/Arduino #6684: Esp01s connects via config portal but fails after exit

quique123, 2019-10-28:
After configuring the Esp01s in the config portal and successfully connecting to the ssid, upon exit the Esp tries to connect on its own but I get a connection result 0 (idle status) in the serial monitor.

My code is: https://github.com/quique123/Autoconnect Kentaylor

I've got the esp01s mounted on a programmer in my laptop's usb port and the programmer has 3.3v connected to enable and the Gnd connected to gpio0 via a push button.  That's the button I use to toggle the esp01s into the config portal.

jandrassy, 2019-10-29:
an USB to TTL adapter usually doesn't supply enough power on 3.3 V pin for WiFi of the eesp8266

devyte, 2019-10-29:
@quique123 this is not the right place to ask for general help, let alone help specific to your project. This is an issue tracker, meant to track issues in our core.
Please request such help at a community forum like esp8266.com or stackoverflow.
Also, when opening an issue please follow the instructions contained in the template.
Closing due to non-compliance.

closed 2019-10-29 by devyte

quique123, 2019-10-29:
OK I asked both in SO and in the Esp forum as suggested.  I have found quote a few posts about this issue.  It seems to be a significant issue.