esp8266/Arduino #6685: Cannot compile code (error: File Exists)

Buzzardbait, 2019-10-29:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [x ] If there is a stack dump, I have decoded it. N/A
- [x ] I have filled out all fields below.

#### Platform

- Hardware:					[NodeMCU1.0]
- Core Version:			[2.5.2]
- Development Env:		[Arduino IDE|1.8.10]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[dio]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v2]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]?
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200)

### Problem Description

Project Based on Extended WebSocketServerAllFunctionsDemo compiled many times before. Recently updated IDE to 1.8.10 but unsure of this contributed to problem. Get one error: File Exists as in 
C:\Users\Martin\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-ar: unable to rename 'C:\Users\Martin\AppData\Local\Temp\arduino_build_156493\libraries\ESP8266WiFi\ESP8266WiFi.a'; reason: File exists

I'm familiar with .o and .d file extensions but never heard of an .a extension. There is no file of any type with an .a extension in the destination folder. xtensa-lx106-elf-ar exists but it is a .exe file extension. 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch N/A

```cpp N/A

#include <Arduino.h> Yes

void setup() { N/A

}

void loop() { N/A

}

```

### Debug Messages N/A

```
Debug messages go here N/A
```

devyte, 2019-10-29:
I also had a 1.8.10 issue and had to roll back to 1.8.9. It doesn't sound like the same issue though, but I suggest trying 1.8.9.
The .a is a code archive, i.e.: a library. By default the build sort of puts all the project code together and compiles everything. We instead build libraries into archives first, because then the .a files can be cached, which reduces the time for incremental builds, and only then link all the archives together. Look at the library folders, there is a library.properties, and in there you can find `dot_a_linkage=true`.

I think this is an issue with your folders or installation, so I suggest a full cleanup of all temps. If that doesn't work, delete the whole core and reinstall it. If that doesn't work, delete Arduino and reinstall it.
For further assistance in this please refer to esp8266.com or stackoverflow.

closed 2019-10-29 by devyte

Buzzardbait, 2019-10-29:
Well, that was the fastest closure of an outstanding issue I've ever seen. So I guess I'll go to stackoverflow.