esp8266/Arduino #6686: WiFi fails with 2.6.0

AcuarioCat, 2019-10-29:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:				Wemos D1 mini r2 (ESP8266EX)
- Core Version:			29/10/2019 (7a43092)
- Development Env:		Visual Studio (Visual Micro)
- Operating System:		Windows

### Settings in IDE

- Module:  			Wemos D1 mini r2
- Flash Mode:			[qio|dio|other]
- Flash Size: 	                4MB
- lwip Variant:   v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|]
- Upload Using:		|SERIAL]
- Upload Speed:   [115200 (serial upload only)

### Problem Description

WiFi connection fails with waitForConnectResult 0.
I have tested with V2.5.2. and connectuion works perfectly.
Tested with Wemos D1 and NodeMCU 
Changing to V2.6.0 and connection fails 100%, reverting back to V2.5.2 and everything is ok again.

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vffffffff
~ld
Startup
Connection Failed! Rebooting...0

### [MCVE](https://stackoverflow.com/help/mcve) Sketch
#include <ESP8266WiFi.h>

void setup(void) {
	Serial.begin(115200);
	const char* ssid = "SSID";
	const char* password = "PASS";
	const char* hostName = "TestProject";

	Serial.println("Startup");

	WiFi.disconnect();
	WiFi.mode(WIFI_STA);
	WiFi.hostname(hostName);
	WiFi.begin(ssid, password);
	while (WiFi.waitForConnectResult() != WL_CONNECTED) {
		Serial.printf("Connection Failed! Rebooting...%d\n", WiFi.waitForConnectResult());
		delay(5000);
		ESP.restart();
	}

}

void loop(void) {
	char result[20];

	sprintf(result, "%d.%d.%d.%d", WiFi.localIP()[0], WiFi.localIP()[1], WiFi.localIP()[2], WiFi.localIP()[3]);
	Serial.println(result);
	delay(1000);
}

devyte, 2019-10-29:
You're not checking any return values for WiFi.blah() methods. Does mode() succeed? does begin()?
What happens if you add a delay(100) after mode()?
Did you wipe the entire flash?

AcuarioCat, 2019-10-30:
I updated the code to check the result codes - all ok except the connect. I added the delay as well and no difference. I did a complete wipe of the flash using the ESP download tool and writing a blank bin to all 4M.  Here is the output:

V2.5.2
=======
Startup
Disconnect: true
Mode: true
Hostname: true
begin: 6
192.168.0.113
192.168.0.113

V2.6.0
=======
Startup
Disconnect: true
Mode: true
Hostname: true
begin: 0
Connection Failed! Rebooting...0

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

devyte, 2019-10-30:
What delay value did you try? only 100? what happens with a longer delay(500)? 1000? 5000?

jandrassy, 2019-10-30:
try to power cycle the esp8266 after uploading the 2.6.2 version. Not simple reset, but disconnect it from power for some seconds.

AcuarioCat, 2019-10-30:
After trying everything I tracked it down to some code in ESP8266WiFiSTA.cpp
For some reason (has the SDK version changed?) I'm passing in the SSID and Password but the SSID is being removed from conf.ssid and set to nothing in the following code (line 174...):

`   
if(WiFi._persistent){
       wifi_station_get_config_default(&conf_compare);
	}
   else {
       wifi_station_get_config(&conf_compare);
	}
`
After commenting out this code the device connects normally.
I tried resetting the WiFi credentials with WiFi.disconnect(true);  so the credentials were deleted from the device (and they were) but it made no difference.

devyte, 2019-10-30:
What is the wifi router security? wep or wpa2? How long is your ssid and password?
This sounds like setting the credentials is being rejected.

AcuarioCat, 2019-10-30:
Security is WPA2-PSK, password 10 characters, SSID 9 characters.
Works perfectly with 2.5.2 (and all previous versions I've used) and my WiFi settings haven't changed.

AcuarioCat, 2019-10-30:
This makes no sense.. It seems wifi_station_get_config is overwriting the conf structure although it is being passed a different structure.
The code:

		Serial.printf("SSID (conf)1: %s\n", conf.ssid);
		wifi_station_get_config(&conf_compare);
		Serial.printf("SSID (conf)2: %s\n", conf.ssid);

gives me the following:
SSID (conf)1: Terrapico
SSID (conf)2: 
when it shouldn't be any different ad the function is being passed conf_compare.

AcuarioCat, 2019-10-30:
Digging further still, it looks like wifi_station_get_config and wifi_station_get_config_default are setting the first byte of conf.ssid to 0. 
If I add a new struct and set its ssid (so it gets memory allocated and conf and conf_compare are not in a contiguous block) before the conf_compare structure is create, then call wifi_station_get_config, then dump the results from both structures ssid I get the following:

54,65,72,72,61,70,69,63,6f,0,fe,3f,fc,75,20,40,50,ff,ff,3f,20,ff,ff,3f,8,0,0,0,f,0,0,0
54,65,72,72,61,70,69,63,6f,0,fe,3f,fc,75,20,40,50,ff,ff,3f,20,ff,ff,3f,8,0,0,0,f,0,0,0

If I then remove my extra struct and re-run the code (so conf and conf_compare are a contiguous block) then I get:

0,65,72,72,61,70,69,63,6f,0,0,0,ec,2c,20,40,48,6f,73,74,6e,61,6d,65,3a,20,74,72,75,65,a,0
54,65,72,72,61,70,69,63,6f,0,fe,3f,fc,75,20,40,50,ff,ff,3f,20,ff,ff,3f,8,0,0,0,f,0,0,0

As can be seen the first byte of the SSID has been overwritten with a 0 (should be 54).

From here on I can't do anything as it appears to be a bug in the SDK code that I don't have access to.

devyte, 2019-10-30:
I didn't understand that last test. Could you please elaborate?
wifi_station_get_config() copies values from the sdk internals into the conf structure passed as arg. Given that at line 174 you haven't set anything up yet, I would expect conf.ssid's 1st char to be set to 0 upon return.
The actual setup is done further down when calling wifi_station_set_config() with the new conf.

devyte, 2019-10-30:
BTW, what happens if you add a delay(500) right after the topmost if statement (below the !enableSTA, so delay at line 133)?

d-a-v, 2019-10-30:
@AcuarioCat Please enable debug options and debug logging.
Have you tried with arduino IDE ?
Do you restart `tools/get.py` and `git submodule update --init` when switching back and forth the core version ?

AcuarioCat, 2019-10-30:
Here is the code I'm using to test:

	//struct station_config conf_compare1;
	//strcpy(reinterpret_cast<char*>(conf_compare1.ssid), "123");
	struct station_config conf_compare;
	if(WiFi._persistent){
		wifi_station_get_config_default(&conf_compare);
	}
   else {
		Serial.printf("Address conf: %p\n", (void*)conf.ssid);
		Serial.printf("Address conf_compare: %p\n", (void*)conf_compare.ssid);
		for (int i = 0; i < sizeof(conf.ssid); i++) {
			Serial.printf("%0x,", conf.ssid[i]);
		}
		Serial.println("");
		Serial.printf("SSID (conf)1: %s\n", conf.ssid);
		wifi_station_get_config(&conf_compare);
		Serial.printf("SSID (conf)2: %s\n", conf.ssid);
	}

	for (int i = 0; i < sizeof(conf.ssid); i++) {
		Serial.printf("%0x,", conf.ssid[i]);
	}
	Serial.println("");
	for (int i = 0; i < sizeof(conf_compare.ssid); i++) {
		Serial.printf("%0x,", conf_compare.ssid[i]);
	}
	Serial.println("");

Calling begin and passing the ssid (and password) copies the ssid into the conf.ssid member of the conf structure at the start of the function. Without the (commented out) structure the first byte of conf.ssid is being overwritten (or changed) to 0 when either of the set_config calls are called. In the setup I have included WiFi.disconnect(true); so I can test with and without saved credentials.

When changing from one core to the other I delete all files relating to the core I'm not using then do a 'clean build' to delete any intermediate files, then a complete re-build of everything.  I'm running on Windows with Visual Studio and the Visual Micro extension.

AcuarioCat, 2019-10-30:
> BTW, what happens if you add a delay(500) right after the topmost if statement (below the !enableSTA, so delay at line 133)?

Exactly the same

d-a-v, 2019-10-30:
> When changing from one core to the other I delete all files relating to the core I'm not using then do a 'clean build' to delete any intermediate files, then a complete re-build of everything. I'm running on Windows with Visual Studio and the Visual Micro extension.

That way has proven to be wrong and leading to similar issues that you have now.

Can you try a regular proper non-git installation using this source: https://d-a-v.github.io/
(remastered 65 seconds ago)