esp8266/Arduino #6688: Crash and reboot when reaching getString() method

MateuszManaj, 2019-10-29:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[nodeMCU dev board rev3]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[don't know]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Board crashed when code reaching
```cpp
String payload = http.getString();
```
Problem occurs when requesting GET from this host: http://192.168.8.218 - this is my stove webserver. For 1 month it was ok but now i cannot getting response. Other host working properly. I tried to change IP of webserver - nothing changed.

Tell me, what needs to happen for getString() method to crash and reboot board ?
In addtion, response from webserver is correct. All packets and whole response is ok.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <ESP8266HTTPClient.h>

#include <WiFiClient.h>

ESP8266WiFiMulti WiFiMulti;

void setup() {

  Serial.begin(115200);
  // Serial.setDebugOutput(true);

  Serial.println();
  Serial.println();
  Serial.println();

  for (uint8_t t = 4; t > 0; t--) {
    Serial.printf("[SETUP] WAIT %d...\n", t);
    Serial.flush();
    delay(1000);
  }

  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP("ABC", "DEF");

}

void loop() {
  // wait for WiFi connection
  if ((WiFiMulti.run() == WL_CONNECTED)) {

    WiFiClient client;

    HTTPClient http;

    Serial.print("[HTTP] begin...\n");
    if (http.begin(client, "http://192.168.8.218")) {  // HTTP

      Serial.print("[HTTP] GET...\n");
      // start connection and send HTTP header
      int httpCode = http.GET();

      // httpCode will be negative on error
      if (httpCode > 0) {
        // HTTP header has been send and Server response header has been handled
        Serial.printf("[HTTP] GET... code: %d\n", httpCode);

        // file found at server
        if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY) {
          String payload = http.getString();
          Serial.println(payload);
        }
      } else {
        Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
      }

      http.end();
    } else {
      Serial.printf("[HTTP} Unable to connect\n");
    }
  }

  delay(10000);
}

```

### Debug Messages

```
14:02:54.744 -> [SETUP] WAIT 4...
14:02:55.348 -> [SETUP] WAIT 3...
14:02:56.371 -> [SETUP] WAIT 2...
14:02:57.344 -> [SETUP] WAIT 1...
14:02:58.354 -> [HTTP] begin...
14:02:58.354 -> [HTTP] GET...
14:02:58.390 -> [HTTP] GET... code: 200
14:02:58.390 -> 
14:02:58.390 -> Exception (3):
14:02:58.390 -> epc1=0x40206d45 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40005301 depc=0x00000000
14:02:58.428 -> 
14:02:58.428 -> >>>stack>>>
14:02:58.428 -> 
14:02:58.428 -> ctx: cont
14:02:58.428 -> sp: 3ffffb20 end: 3fffffc0 offset: 01a0
14:02:58.428 -> 3ffffcc0:  3fffc278 40101dc8 3fffc200 0000005c  
14:02:58.428 -> 3ffffcd0:  00000046 4021850c 0000006c 40206e3c  
14:02:58.428 -> 3ffffce0:  4000066f 00000014 00000000 4021850c  
14:02:58.428 -> 3ffffcf0:  4000066d 34a9edba 00000ffd 00000000  
14:02:58.428 -> 3ffffd00:  00000041 34a9edba 000007b5 4021325c  
14:02:58.428 -> 3ffffd10:  0000003f 3ffe8a14 3ec1edba 003e7516  
14:02:58.464 -> 3ffffd20:  00418937 00127cc4 ffffffff 00000050  
14:02:58.464 -> 3ffffd30:  fa190000 00000000 00000000 40213d39  
14:02:58.464 -> 3ffffd40:  11f0ede3 00000010 00000860 0000e0b9  
14:02:58.464 -> 3ffffd50:  3ffee8e8 3ffefc24 ffffffff 3ffef40c  
14:02:58.464 -> 3ffffd60:  00000000 00000000 3ffef40c 40213dfd  
14:02:58.464 -> 3ffffd70:  00000010 00000860 3ffef40c 3ffef40c  
14:02:58.464 -> 3ffffd80:  3fff01dc 00000000 3ffef40c 40214726  
14:02:58.464 -> 3ffffd90:  00000000 00000001 00000860 4021483e  
14:02:58.464 -> 3ffffda0:  3ffffe40 0000003f 3ffffe1c 402046c4  
14:02:58.502 -> 3ffffdb0:  00000adc 00000194 00000194 40206bf8  
14:02:58.502 -> 3ffffdc0:  3ffef974 00000000 3ffef69c 40207020  
14:02:58.502 -> 3ffffdd0:  00000000 4bc6a7f0 00000ffe ffffffff  
14:02:58.502 -> 3ffffde0:  3ffefc24 00000001 3ffef40c 4020faa5  
14:02:58.502 -> 3ffffdf0:  3ffef69c 3ffef974 3ffef1b4 402026ac  
14:02:58.502 -> 3ffffe00:  3ffefc24 3fffff68 0000002d 40202726  
14:02:58.502 -> 3ffffe10:  3ffefc24 3fffff68 3fffff68 40204307  
14:02:58.502 -> 3ffffe20:  3ffee3e4 00000008 000007ba 4020434b  
14:02:58.539 -> 3ffffe30:  ff000000 00000000 00000000 ffffffff  
14:02:58.539 -> 3ffffe40:  3ffef2a4 ffffffff 3ffefc24 40207380  
14:02:58.539 -> 3ffffe50:  203a6564 3ffef2a4 3ffefc24 40203528  
14:02:58.539 -> 3ffffe60:  00000000 000005b4 00000000 3ffffee0  
14:02:58.539 -> 3ffffe70:  3ffe86a5 000000c8 3ffffee0 3ffee3e4  
14:02:58.539 -> 3ffffe80:  3ffe8641 00000000 3ffffee0 3ffffee0  
14:02:58.539 -> 3ffffe90:  3ffef2a4 3ffef2a4 3ffffee0 402035b0  
14:02:58.539 -> 3ffffea0:  00000000 3ffef2a4 3ffef2b4 40204704  
14:02:58.539 -> 3ffffeb0:  3ffe8641 00000000 0000001c 3ffee3e4  
14:02:58.577 -> 3ffffec0:  00000000 3ffef2a4 3ffffee0 40203d7e  
14:02:58.577 -> 3ffffed0:  3ffee3e4 00000000 3fffff80 4020117f  
14:02:58.577 -> 3ffffee0:  00000000 00000000 3fffff68 3ffef21c  
14:02:58.577 -> 3ffffef0:  000d000f ffffc6fc 00000050 40001388  
14:02:58.577 -> 3fffff00:  00000000 00000000 0000049c 70747468  
14:02:58.577 -> 3fffff10:  00000000 0020f102 00000000 00000000  
14:02:58.577 -> 3fffff20:  00fe858b 3ffef18c 0011001f ff20f102  
14:02:58.577 -> 3fffff30:  00000000 00000000 00fee388 00000000  
14:02:58.612 -> 3fffff40:  00000000 000000c8 ffffffff 00000000  
14:02:58.612 -> 3fffff50:  3ffe000a 00000000 00000000 00fee44c  
14:02:58.612 -> 3fffff60:  00000000 3ffef2a4 40207740 00000000  
14:02:58.612 -> 3fffff70:  00001388 00000ffe 00000000 3ffef1b4  
14:02:58.612 -> 3fffff80:  00000000 00000000 fffee3e4 402018ac  
14:02:58.612 -> 3fffff90:  3fffdad0 00000000 3ffee3e4 3ffee44c  
14:02:58.612 -> 3fffffa0:  3fffdad0 00000000 3ffee41c 40205614  
14:02:58.612 -> 3fffffb0:  feefeffe feefeffe 3ffe8508 40100459  
14:02:58.648 -> <<<stack<<<
14:02:58.648 -> 
14:02:58.648 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,0)
14:02:58.648 -> 
14:02:58.648 -> load 0x4010f000, len 1384, room 16 
14:02:58.648 -> tail 8
14:02:58.648 -> chksum 0x2d
14:02:58.648 -> csum 0x2d
14:02:58.648 -> v8b899c12
14:02:58.648 -> ~ld
14:02:58.685 -> 
14:02:58.685 -> 
14:02:58.685 -> 
```

devyte, 2019-10-29:
> [] If there is a stack dump, I have decoded it.

Why isn't the stack decoded? Please decode it and add it above.

MateuszManaj, 2019-10-29:
I do not know how :)

devyte, 2019-10-29:
Just google "decode esp8266 stack". Also, readthedocs.

MateuszManaj, 2019-10-29:
Ok, thank you. I edited issue. Is this memory leak ?

devyte, 2019-10-29:
You said:
>  [x] I have tested that the issue is present in current master branch (aka latest git). 

But from your stack dump, you're using core v2.5.2. 

Retesting with latest git is needed before this can be looked at further, because there have been several issues fixed, including interrupt and String issues. In other words, it makes no sense to look at this if the issue is already fixed in the latest code base.

Instructions for installing latest git are in readthedocs.

I am closing for now. If you find more information, please post it here and we can continue to discuss.

MateuszManaj, 2019-10-30:
@devyte ok, thank you for instructions. 
I've updated already ESP8266 lib and there is my another decoded exception stack:

```
Decoding stack results
0x40217314: ip4_output_if_opt_src at core/ipv4/ip4.c line 1764
0x401008e8: malloc(size_t) at /Users/mateuszmanaj/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.cpp line 511
0x40217e10: mem_malloc at core/mem.c line 210
0x40211d54: pbuf_alloc_LWIP2 at core/pbuf.c line 284
0x40212805: tcp_output_alloc_header_common at core/tcp_out.c line 1825
0x402128c9: tcp_output_alloc_header at core/tcp_out.c line 1863
0x402131f2: tcp_send_empty_ack at core/tcp_out.c line 2042
0x4021330a: tcp_output at core/tcp_out.c line 1319
0x40100154: ets_post(uint8, ETSSignal, ETSParam) at /Users/mateuszmanaj/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 156
0x4020e57d: tcp_recved at core/tcp.c line 1003
0x40202ba4: ClientContext::_consume(unsigned int) at /Users/mateuszmanaj/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/include/ClientContext.h line 556
0x40202c1a: WiFiClient::read() at /Users/mateuszmanaj/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/include/ClientContext.h line 239
0x402047fc: Stream::timedRead() at /Users/mateuszmanaj/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Stream.cpp line 34
0x4020484b: Stream::readBytes(char*, unsigned int) at /Users/mateuszmanaj/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Stream.cpp line 215
0x40207590: Stream::readBytes(unsigned char*, unsigned int) at /Users/mateuszmanaj/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Stream.h line 92
0x4020373c: HTTPClient::writeToStreamDataBlock(Stream*, int) at /Users/mateuszmanaj/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 1401
0x4020416f: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at /Users/mateuszmanaj/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 725
0x402037be: HTTPClient::writeToStream(Stream*) at /Users/mateuszmanaj/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 934
0x40204262: HTTPClient::getString() at /Users/mateuszmanaj/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 1019
0x40201180: loop() at /Users/mateuszmanaj/Documents/Arduino/BasicHttpClient/BasicHttpClient.ino line 62
0x40201abb: std::vector   >::_M_emplace_back_aux (WifiAPEntry const&) at /Users/mateuszmanaj/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/xtensa-lx106-elf/include/c++/4.8.2/ext/new_allocator.h line 104
0x40201bc0: ESP8266WiFiMulti::addAP(char const*, char const*) at /Users/mateuszmanaj/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp line 39
0x40205c90: loop_wrapper() at /Users/mateuszmanaj/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 176
```

Can you help me with this ?

devyte, 2019-10-30:
How big is the resource you're trying to GET?

MateuszManaj, 2019-10-30:
It's very small ~25kb
```html

  | <!DOCTYPE HTML PUBLIC "-//W3C//DTD XHTML 1.0 Strict//EN" "http://www.w3.org/TR/xhtml1/DTD/xhtml1-strict.dtd">
-- | --
  | <html>
  | <head>
  | <meta HTTP-EQUIV="content-type" content="text/html; charset=utf-8" />
  | <meta HTTP-EQUIV="PRAGMA" CONTENT="NO-CACHE">
  | <meta name="handheldFriendly" content="true" />
  | <meta name="viewport" content="width=device-width; initial-scale=1"; user-scalable="yes" />
  |  
  | <title>MultiFun Control System</title>
  | <link ID="styl_id" rel="stylesheet" type="text/css" media="screen"/> <!--href="style.css"-->
  | <script language="javascript">
  | var szer = window.screen.width;
  | if(navigator.userAgent.indexOf('Opera Mini') > -1)
  | {
  | if(szer <= 1024){ document.getElementById('styl_id').href = "opmini.css"; }
  | else{ document.getElementById('styl_id').href = "style.css"; }
  | }
  | else
  | {
  | if(szer <= 800) {document.getElementById('styl_id').href = "styleh.css"; }
  | else if(szer <= 1024){document.getElementById('styl_id').href = "style600.css";}
  | else{document.getElementById('styl_id').href = "style.css";}
  | }
  | </script>
  | </head>
  | <body>
  | <div id="glowny">
  | <div id="obrazek">
  | <div id="ziel"><h1 class="green">MultiFun Control System</h1></div>
  | <!--temperatury od róry-->
  | <!--spaliny-->
  | <div id="spaliny"><h1 class="black">32</h1></div>
  | <!--zewnętrzna-->
  | <div id="zewnetrzna"><h1 style="font-size: 42px;" class="blue"></h1></div>
  | <!--pokojowa1-->
  | <div id="pokojowa1"><h1 class="cherry"></h1></div>
  | <!--pokojowa2-->
  | <div id="pokojowa2"><h1 class="cherry"></h1></div>
  | <!--CO1-->
  | <div id="CO1"><h1 class="red">23</h1></div>
  | <!--CO2-->
  | <div id="CO2"><h1 class="red">26</h1></div>
  | <!--Kocioł-->
  | <div id="kociol"><h1 class="red">34</h1></div>
  | <!--CWU-->
  | <div id="CWU"><h1 class="red">37</h1></div>
  | <!--Wyliczone-->
  | <!--Pokój 1-->
  | <div id="pokoj1"><h2></h2></div>
  | <!--Pokój 2-->
  | <div id="pokoj2"><h2></h2></div>
  | <!--CO 1-->
  | <div id="WCO1"><h2>20</h2></div>
  | <!--CO 2-->
  | <div id="WCO2"><h2></h2></div>
  | <!--Kotla-->
  | <div id="WKOC"><h2>0</h2></div>
  | <!--CWU-->
  | <div id="WCWU"><h2>0</h2></div>
  | <form method="get" action="/opts.cgi">
  | <div id="th1"><input type="text" name="TH1" size="4" value="35"></div>
  | <div id="th2"><input type="text" name="TH2" size="4" value=""></div>
  | <div id="th3"><input type="text" name="TH3" size="4" value="45"></div>
  | <div id="op1"><input value="1" name="op" type="checkbox"></div>
  | <div id="op2"><input value="2" name="op" type="checkbox"></div>
  | <div id="op3"><input value="3" name="op" type="checkbox"></div>
  | <div id="op4"><input value="4" name="op" type="checkbox"></div>
  | <div id="op5"><input value="5" name="op" type="checkbox"></div>
  | <div id="pas"><input name="pass" size="4" type="password"></div>
  | <div id="but"><input value="OK" type="submit"></div>
  | </form>
  | </div>
  | </div>
  | </body>
  | </html>

```

MateuszManaj, 2019-10-30:
I must add that this module works perfectly for about 2-3 months and suddenly stopped working (of course on previous git version of board). I tested this code on other same nodeMCU's but nothing change. This is very strange.
The only thing could change is my stove webserver but i do not see any strange notation or differencies. Also head od this document is ok. There is no special signs in code and encoding is UTF-8 wo. BOM.

devyte, 2019-10-30:
25KB is most definitely NOT very small, it's about half the total amount of available heap of an empty sketch, and yours is not an empty sketch.
My guess is you're oom. Please check available heap right before the call to getString().

MateuszManaj, 2019-10-30:
> 25KB is most definitely NOT very small, it's about half the total amount of available heap of an empty sketch, and yours is not an empty sketch.
> My guess is you're oom. Please check available heap right before the call to getString().

Sorry, it's 2.9 - 3.1 kb.
What OOM stands for ?

MateuszManaj, 2019-10-30:
Ok, I have Exception 3 which stands for: LoadStoreErrorCause
Processor internal physical address or data error during load or store
Free heap before getString() is: 50864

devyte, 2019-10-30:
3KB should be ok, then I don't think it's oom (out of memory).
I think the only thing missing are the headers sent with your resource. Can you figure out how to get them and post them here? I think that should be enough info for someone to investigate.

MateuszManaj, 2019-10-30:
@devyte Can you reopen this issue ?

MateuszManaj, 2019-10-30:
@devyte thank you

devyte, 2019-10-30:
Just to be clear, you did install latest git following the instructions in readthedocs, right? And not some copy-over-2.5.2 folders or other install weirdness?

MateuszManaj, 2019-10-30:
right sir. I did this with readthedocs in 100%.

devyte, 2019-10-30:
The HTML you posted isn't valid, there seem to be syntax issues with it. Reported by Chrome and online validation tools.
That being said, it shouldn't crash the ESP.

devyte, 2019-10-31:
Alright, I put the (bad) html in one ESP behind a (async)webserver, and put the sketch in another ESP to test. Building with today's core, it's been running for 30mins with no crash.
This is an excerpt of what I see in the serial monitor.
```
HTTP] begin...
[HTTP] GET...
[HTTP] GET... code: 200
About to getString!!
Done with getString!!
<!DOCTYPE HTML PUBLIC "-//W3C//DTD XHTML 1.0 Strict//EN" "http://www.w3.org/TR/xhtml1/DTD/xhtml1-strict.dtd">
<html>
<head>
... 
</html>

```
I added the About to getString!! and Done with getString!! prints to fence the getString() call.

In other words, **I can't reproduce the crash**.

My guess at this point is that you have either a core installation issue which results in a bad binary or  you have a bad board.

My suggestion is you pursue this at your end, i.e.:
1. do a full cleanout of everything Arduino-related, including IDE, arduino15, temp dirs, whatever
2. do a full reinstall from scratch
3. rebuild cleanly and retest
4. if the above still results in a crash, debug at your end, e.g.: go into getString() add printouts etc and figure out why the crash happens.

Closing for now due to can't reproduce. I'll reopen if more info presents itself.

closed 2019-10-31 by devyte

MateuszManaj, 2019-11-01:
Hi there,
I tested this code on another board and still prints an exception but different:
```
20:08:50.970 -> connected with ABC, channel 5
20:08:50.970 -> dhcp client start...
20:08:51.146 -> ip:192.168.8.115,mask:255.255.255.0,gw:192.168.8.1
20:08:51.803 -> [SETUP] WAIT 3...
20:08:52.806 -> [SETUP] WAIT 2...
20:08:53.803 -> [SETUP] WAIT 1...
20:08:54.828 -> FREE HEAP: 51856
20:08:54.828 -> [HTTP] begin...
20:08:54.828 -> [HTTP] GET...
20:08:54.828 -> FREE HEAP: 50136
20:08:54.828 -> [HTTP] GET... code: 200
20:08:54.865 -> Fatal exception  (LoadStoreErrorCause):
20:08:54.865 -> epc1=0x
20:09:02.394 ->  ets Jan  8 2013,rst cause:4, boot mode:(3,6)
20:09:02.394 -> 
20:09:02.394 -> wdt reset
20:09:02.427 -> load 0x4010f000, len 1384, room 16 
20:09:02.427 -> tail 8
20:09:02.427 -> chksum 0x2d
20:09:02.427 -> csum 0x2d
20:09:02.427 -> v9f2cfb82
20:09:02.427 -> ~ld
```

There is no long tail of errors but still... Can you help me with this a little bit more ?

devyte, 2019-11-02:
Without the ability to reproduce my help is very limited. What I suggest is editing the getString() function by adding Serial.printwhatever, then check in the console which is the last message to print out. That tells you hownthe code execution progresses. If you see that some other function crashes, then you go into that one and do the same. As you go, print out pieces of data (local variables, args, whatever). Eventually you'll reach some code that crashes due to data, e. g. null pointer access or buffer with garbage or missing a null termination, or something that doesn't make sense, like one statement that modifies a completely unrelated piece of data (=> corruption).
It takes time, but you're the only person who can pursue your problem (welcome to open source), at least until you find something that shows a bug in the core code.
Look me up in gitter, I might be able to help a bit better from there.

ingeniero-david, 2019-12-21:
Try this:

void loop() {
  // wait for WiFi connection
  if ((WiFiMulti.run() == WL_CONNECTED)) {

    WiFiClient client;

    HTTPClient http;

    Serial.print("[HTTP] begin...\n");
    if (http.begin(client, "http://192.168.8.218")) {  // HTTP

      Serial.print("[HTTP] GET...\n");
      // start connection and send HTTP header
      int httpCode = http.GET();

      // httpCode will be negative on error
      if (httpCode > 0) {
        // HTTP header has been send and Server response header has been handled
        Serial.printf("[HTTP] GET... code: %d\n", httpCode);

        // file found at server
        if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY) {
          const String&  payload = http.getString();
          Serial.println(payload);
        }
      } else {
        Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
      }

      http.end();
    } else {
      Serial.printf("[HTTP} Unable to connect\n");
    }
  }

  delay(10000);
}

MatejKotas, 2020-12-16:
I ran into the same issue. It turned out to be a problem with a missing `Content-Length` header. When the server doesn't return the header, `http.getString()` crashes. I fixed it by returning the content length header from the server.

MateuszManaj, 2020-12-17:
My server is simple blackbox device. I cannot modify its code. I'm afraid I have to edit main lib and fix this issue.
What do you think?

Mateusz Manaj
Kierownik IT
softgraf.pl

(Krótko bo z iPhone’a)

d-a-v, 2020-12-17:
Can you try latest master, I think #7691 solved this issue.

Flensb, 2021-03-29:
Same here: very repeatable crash on getString(). Under certain very repeatable conditions, I have received "Exception (9) - LoadStoreAlignmentCause: Load or store to an unaligned address". Details are available, if needed by anyone. 
Cure was exactly as described by ArnoldSchw. Thank you for posting.

asinkxcoswt, 2021-04-10:
I also encounter a similar problem and can be fixed by the solution as described by @Flensb and @ArnoldSchw 

In my case, it crashed on a simple GET request to an API which response with a simple short string such as "zone1".

What is weird is that it was working fine at first when I run the API using NodeJs express on my Mac. But It crashed when I deploy the real API in another NodeMCU, even though the request and response are mostly the same.

reivaxy, 2021-06-18:
> I experienced the same behaviour: crash on getString(). For me it worked correctly after rewriting
> HTTPClient http;
> http.begin( url );
> to
> WiFiClient client;
> HTTPClient http;
> http.begin( client, url );
> 
> The top http.begin( url ) statement was deprecated, but it is still in the examples.
> Hope this helps for someone else.

It definitely helped me. I stopped working on my project one year ago out of boredom at being unable to solve this crappy issue.

thebacons, 2024-02-08:
Thank for that reivaxy.  I also got a dump on my Wemos D1 R2 Mini  when using  http.getString(); 

I reqrote from: 
HTTPClient http;
WiFiClient client;
http.begin( client, url );

and change  the sequence of WiFiClient,  as your code above, and it worked ok
WiFiClient client;
HTTPClient http;
http.begin( client, url );