esp8266/Arduino #6691: LittleFS library linked when not used

TD-er, 2019-10-29:
I've been browsing through my builds using the very nice and new Inspect tool from the (not yet released) PIO 4.1.0rc1 which allows me to browse through the binary and see all libs included and its size.  (big, big kudo's for that tool!)

I noticed the LittleFS lib is linked even though I am not using it in my build (as far as I know)
Is this needed to be linked in when only using SPIFFS?
It does occupy about 20k in the build (compiled with debug, so release size my be less)

[image]

TD-er, 2019-10-31:
Is it needed to have LittleFS included like this?
It does seem to be used only here:

https://github.com/esp8266/Arduino/blob/3733ece7e8c4ece1aaa99cef98dc9152bb277539/libraries/ESP8266HTTPUpdateServer/src/ESP8266HTTPUpdateServer-impl.h#L95-L102

Wouldn't it make more sense to have a define stating which filesystem has to be included?
Not including LittleFS would make the 1M builds a lot more useful, since they are now too big to be able to perform a 2-step OTA flash and it is really hard to get them small enough.

earlephilhower, 2019-10-31:
You've a good point.  If you were using SPIFFS, would not want to include LittleFS code.  And if using LittleFS then SPIFFS code (almost as big) is wasted, too.

A `#define NOSPIFFSUPDATE` or `#define NOLITTLEFSUPDATE` before including the HTTPUpdateServer.h file and a `#ifndef NOSPIFFSUPDATE / SPIFFS.end() / #endif` would do it, if slightly hacky

Alternative, probably 3.0 since it changes API, is to require the app to stop the FS itself before beginning an update process.  So *you* would call FS.end() manually before httpupdate.begin().

dirkmueller, 2019-10-31:
I thought the documentation says that anyway already...

TD-er, 2019-10-31:
> I thought the documentation says that anyway already...

Do you have a link to that?

dirkmueller, 2019-10-31:
https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html#end

TD-er, 2019-10-31:
But where does it state that the HTTPUpdateServer should only be called after stopping the filesystem?
Right now that update server does call the end on both filesystem types, which causes both to be included.
I do agree with @earlephilhower that it should not be the responsibility of this update server to end the filesystem processes, but that does mean a change in the interface of this update server.

If it was always meant to be called only with a running filesystem, then it is just a matter of removing these lines from that HTTPUpdateServer code.

d-a-v, 2019-10-31:
Does it make sense, instead of calling `SPIFFS.end()` and `LittleFS.end()`, to call two weak empty functions that do nothing.
These two functions would be redefined each in their respective file (spiffs, littlefs) and this time do the right thing (call `.end()`).
The same way we disable WPS in https://github.com/esp8266/Arduino/blob/master/cores/esp8266/core_esp8266_app_entry_noextra4k.cpp#L13

TD-er, 2019-10-31:
I guess that does make sense, although I do not yet see the implications of these kind of implementations with regard to these `extern` declarations. (in a .cpp file)
I do not fully understand how that code does what it does suggest it is doing, no matter what order the files are being compiled.

earlephilhower, 2019-11-01:
@TD-er , give  #6697 a try.  I've manually tested the WebUpdater example with no SPIFFS/LittleFS.begin() in it and seen no FS code linked, then added SPIFFS/LittleFS.begin, rebuilt, and seen full code linked in as expected.

TD-er, 2019-11-01:
Will try it later this evening.
Not before 22h Dutch time. (GMT + 1)

PR #6699 (2019-11-07): Make SPIFFS and LittleFS stay out of link when not needed
- define two weak functions defaulting to no-op
- redefine them to do something useful when either spiffs or littlefs are used/linked

This comes after a discussion with @earlephilhower 
(who will be too busy for the next days to come to continue on this)

replaces and closes #6697
fixes #6691 

Tested with all 4 uncommented `ets_printf` from this PR,
calling the two weak functions in `setup()`, and calling or not `SPIFFS.begin()` / `LittleFS.begin()`.
closed 2019-11-07 by devyte