esp8266/Arduino #6695: avr/dtostrf() cannot be #included by #include <avr/dtostrf.h>

dsyleixa, 2019-10-31:
avr/dtostrf() cannot be #included by
#include <avr/dtostrf.h>

### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [ x] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12
- Core Version:			2.5.2 
- Development Env:		[Arduino IDE
- Operating System:		[Windows| 7/64 pro

### Settings in IDE

- Module:  			nodemcu 1.0
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[     40Mhz]
- CPU Frequency:		[80Mhz|]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200| 

avr/dtostrf() cannot be #included by
`#include <avr/dtostrf.h>`

this code works with ARM Cortex SAMD21, SAMD51 and M3/Due though.

### Description:
Describe your problem here
avr/dtostrf() cannot be included by
`#include <avr/dtostrf.h>`
this code works with ARM Cortex SAMD21, SAMD51 and M3/Due though.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  delay(1000);
  Serial.println();
  
  float f=1234.567;
  char s[20]="";
  dtostrf(f,8,3,s);
  Serial.println(f);
  Serial.println(s);
  Serial.println();
  
}

void loop() {
  // put your main code here, to run repeatedly:

}

```

### Debug Messages:
```

### Debug Messages

```
Arduino: 1.8.9 (Windows 7), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

D:\arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=80,vt=flash,exception=disabled,ssl=all,eesz=4M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -ide-version=10809 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_732431 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=D:\arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=D:\arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.python.path=D:\arduino\portable\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.python-3.7.2-post1.path=D:\arduino\portable\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=D:\arduino\portable\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=D:\arduino\portable\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -verbose D:\Arduino\portable\sketchbook\dtostrftest000\dtostrftest000.ino
D:\arduino\arduino-builder -compile -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=80,vt=flash,exception=disabled,ssl=all,eesz=4M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -ide-version=10809 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_732431 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=D:\arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=D:\arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.python.path=D:\arduino\portable\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.python-3.7.2-post1.path=D:\arduino\portable\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=D:\arduino\portable\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=D:\arduino\portable\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -verbose D:\Arduino\portable\sketchbook\dtostrftest000\dtostrftest000.ino
Using board 'nodemcuv2' from platform in folder: D:\arduino\portable\packages\esp8266\hardware\esp8266\2.5.2
Using core 'esp8266' from platform in folder: D:\arduino\portable\packages\esp8266\hardware\esp8266\2.5.2
Build-Optionen wurden verändert, alles wird neu kompiliert
Detecting libraries used...
"D:\\arduino\\portable\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-ID:\\arduino\\portable\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/include" "-ID:\\arduino\\portable\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/lwip2/include" "-ID:\\arduino\\portable\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\hw\\AppData\\Local\\Temp\\arduino_build_732431/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10809 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DFLASHMODE_DIO -DESP8266 "-ID:\\arduino\\portable\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\cores\\esp8266" "-ID:\\arduino\\portable\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\variants\\nodemcu" "C:\\Users\\hw\\AppData\\Local\\Temp\\arduino_build_732431\\sketch\\dtostrftest000.ino.cpp" -o nul
dtostrftest000:1:25: error: avr/dtostrf.h: No such file or directory

 #include <avr/dtostrf.h>

                         ^

compilation terminated.

exit status 1
avr/dtostrf.h: No such file or directory
```

dsyleixa, 2019-11-01:
solved:
esp does not need 
#include <avr/dtostrf.h>
at all.

there always was a need to
#include <avr/dtostrf.h>
for all non-AVR-boards (e.g. ARM Cortex M0, M3, M4) so I took that line also for ESP.

for AVR it's also not needed at all.

issue solved.

d-a-v, 2019-11-01:
We already have 'avr/pgmspace.h' to facilitate sketch portability.
We can add an empty 'avr/dtostrf.h' for free.
Just propose your PR mentioning this issue.

this single line would be sufficient:
```
#include <stdlib_noniso.h>
```

dsyleixa, 2019-11-01:
thank you, I do not know how to make a PR, but it's fine so far as it's compatible to Arduino AVR.
automatic compatibility additionally, also to SAM/SAMD, of course would be a perfect feature.

closed 2019-11-02 by d-a-v

dsyleixa, 2019-11-02:
thank you very much!
will that work for my ESP32 too in future?

d-a-v, 2019-11-02:
Not with this commit :-]

PR #6700 (2019-11-02): increase include file compatibility with other architectures
fixes #6695