esp8266/Arduino #6698: Does not show ipv6 in serial

sirwez, 2019-11-01:
I used the code `<https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/IPv6/IPv6.ino>,` but it doesn't show ipv6, only ipv4, I looked in several places for some solutions, but I didn't get any answer for that, only appears in serial:

------------------------------
SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
dns0: 192.168.1.1
Try me at these addresses:
(with 'telnet <addr> or 'nc -u <addr> 23')
IF='st' IPv6=0 local=0 hostname='ipv6test' addr= 192.168.1.6 / mask:255.255.255.0 / gw:192.168.1.1
resolving www.google.com: 172.217.30.4
resolving ipv6.google.com: timeout or not found
------------------------------
SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
dns0: 192.168.1.1
Try me at these addresses:
(with 'telnet <addr> or 'nc -u <addr> 23')
IF='st' IPv6=0 local=0 hostname='ipv6test' addr= 192.168.1.6 / mask:255.255.255.0 / gw:192.168.1.1
resolving www.google.com: 172.217.30.4
resolving ipv6.google.com: timeout or not found
------------------------------
SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
dns0: 192.168.1.1
Try me at these addresses:
(with 'telnet <addr> or 'nc -u <addr> 23')
IF='st' IPv6=0 local=0 hostname='ipv6test' addr= 192.168.1.6 / mask:255.255.255.0 / gw:192.168.1.1
resolving www.google.com: 172.217.30.4
resolving ipv6.google.com: timeout or not found
------------------------------

d-a-v, 2019-11-01:
lwIP variant: select IPv6 in arduino IDE menu

sirwez, 2019-11-01:
how to do this?

sirwez, 2019-11-01:
IDE Arduino
Tools>lwIP Variante> choose  "v2 ipv6 Lower Memory"

Thanks for Answer

closed 2019-11-01 by sirwez