esp8266/Arduino #6704: HTTPClient doesn't request AAAA lookup for IPv6

brianjmurrell, 2019-11-02:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
 - you need to include instructions in this template on how to test with master git
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12E
- Core Version:			2.5.2
- Development Env:		Arduino IDE
- Operating System:		Fedora 31

### Settings in IDE

- Module:  			NodeMCU-1.0 (ESP12-E Module)
- Flash Mode:			?
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		?
- Flash Frequency:	?
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

Trying to use HTTPClient.begin() does not request AAAA records when looking up a name and as such doesn''t work with a host that has only an IPv6 name in DNS.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

WiFiClient client;

void setup() {
  Serial.begin(115200);
  delay(500);

  const char* ssid = "[redacted]";
  const char* password = "[redacted";
  const char* hostname = "[redacted]";
  
  WiFi.hostname(hostname);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
  
}

void loop() {
  HTTPClient http;

  http.begin("http://ip-v6-only-name.example.com:8081/");
  int httpCode = http.GET();
  delay(1000);
}
```

### Debug Messages

Using `tcpdump` we can see that only an IPv4 `A` record is requested:

```
11:57:54.826651 IP 10.75.22.187.61367 > 10.75.22.247.53: 105+ A? ip-v6-only-name.example.com. (40)
11:57:54.827474 IP 10.75.22.247.53 > 10.75.22.187.61367: 105* 0/1/0 (94)
11:57:55.832125 IP 10.75.22.187.61367 > 10.75.22.247.53: 106+ A? ip-v6-only-name.example.com. (40)
11:57:55.832841 IP 10.75.22.247.53 > 10.75.22.187.61367: 106* 0/1/0 (94)
11:57:56.840136 IP 10.75.22.187.61367 > 10.75.22.247.53: 107+ A? ip-v6-only-name.example.com. (40)
11:57:56.840694 IP 10.75.22.247.53 > 10.75.22.187.61367: 107* 0/1/0 (94)
11:57:57.844558 IP 10.75.22.187.61367 > 10.75.22.247.53: 108+ A? ip-v6-only-name.example.com. (40)
11:57:57.845158 IP 10.75.22.247.53 > 10.75.22.187.61367: 108* 0/1/0 (94)
11:57:58.849006 IP 10.75.22.187.61367 > 10.75.22.247.53: 109+ A? ip-v6-only-name.example.com. (40)
11:57:58.849828 IP 10.75.22.247.53 > 10.75.22.187.61367: 109* 0/1/0 (94)
11:57:59.853042 IP 10.75.22.187.61367 > 10.75.22.247.53: 110+ A? ip-v6-only-name.example.com. (40)
11:57:59.853843 IP 10.75.22.247.53 > 10.75.22.187.61367: 110* 0/1/0 (94)
11:58:00.860603 IP 10.75.22.187.61367 > 10.75.22.247.53: 111+ A? ip-v6-only-name.example.com. (40)
11:58:00.861315 IP 10.75.22.247.53 > 10.75.22.187.61367: 111* 0/1/0 (94)
11:58:01.867379 IP 10.75.22.187.61367 > 10.75.22.247.53: 112+ A? ip-v6-only-name.example.com. (40)
11:58:01.868181 IP 10.75.22.247.53 > 10.75.22.187.61367: 112* 0/1/0 (94)
11:58:02.874121 IP 10.75.22.187.61367 > 10.75.22.247.53: 113+ A? ip-v6-only-name.example.com. (40)
11:58:02.874827 IP 10.75.22.247.53 > 10.75.22.187.61367: 113* 0/1/0 (94)
11:58:03.880494 IP 10.75.22.187.61367 > 10.75.22.247.53: 114+ A? ip-v6-only-name.example.com. (40)
```

d-a-v, 2019-11-02:
```
lwip Variant: v2 Lower Memory
```

If that's true, there is no IPv6 capability in your sketch. Please select IPv6 in lwIP variants.
Also when IPv6 is enabled and you want to use it specifically, you need to [wait for it properly](https://github.com/esp8266/Arduino/blob/bf4a4351b6f68c9c0fbbfbf66c71ed318806c0c0/libraries/ESP8266WiFi/examples/IPv6/IPv6.ino#L121).

devyte, 2019-11-03:
Closing due to user error.

closed 2019-11-03 by devyte