esp8266/Arduino #6705: esp8266 softap waiting all the time

lou-lan, 2019-11-02:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[latest]
- Development Env:		[Arduino IDEr]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Iphone connection softap has been waiting for a long time, arduino firmware doesn't work well. Nodemcu firmware can work.
[image]

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <Arduino.h>
#include <ESP8266WiFi.h>

IPAddress local_IP(192,168,4,1);
IPAddress gateway(192,168,4,10);
IPAddress subnet(255,255,255,0);

void setup() {
    Serial.begin(115200);
    Serial.println();

    for(uint8_t t = 3; t > 0; t--) {
        Serial.printf("[SETUP] BOOT WAIT %d...\n", t);
        Serial.flush();
        delay(1000);
    }

    Serial.print("Setting soft-AP configuration ... ");
    Serial.println(WiFi.softAPConfig(local_IP, gateway, subnet) ? "Ready" : "Failed!");

    Serial.print("Setting soft-AP ... ");
    Serial.println(WiFi.softAP("A1234", "12345678") ? "Ready" : "Failed!");

    Serial.print("Soft-AP IP address = ");
    Serial.println(WiFi.softAPIP());
}

void loop() {
}
```

### Debug Messages

```
Debug messages go here
```

lou-lan, 2019-11-04:
[image]

arihantdaga, 2019-11-18:
I too faced this issue sometimes with ios.

d-a-v, 2019-11-18:
error -13 is lwIP-v2's `connection aborted`.
We'd need to reproduce this error.

@arihantdaga sometimes it works ?

@lou-lan would you be able to provide a tcpdump or wireshark capture of the traffic during this error ?

arihantdaga, 2019-11-19:
@d-a-v It works most of the time. Few of my clients said that it did not work in their iphones and they sent similar screenshots and while talking to them they explained the same. Currently i don't have tcpdump, but i'll try to reproduce it in my place and gather logs this time.

lou-lan, 2019-11-19:
@d-a-v  I don't have tcpdump for ios system, but I can provide operating video and code.

d-a-v, 2019-11-19:
Best is to run tcpdump on your router if you can,
or use the integrated packet capture/dump facility directly in the esp8266
(#6518 not merged yet, or the old [simple NetDump](https://github.com/d-a-v/EspGoodies)

tablatronix, 2019-12-04:
What version of IOS ?
17C5046a might have fixes for this

arihantdaga, 2019-12-19:
In my case, the problem seems to be related with adc reads. If we read adc in every loop, then its affecting the wifi performance.  Related to https://arduino.stackexchange.com/questions/19787/esp8266-analog-read-interferes-with-wifi