esp8266/Arduino #6706: IRAM attribute thrown away if function gets inlined

Makuna, 2019-11-02:
### Platform
- Hardware:			ESP-12
- Core Version:			2.5.2
- Development Env:		Arduino IDE
- Operating System:		Windows

### Problem Description
Any method that is marked with IRAM attribute and the compiler decides to inline it into code that is not IRAM will loose the IRAM attribute.

This was a nasty issue to diagnose as the mere presence of code to check the address for conformity will force the method to not be inline and thus it will look to be as though it was correctly attributed.  But instead of checking the addr in code, if you disassemble the code you will see the method call is not present as it got inlined and thus it can't be in placed in IRAM.

I would suggest the attribute IRAM also include attributes to stop the compiler from inlining the method. 

BUT, are there places today that the call site is already in IRAM and the inline makes sense and this would cause worse code to be generated in places?  Can the compiler be smart enough for this situation?

The work around is for the author to mark all methods that are never expected to inline but also IRAM, with the `__attribute__((noinline))`.
The other work around is to explicitly check the address like below which will also force the compiler to not inline the call.
```
        uint32_t addr = reinterpret_cast<uint32_t>(methodInIram);
        if (addr >= 0x40200000)
        {
            ::printf((PGM_P)F("methodInIram is not in IRAM!\r\n"));
            abort();
        }
```

d-a-v, 2019-11-02:
Do you suggest to try with updating the `IRAM_ATTR` / `ICACHE_RAM_ATTR` macro with `__attribute__((noinline))` ?

Makuna, 2019-11-02:
Did you read above ;-)
I stated two workarounds, the first one was marking with the attribute and yes it does work.

Makuna, 2019-11-02:
Asking alot for this to happen; but the BEST solution is...
The compiler should honor IRAM attribute and not inline it if the calling location is in flash and inline it if its already in a method that is in IRAM.

d-a-v, 2019-11-02:
What you're asking to the compiler is to prevent inlining when mapping section is different.
After checking I don't see any gcc option about this... Maybe @earlephilhower can comment.

Did you try to redefine [`ICACHE_RAM_ATTR`](https://github.com/esp8266/Arduino/blob/master/tools/sdk/include/c_types.h#L90) ?

Makuna, 2019-11-02:
not yet

earlephilhower, 2019-11-03:
Making all IRAM function non-inline is a non-starter for me.  When a function is inlined inside another, it takes the segment of the upper-level function.  There's no other sane thing for GCC to do, and that's what it does (since there is no separate function w/inlined function, any function section on that bit disappears).

Your upper level function, if called from IRQ and needs its subfcn as in IRAM, should also be in IRAM.

If your upper level function is only run in userland, but wants a specific small, inlineable function that you're putting in IRAM for speed purposes, then you need to define that specific function with whatever you need to do.

That's a very specific case and not one to push out to every IRAM function out there.  In your own code, sure, set that tiny function with a special `__attribute`.

Makuna, 2019-11-04:
@earlephilhower I don't disagree, and we have discussed elsewhere the new PRECACHE stuff as an appropriate solution instead.

BUT, I hate that everyone considers IRAM in the case of ISRs but it has other uses.  I really wish another attribute was exposed that was ISR specific rather than use the ICACHE_RAM_ATTR which is rather generic term.  This is all a different topic so I will shut up now on it ;-)

devyte, 2019-11-04:
Closing due to previous comments.

closed 2019-11-04 by devyte