esp8266/Arduino #6710: new release needed on esp8266wifi to improve behaviour on multiAP enviroment

teldrive, 2019-11-04:
we have detected that esp8266 arduino v2.52 can't connect to closer AP in a multi Ap enviroment(with same SSID), this issue sometimes causes lack of quality link
3 network(s) found:
1: floured, Ch:6 (-54dBm) 
2: prueba, Ch:6 (-54dBm) 
3: prueba, Ch:11 (-70dBm)
we contacted to espressif support and they have upgraded their sdk, so we need some with the deep acknoledge of theese libraries upgrade it with the new features introduced by espressif

devyte, 2019-11-04:
What are the commits of those upgrades in their repo? Were the changes backported to their 2.2.x branch, which is what we use?

devyte, 2019-11-04:
Also, can you please elaborate on the issue? I have a multiap environment with same ssid, so I would think I would have seen this.
Did you test with our latest git? There are known issues with 2.5.2 which can show as connection issues. 

It would be best if you edit your original post and add the issue template with the info fields filled out.

TD-er, 2019-11-04:
I would also like to know a bit more on this issue.
Were you only using the SSID to connect, or does it also happen in the scenario where you perform a scan and then pick the one with the best RSSI and use the BSSID/channel taken from that scan?

So is it a matter of "cannot connect" or "does not choose the strongest AP" ?

d-a-v, 2019-11-04:
Currently in master we have as default: nonos-sdk-v2.2.x from 19-07-03
And we have optionally (when generic board and proper fw are selected) as testing: nonos-sdk-v2.2.x from 19-10-24
Check #6672 

@teldrive 
Can you verify with the available testing firmware which is currently the latest of their 2.2.x branch ?

teldrive, 2019-11-04:
Hi you can see this issue

We have same ssid “prueba” for all 16 temperature sensors but we had some sensors that don’t connect to any AP and the perform was very erratical

 

Wifi-prueba

Due we have not answer from Espressif  we decided define 3 ssid not yet implemented because we have to dismount them and mount new one’s

Just currently: secadero4,secadero5,secadero6,bodega1 are working with this new idea (different ssid’s) on ssid “prueba2” with AP wifi-prueba2

*         Wifi-prueba

*         Wifi-prueba1

*         Wifi-prueba2

*         Wifi-prueba3

 

You can see that “secadero2”(point 9) has wifi-prueba1 to 3 meters distance instead it connects to 20 meters far wifi-AP called “wifi-prueba”

Enviado el: lunes, 4 de noviembre de 2019 8:48
Para: esp8266/Arduino <Arduino@noreply.github.com>
CC: teldrive <info@teldrive.com>; Author <author@noreply.github.com>

teldrive, 2019-11-04:
Hi you can see this issue

We have same ssid “prueba” for all 16 temperature sensors but we had some sensors that don’t connect to any AP and the perform was very erratical

 

Wifi-prueba

Due we have not answer from Espressif  we decided define 3 ssid not yet implemented because we have to dismount them and mount new one’s

Just currently: secadero4,secadero5,secadero6,bodega1 are working with this new idea (different ssid’s) on ssid “prueba2” with AP wifi-prueba2

*         Wifi-prueba

*         Wifi-prueba1

*         Wifi-prueba2

*         Wifi-prueba3

 

You can see that “secadero2”(point 9) has wifi-prueba1 to 3 meters distance instead it connects to 20 meters far wifi-AP called “wifi-prueba”

Enviado el: lunes, 4 de noviembre de 2019 8:41

teldrive, 2019-11-04:
Hi David

We have used 2.52 wifi esp8266 arduino libraries

We received this message today, below is the issue we have detected

-----Mensaje original-----
De: Espressif Systems [mailto:sales@espressif.com] 
Enviado el: lunes, 4 de noviembre de 2019 7:01
Para: info@teldrive.com <mailto:info@teldrive.com> 
Asunto: RE: latest advances [ ref:_00D7FyCNM._5007Fp3nOc:ref ]

 

Hi,

    Thank you for your information!

    Actually we have done similar optimization in our official SDK. Since ESP8266 Arduino is not maintained by us now, is it possible to post this information on ESP8266 Arduino Github? 

    Thank you for your great effort.

 

Kind regards,

 

_________________________________

 

Wei DAI

 

Technical Support | Espressif Systems

 

www.espressif.com <http://www.espressif.com> 

 

#101, Block 2, 690 Bibo Road, 

 

Zhang Jiang High-Tech Park, Shanghai, China

 

We have same ssid “prueba” for all 16 temperature sensors but we had some sensors that don’t connect to any AP and the perform was very erratical

 

Wifi-prueba

Due we have not answer from Espressif  we decided define 3 ssid not yet implemented because we have to dismount them and mount new one’s

Just currently: secadero4,secadero5,secadero6,bodega1 are working with this new idea (different ssid’s) on ssid “prueba2” with AP wifi-prueba2

*         Wifi-prueba

*         Wifi-prueba1

*         Wifi-prueba2

*         Wifi-prueba3

 

You can see that “secadero2”(point 9) has wifi-prueba1 to 3 meters distance instead it connects to 20 meters far wifi-AP called “wifi-prueba”

Enviado el: lunes, 4 de noviembre de 2019 12:02

CC: teldrive <info@teldrive.com>; Mention <mention@noreply.github.com>

d-a-v, 2019-11-04:
@teldrive 
Current development version use more recent nonos-sdk firmwares than last stable release v2.5.2.
I don't understand the rest of the messages above.
You can
- try the git version of this core https://github.com/esp8266/Arduino#using-git-version
- try a recent arduino installer (unsupported) snapshot https://d-a-v.github.io

In either case, use Arduino IDE, select the generic board, then select the testing firmware (see my previous post, follow the included link).

teldrive, 2019-11-04:
Hi David

Today 

Espressif has released a new firmware due some optimizations related to some issues reported by us

So we think there should be some update in android firmware assuming this new features

 

-----Mensaje original-----
De: Espressif Systems [mailto:sales@espressif.com] 
Enviado el: lunes, 4 de noviembre de 2019 7:01
Para: info@teldrive.com <mailto:info@teldrive.com> 
Asunto: RE: latest advances [ ref:_00D7FyCNM._5007Fp3nOc:ref ]

 

Hi,

    Thank you for your information!

    Actually we have done similar optimization in our official SDK. Since ESP8266 Arduino is not maintained by us now, is it possible to post this information on ESP8266 Arduino Github? 

    Thank you for your great effort.

 

Kind regards,

Enviado el: lunes, 4 de noviembre de 2019 14:26

d-a-v, 2019-11-04:
@teldrive 

They did, but **not today**, it was 11 days ago.
This is verifiable on their repository which I just checked.

We already have this update, but ***not*** in the *latest stable* 2.5.2 version you are using.
Please try our *latest* ***unstable*** version.
I gave you the howto-links in my previous message.

teldrive, 2019-11-04:
Thanks david, we’ll test it

and we’ll fedback you

regards

Enviado el: lunes, 4 de noviembre de 2019 15:53

devyte, 2019-11-06:
@teldrive I asked:
> What are the commits of those upgrades in their repo? Were the changes backported to their 2.2.x branch, which is what we use?

We still need to know this. If the changes implemented by Espressif per your reports are in their 3.x branch, but have not been backported to their 2.2.x branch, then we can't pick them up.

Also, what is the status with you testing latest git?

devyte, 2019-11-09:
Latest SDK was included as optional in release 2.6.0.
If that one doesn't include your needed fixes, please request at the Espressif repo that your changes be backported to their 2.2.x branch, then open a new issue requesting an SDK update.
Closing.

closed 2019-11-09 by devyte