esp8266/Arduino #6711: probeMaxFragmentLength returns false but MFLN is working

edodad, 2019-11-04:
Hi

following this article

https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/bearssl-client-secure-class.html

I tried to save memery with Fragment Length Negotiation (see chapter "MFLN or Maximum Fragment Length Negotiation (Saving RAM)")

unfortunately it seems that our backend (Azure IoT Hub) does not support MFLN

We supposed that because the probeMaxFragmentLength always returns false

BTW, we noticed that we can still call for example

espClient.setBufferSizes(1024, 1024);

even if probeMaxFragmentLength returns false

and it works! it saves a lot of memory

Why probeMaxFragmentLength returns false? is that an issue in the library?

is there any risk in calling setBufferSizes even if the probeMaxFragmentLength returns false?

Thanks

edodad, 2019-11-04:
ESP8266
Arduino Core 2.5.2

Jeroen88, 2019-11-04:
No there is no issue with probeMaxFragmentLength, it works correctly. 

If you can assure that the response you receive from your server is **always less than the buffer size minus some overhead** it will work even without MFLN. This is tricky though, because on the client side you rely on server behaviour that you can not control. If the server goes beyond your buffer size the ESP is likely to crash because it will access memory not available to the client. 

You can always reduce the send buffer size though, I think the minimum is 512 bytes. If the data you send is less than this buffer size, reducing this buffer has little or no effect. If you send more data than the buffer size (also minus overhead) transfer will slow down because it will be send in chunks. In this case the choice is a trade off between memory and speed.

devyte, 2019-11-04:
From the same article of MFLN linked above, section on setBufferSizes():
> Once you have verified (or know beforehand) that MFLN is supported you can use this call...

Your probe says not supported, which means you shouldn't call setBufferSizes. I think doing so to set the receive buffer to less than 16KB will likely result in mem corruption or just plain failure to communicate.

Closing.

closed 2019-11-04 by devyte

edodad, 2019-11-05:
Hi

thanks

BTW, the article also says that

"
In certain applications where the TLS server does not support MFLN (not many do as of this writing as it is relatively new to OpenSSL), but you control both the ESP8266 and the server to which it is communicating, you may still be able to setBufferSizes() smaller if you guarantee no chunk of data will overflow those buffers.
"

Jeroen88, 2019-11-05:
Yes that is more or less the same as I stated. If you are in control of the Azure IoT and if SSL is provided by OpenSSL (which I do not know, but OpenSSL is widely used) then upgrading OpenSSL on the Azure server may well provide the MFLN feature you are looking for.

RaphaelSquair, 2024-10-30:
Hey it's 2024 and I'm having a similar problem but with AWS, their documentation says the tls server is based on openssl and we are using tls1.3 /tls1.2 so we should have mfln supported but we have the same problem. the client says the server dont support mfln, any tip on debugin this with the AWS server?