esp8266/Arduino #6712: Adding SparkFun Blynk to variants?

edspark, 2019-11-04:
Hello!

I'm an Engineer at SparkFun Electronics and I have a question about a potential pull request.  Hosted in this repository is support for SparkFun's ESP8266 Thing and Thing Dev boards which both use the [thing variant](https://github.com/esp8266/Arduino/tree/master/variants/thing) pin mapping. We have a third board, the [SparkFun Blynk Board](https://www.sparkfun.com/blynk) that _also_ uses the thing variant but that we instead host in our Arduino Boards IDE JSON file. If possible I'd like to offer a pull request, so that the board can be offered through the ESP8266 JSON file like the SparkFun Thing and Thing Dev. Let me know,

Thanks!

d-a-v, 2019-11-04:
Be our guest !

edspark, 2019-11-04:
Pull request pending [here](https://github.com/esp8266/Arduino/pull/6713)

edspark, 2019-11-05:
Thanks for you help! I'll close this out since PR was pulled in. Any idea when that will be rolled out in new release, or I guess when your next release will happen?

closed 2019-11-05 by edspark

d-a-v, 2019-11-05:
It was supposed to be yesterday, so I guess you're lucky.

edspark, 2019-11-05:
😤