esp8266/Arduino #6714: Compile error on FS.h: missing time.h import

Niek, 2019-11-05:
See https://github.com/esp8266/Arduino/commit/72dd589599e0b7d2e208eafeeca01a83b2d4cb94#r35810441: that commit breaks compilation for me because a ``<time.h>`` include is missing:
```c++
xtensa-lx106-elf-g++ -o .pio/build/myenv/libf93/ESP8266WiFi/ESP8266WiFiMulti.cpp.o -c -fno-rtti -std=c++11 -g -w -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -fno-exceptions -Wall -DPLATFORMIO=40003 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP01 -DMQTT_MAX_PACKET_SIZE=1024 -DNO_GLOBAL_EEPROM -DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY_LOW_FLASH -DBEARSSL_SSL_BASIC -DUSE_CUSTOM_H -DSECURE_CLIENT=SECURE_CLIENT_BEARSSL -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO=10805 -DARDUINO_BOARD=\"PLATFORMIO_ESP01_1M\" -DFLASHMODE_DOUT -DLWIP_OPEN_SRC -DNONOSDK22x_190703=1 -DTCP_MSS=536 -DLWIP_FEATURES=0 -DLWIP_IPV6=0 -DVTABLES_IN_FLASH -I~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/libraries/ESP8266WiFi/src -I.pio/build/myenv/core -I~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/include -I~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/libc/xtensa-lx106-elf/include -I~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/cores/esp8266 -I~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/lwip2/include -I~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/variants/generic ~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp
In file included from ~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/libraries/ESP8266WiFi/src/CertStoreBearSSL.h:26:0,
                 from ~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.h:30,
                 from ~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/libraries/ESP8266WiFi/src/WiFiClientSecure.h:41,
                 from ~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/libraries/ESP8266WiFi/src/WiFiServerSecure.h:20,
                 from ~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/libraries/ESP8266WiFi/src/ESP8266WiFi.h:41
~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/cores/esp8266/FS.h: In static member function 'static time_t fs::FS::_defaultTimeCB()':
~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/cores/esp8266/FS.h:244:58: error: 'time' was not declared in this scope
     static time_t _defaultTimeCB(void) { return time(NULL); }
                                                          ^
```

Simply adding ``#include <time.h>`` to FS.h doesn't resolve the issue, seemingly because it includes the wrong header file. Adding ``#include "~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/libc/xtensa-lx106-elf/include/time.h"`` or ``#include "../../tools/sdk/libc/xtensa-lx106-elf/include/time.h"`` does work, but that doesn't seem like a good solution. Any ideas?

d-a-v, 2019-11-05:
From what I can see, `time.h` should be included in this directory (copy-paste from your cmdline):
```
-I~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/libc/xtensa-lx106-elf/include
```
I think this is an issue with the character `~` which is not properly expanded to your home directory.
Can you change all occurrences of `~` with `/home/you` in your graphical/file configuration ?

Niek, 2019-11-08:
Found the issue: Time.h from Arduino Time Library conflicts on case-insensitive filesystems. See also:
https://github.com/esp8266/Arduino/issues/2341
https://github.com/PaulStoffregen/Time/issues/74
https://github.com/me-no-dev/ESPAsyncWebServer/issues/60

Closing this issue as it's a problem with the library.

closed 2019-11-08 by Niek

d-a-v, 2019-11-08:
Thanks for the links.
Instead of removing the offending `Time.h`, can you check if you can replace our `#include <time.h>` by `#include <../include/time.h>` and see if it works ?

Niek, 2019-11-08:
@d-a-v good one, that works as well. So adding ``#include <../include/time.h>`` at the top of FS.h fixes the issue too.

d-a-v, 2019-11-08:
Are you willing to make a PR with this ?

devyte, 2019-11-08:
I don't understand the core issue here, or how the suggested fix addresses it.

Niek, 2019-11-08:
> Are you willing to make a PR with this ?
> 
> _edit_
> There are a number of files:
> 
> ```
> cores/esp8266/HardwareSerial.h
> cores/esp8266/libc_replacements.cpp
> cores/esp8266/time.cpp
> libraries/ESP8266WiFi/examples/BearSSL_CertStore/BearSSL_CertStore.ino
> libraries/ESP8266WiFi/examples/BearSSL_Server/BearSSL_Server.ino
> libraries/ESP8266WiFi/examples/BearSSL_ServerClientCert/BearSSL_ServerClientCert.ino
> libraries/ESP8266WiFi/examples/BearSSL_Sessions/BearSSL_Sessions.ino
> libraries/ESP8266WiFi/examples/BearSSL_Validation/BearSSL_Validation.ino
> libraries/ESP8266WiFi/examples/HTTPSRequestCACertAxTLS/HTTPSRequestCACertAxTLS.ino
> libraries/ESP8266httpUpdate/examples/httpUpdateSecure/httpUpdateSecure.ino
> libraries/ESP8266mDNS/examples/LEAmDNS/mDNS_Clock/mDNS_Clock.ino
> libraries/esp8266/examples/NTP-TZ-DST/NTP-TZ-DST.ino
> libraries/LittleFS/examples/LittleFS_Timestamp/LittleFS_Timestamp.ino
> ```

I made one for FS.h only: https://github.com/esp8266/Arduino/pull/6730
You want to change all time.h includes?

Niek, 2019-11-08:
> I don't understand the core issue here, or how the suggested fix addresses it.

Arduino Time Library (a popular library that offers some time-related functions) ships a header file called ``Time.h`` (capital T). When using this library, the issue occurs because on at least macOS and Windows the ``Time.h`` of the library is included instead of the correct ``time.h``. Using the relative path forces the compiler to include the correct ``time.h``.

d-a-v, 2019-11-08:
@devyte 

That comment from @me-no-dev 
https://github.com/me-no-dev/ESPAsyncWebServer/issues/60#issuecomment-330153858
> but since it's too late to change that, we are in a sticky situation now and from now on...

and this one from @PaulStoffregen
> https://github.com/PaulStoffregen/Time/issues/74#issuecomment-300629882

d-a-v, 2019-11-08:
@Niek thanks for the PR.
This workaround may benefit to others, from links you referenced above.

NitrofMtl, 2022-01-23:
> Using the relative path forces the compiler to include the correct time.h

I install the library via the lib manager, so it is on \libdeps. I did not manage to get it work, what would be the path ?

`#include <../libdeps/uno/Time/Time.h>`