esp8266/Arduino #6718: OTA stuck when upload 8% - 44% progress

rashn, 2019-11-06:
#### Platform

- ESP8266MOD (ESP-07)
- Visual Studio Code
- PlatformIO (ArduinoOTA)
- iMac 27, Mac OS Catalina

### Settings in IDE

- Module:  			ESP8266MOD
- Flash Mode:			OTA
- Flash Size: 		1MB
- Upload Using:		OTA

### Problem Description

Each time of upload stuck on 43-44% without any reasons (some times 3-8%, but often 43%).   
In the serial i see error [4] - OTA_END_ERROR

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include <elapsedMillis.h>
#include <SoftwareSerial.h>

// флаг процесса обновления устройства
byte  upDateMe = 0;

int isConnected = 0 ;
const char* ssid = "newrn";
const char* password = "kurban12lam";
elapsedMillis timeElapsed;
byte pina = 0;
void setup() {
  Serial.begin(115200);
  Serial.println("Booting");  //  "Загрузка"
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  pinMode(0, OUTPUT);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    //  "Соединиться не удалось! Перезагрузка..."
    delay(5000);
    ESP.restart();
  }
  delay(3000);
  
  ArduinoOTA.setHostname("Elekrik31-2");

   ArduinoOTA.onStart([]() {
    Serial.println("Start OTA");  //  "Начало OTA-апдейта"
       upDateMe = 1;

  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd OTA");  //  "Завершение OTA-апдейта"
      upDateMe = 0;
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    //  "Ошибка при аутентификации"
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    //  "Ошибка при начале OTA-апдейта"
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    //  "Ошибка при подключении"
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    //  "Ошибка при получении данных"
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
    //  "Ошибка при завершении OTA-апдейта"

      upDateMe = 0;
  });
  ArduinoOTA.begin();
  Serial.println("Ready");  //  "Готово"
  Serial.print("IP address: ");  //  "IP-адрес: "
  Serial.println(WiFi.localIP());
}

void loop() {
  ArduinoOTA.handle();

  if (upDateMe == 0)
  {
    if (timeElapsed > 550)
    {
      timeElapsed = 0;
      pina ^= 1;
      digitalWrite(0, pina * 255);
    // Serial.println("Reba");
    }
  }
}

void onConnected(const WiFiEventStationModeConnected& event) {
  Serial.println ( "Connected to AP." );
  isConnected = 1;

}

void onDisconnect(const WiFiEventStationModeDisconnected& event) {
  Serial.println ( "WiFi On Disconnect." );
  isConnected = 0;

}

void onGotIP(const WiFiEventStationModeGotIP& event) {
  Serial.print ( "Got Ip: " );
  isConnected = 2;

  Serial.println(WiFi.localIP());
}

### PLATFORMIO INI

;PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html
[platformio]
default_envs = esp07

[common_env_data]
build_flags =
    -D VERSION=1.2.3
    -D DEBUG=0

[env:esp07]
platform = espressif8266
board = esp07
framework = arduino
build_flags = -Wl,-Teagle.flash.1m.ld ;eagle.flash.1m128.ld
; OTA
build_flags = -DDEBUG_ESP_OTA

upload_port = 192.168.1.101 ; Elekrik31-2.local
upload_protocol = espota
upload_speed = 921600

upload_flags = --host_ip=192.168.1.179
upload_flags = --port=8266

monitor_port = /dev/cu.wchusbserial1410
monitor_speed = 115200

targets = upload 

### Debug Messages

Ready
IP address: 192.168.1.101
Not connected to Blynk server
[17381] Connecting to 192.168.1.139
[17394] Ready (ping: 9ms).
Connected to Blynk server
Start
Progress: 0%
Progress: 0%
Progress: 0%
Progress: 1%
Error[4]: End Failed

rashn, 2019-11-06:
Compiling .pio/build/esp07/src/elektreckiy.ino.cpp.o
Linking .pio/build/esp07/firmware.elf
Retrieving maximum program size .pio/build/esp07/firmware.elf
Building .pio/build/esp07/firmware.bin
Checking size .pio/build/esp07/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [===       ]  33.9% (used 27744 bytes from 81920 bytes)
PROGRAM: [===       ]  29.3% (used 306332 bytes from 1044464 bytes)
Creating BIN file ".pio/build/esp07/firmware.bin" using ".pio/build/esp07/firmware.elf"
Configuring upload protocol...
AVAILABLE: espota, esptool
CURRENT: upload_protocol = espota
Uploading .pio/build/esp07/firmware.bin
19:55:30 [DEBUG]: Options: {'esp_ip': '192.168.1.101', 'host_port': 41353, 'image': '.pio/build/esp07/firmware.bin', 'host_ip': '0.0.0.0', 'auth': '', 'esp_port': 8266, 'spiffs': False, 'debug': True, 'progress': True}
19:55:30 [INFO]: Starting on 0.0.0.0:41353
19:55:30 [INFO]: Upload size: 310480
19:55:30 [INFO]: Sending invitation to: 192.168.1.101
19:55:30 [INFO]: Waiting for device...
Uploading: [===========================                                 ] 44% 
19:55:30 [ERROR]: Error Uploading
*** [upload] Error 1
============================================================================================= [FAILED] Took 2.29 seconds =============================================================================================
Процесс терминала завершен с кодом выхода: 1

TD-er, 2019-11-06:
I have also seen this happen (also with recent core 2.6.0 builds)
I have not tested it very thoroughly, but I noticed that it has an almost 100% failure rate when the computer used to upload the OTA firmware and the ESP node are on the same AP.
Switch to wired network, or connecting the ESP or computer to another AP and the update is applied just fine.

Problem with this stuck behavior is that you have to reboot the unit to try again. (reboot cannot be done remote :( )

pfeerick, 2019-11-09:
Hm... so does this mean the ESP is crashing? As shouldn't `ArduinoOTA.onError` be catching this, and either allowing the program to continue or for the unit to be automatically restarted? ouch! 😨

devyte, 2019-11-09:
Investigation and PRs are welcome for consideration.

Teddyz, 2019-11-17:
I had similar behaviour when OTA-uploading to Itead/Sonoff switches. This was because of poor quality flash chips. I changed to a better 4MB slash (larger footprint, but works) and the problems went away. 
Your problem might or might not be related to bad flash memory. Mine worked fine for a year before I got problems flashing (and running them reliably even if flashing was successful).

jandrassy, 2019-11-17:
> Investigation and PRs are welcome for consideration.

connected() true before all data read?

bill-orange, 2019-11-23:
I have also experience this problem. My solution has been the same as others, a manual reboot. 

I assumed (with no real evidence) that the problem was poor WiFi connectivity. This belief was reinforced by the observation that the problem has not occurs when the ESP8266 was near the router. The trouble with intermittents, of course is that it is easy to latch on to false leads.

jandrassy, 2019-11-23:
> I have also experience this problem. My solution has been the same as others, a manual reboot.
>
> I assumed (with no real evidence) that the problem was poor WiFi connectivity. This belief was [...]

could you try this change? https://github.com/esp8266/Arduino/pull/6702/commits/e219cb03735d67be26065ab7df62fe964b764b9d#diff-a5fd274181d082a3211a59cb42a7e0d8L334

bill-orange, 2019-11-23:
Thanks.  I will have to wait until the problem occurs again.

devyte, 2019-11-24:
Given that #6827 is merged, is this issue still relevant in latest git?

d-a-v, 2019-12-10:
I think we can close it.
Thanks @jandrassy

closed 2019-12-10 by d-a-v

jandrassy, 2019-12-10:
> I think we can close it.
> Thanks @jandrassy

The connected() function should return true if data are available

ZoserLock, 2020-12-31:
I don't know if related but today my OTA upload was failing 100% of the times between 9% and 24% of the uploading. It was fixed disabling all work in loop function using a boolean and detaching all interruptions during the OTA upload.

```cpp
// setup function
ArduinoOTA.onStart([]() 
{
    _handlingOTA = true;
    _interruptTimer.detachInterrupt();
});
```
    
```cpp
// loop function
void loop()
{
    ArduinoOTA.handle();

    if(!_handlingOTA)
    {
        // rest of loop function
    }
}
```

pfeerick, 2021-01-01:
I can't say I've used interrupts (actually, that may be a lie... see further) with OTA on an ESP8266, and I've not encountered this issue, so perhaps that is the main culprit. I would have thought that once the onStart fires, only that and the other OTA functions run  - or at least this is what I rely on in my programs when I do things like screen updates when in OTA, etc - I've never had to use flags in the main loop to say 'inOtaSoDoNothing'.

On second thought... I do remember having to stop a ticker function from running, or something similar in one project, which I believes does rely on interrupts to function.