esp8266/Arduino #6723: [Feature Req] HardwareSerial: Add support for 9 bit UART (9N1, 9N2, 9P1 etc)

TargoDove, 2019-11-07:
Would it be possible to add support for transmitting 9 bits over UART? I need to use the 9th bit to differentiate between "commands" and "data". It has been done here: https://forum.arduino.cc/index.php?topic=517063.0

Would it be possible to adapt that for the ESP8266 "HardwareSerial" library?

devyte, 2019-11-07:
I am not 100% sure, but I think the hw doesn't support 9bits (did anqyick search and code lookup) 
SoftwareSerial was recently improved to support 9bits.
Closing due to not an issue.

closed 2019-11-07 by devyte

TargoDove, 2019-11-07:
@devyte SoftwareSerial doesn't currently support 9bits either. Where would I post a feature request then?

devyte, 2019-11-07:
Ah, looks like it's still wip, I thought it was done already.
Look at the espsoftwareserial repo, issues/prs etc.