esp8266/Arduino #6727: SPIFFS read slow at some position

c-m-02, 2019-11-08:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12S]
- Core Version:			2.5.2
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[DOUT]
- Flash Size: 		[4MB/3MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Actually, my problem is the same as:
SPIFFS read speed varies depending on location in file #2450
But I am using ESP8266 Arduino-core v2.5.2 that already includes the latest version of SPIFFS library (V0.3.7)

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include "FS.h"

void setup()
{
	Serial.begin(9600);
	SPIFFS.begin();

	Dir dir = SPIFFS.openDir("/");
	String Filename((char *)0);
	FSInfo fs_info;

	Serial.println("");
	while (dir.next())
	{
		Filename = dir.fileName();
		size_t fileSize = dir.fileSize();
		Serial.printf_P(PSTR("File: %s, size: %lu\r\n"), Filename.c_str(), fileSize);
	}

	SPIFFS.info(fs_info);
	Serial.printf_P(PSTR("Block size: %lu, Page size: %lu\r\n"), fs_info.blockSize, fs_info.pageSize);
	Serial.printf_P(PSTR("Total bytes: %lu, Used bytes: %lu\r\n"), fs_info.totalBytes, fs_info.usedBytes);
	Serial.println("");
}

void loop()
{
	File my_file;
	unsigned long start_time;
	unsigned long delta;
	unsigned long byte_count;

	my_file = SPIFFS.open("/file1.bin","r");
	Serial.println(F("Reading file1.bin..."));
	byte_count=0;
	while (my_file.available())
	{
		start_time=millis();
		my_file.read();

		delta = millis()-start_time;
		byte_count++;
		if (delta > 1)
			Serial.printf_P(PSTR("Pos=%lu, time=%d\r\n"), byte_count, delta);
	}
	my_file.close();
	Serial.printf_P(PSTR("Total bytes read=%lu\r\n"), byte_count);

	my_file = SPIFFS.open("/file2.bin","r");
	Serial.println(F("Reading file2.bin..."));
	byte_count=0;
	while (my_file.available())
	{
		start_time=millis();
		my_file.read();

		delta = millis()-start_time;
		byte_count++;
		if (delta > 1)
			Serial.printf_P(PSTR("Pos=%lu, time=%d\r\n"), byte_count, delta);
	}
	my_file.close();
	Serial.printf_P(PSTR("Total bytes read=%lu\r\n"), byte_count);

	my_file = SPIFFS.open("/alm.mp3","r");
	Serial.println(F("Reading alm.mp3..."));
	byte_count=0;
	while (my_file.available())
	{
		start_time=millis();
		my_file.read();

		delta = millis()-start_time;
		byte_count++;
		if (delta > 1)
			Serial.printf_P(PSTR("Pos=%lu, time=%d\r\n"), byte_count, delta);
	}
	my_file.close();
	Serial.printf_P(PSTR("Total bytes read=%lu\r\n"), byte_count);
	Serial.println("");

}

```
### Debug Messages
```
File: /file1.bin, size: 645328
File: /file2.bin, size: 1290656
File: /alm.mp3, size: 161332
Block size: 8192, Page size: 256
Total bytes: 2949250, Used bytes: 2115177

Reading file1.bin...
Pos=1, time=2
Total bytes read=645328
Reading file2.bin...
Pos=26356, time=26
Pos=462092, time=2
Pos=679960, time=3
Pos=711084, time=4
Pos=742208, time=4
Pos=773332, time=3
Pos=804456, time=4
Pos=835580, time=4
Pos=866704, time=4
Pos=897828, time=4
Pos=928952, time=5
Pos=960076, time=5
Pos=991200, time=3
Pos=1022324, time=3
Pos=1053448, time=3
Pos=1084572, time=3
Pos=1115696, time=3
Pos=1146820, time=3
Pos=1177944, time=3
Pos=1209068, time=3
Pos=1240192, time=4
Pos=1271316, time=3
Total bytes read=1290656
Reading alm.mp3...
Total bytes read=161332

```

devyte, 2019-11-09:
Please retry with release 2.6.0.
Please try switching to LittleFS. It was integrated as an alternative to SPIFFS to address precisely this kind of use case.

c-m-02, 2019-11-11:
OK, I will retry with release 2.6.0 and LittleFS.
I will report the result here.
Thanks for the advice.

c-m-02, 2019-11-12:
After switching to 2.6.0 and LittleFS, every byte read < 1ms.
I think this issue can be closed.
Thanks.

closed 2019-11-12 by c-m-02