esp8266/Arduino #6731: Cannot automatically assign a static ip to esp8266

UniTechMaxed, 2019-11-08:
> 
> ### Basic Infos
> 
>     * [x]  This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
> 
>     * [x]  I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
> 
>     * [x]  I have tested that the issue is present in current master branch (aka latest git).
> 
>     * [x]  I have searched the issue tracker for a similar issue.
> 
>     * [x]  If there is a stack dump, I have decoded it.
> 
>     * [x]  I have filled out all fields below.
> 
> 
> #### Platform
> 
>     * Hardware:					[ESP-12S]
> 
>     * Core Version:			2.5.2
> 
>     * Development Env:		[Arduino IDE]
> 
>     * Operating System:		[Windows]
> 
> 
> ### Settings in IDE
> 
>     * Module:  			[Generic ESP8266 Module]
> 
>     * Flash Mode:			[DOUT]
> 
>     * Flash Size: 		[4MB/3MB]
> 
>     * lwip Variant:   [v2 Lower Memory]
> 
>     * Reset Method:		[ck]
> 
>     * Flash Frequency:	[40Mhz]
> 
>     * CPU Frequency:		[160MHz]
> 
>     * Upload Using:		[SERIAL]
> 
>     * Upload Speed:   [115200] (serial upload only)
> 
> 
> ### Problem Description
> 
> I have found so many ways online on how to set a static ip to esp8266 but then it you have to manually assign it yourself. I would like to set a static ip without manually assigning it. How would I be able to achieve this?
> 
> ### Debug Messages
> None

devyte, 2019-11-08:
Off the top of my head, use a static dhcp lease setup in your router/dhcp server.
The only other wsy I know of is via arp, and I don't know if that works on the ESP.
This is a "how do I" type question, and as stated in the very first line of the issue template that you saw when opening, this isn't the right place to ask. This is an issue tracker, meant for tracking issues in the core. 
Closing.

closed 2019-11-08 by devyte

d-a-v, 2019-11-08:
> Off the top of my head, use a static dhcp lease setup in your router/dhcp server.

:+1:

Like any other network device: static IP is setup either on the device (static address on device/esp), or setup in the DHCP server (esp is then dhcp client).