esp8266/Arduino #6737: no matching function for call to 'SoftwareSerial::SoftwareSerial(const uint8_t&, const uint8_t&)' in ver 2.6.0

tihoangyeudau, 2019-11-09:
fatal: not a git repository (or any of the parent directories): .git
GyverMatrixOS_v1.13:478:34: error: no matching function for call to 'SoftwareSerial::SoftwareSerial(const uint8_t&, const uint8_t&)'

 SoftwareSerial mp3Serial(SRX, STX);

                                  ^

C:\Users\tuant\Desktop\GyverMatrixWiFi-master\firmware\GyverMatrixOS_v1.13\GyverMatrixOS_v1.13.ino:478:34: note: candidate is:

In file included from C:\Users\tuant\Desktop\GyverMatrixWiFi-master\firmware\GyverMatrixOS_v1.13\GyverMatrixOS_v1.13.ino:18:0:

C:\Users\tuant\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0\libraries\SoftwareSerial\src/SoftwareSerial.h:49:5: note: SoftwareSerial::SoftwareSerial()

     SoftwareSerial();

     ^

C:\Users\tuant\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0\libraries\SoftwareSerial\src/SoftwareSerial.h:49:5: note:   candidate expects 0 arguments, 2 provided

C:\Users\tuant\Desktop\GyverMatrixWiFi-master\firmware\GyverMatrixOS_v1.13\dfplayer.ino: In function 'void InitializeDfPlayer1()':

dfplayer:2:23: error: no matching function for call to 'SoftwareSerial::begin(int)'

   mp3Serial.begin(9600);  

                       ^

C:\Users\tuant\Desktop\GyverMatrixWiFi-master\firmware\GyverMatrixOS_v1.13\dfplayer.ino:2:23: note: candidate is:

In file included from C:\Users\tuant\Desktop\GyverMatrixWiFi-master\firmware\GyverMatrixOS_v1.13\GyverMatrixOS_v1.13.ino:18:0:

C:\Users\tuant\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0\libraries\SoftwareSerial\src/SoftwareSerial.h:53:10: note: void SoftwareSerial::begin(uint32_t, int8_t, int8_t, SoftwareSerialConfig, bool, int, int)

     void begin(uint32_t baud, int8_t rxPin, int8_t txPin = -1,

          ^

C:\Users\tuant\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0\libraries\SoftwareSerial\src/SoftwareSerial.h:53:10: note:   candidate expects 7 arguments, 1 provided

Multiple libraries were found for "DFRobotDFPlayerMini.h"
 Used: C:\Users\tuant\Documents\Arduino\libraries\DFRobotDFPlayerMini
Multiple libraries were found for "GyverButton.h"
 Used: C:\Users\tuant\Documents\Arduino\libraries\GyverButton
Multiple libraries were found for "GyverFilters.h"
 Used: C:\Users\tuant\Documents\Arduino\libraries\GyverFilters
Multiple libraries were found for "ESP8266WiFi.h"
 Used: C:\Users\tuant\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0\libraries\ESP8266WiFi
Multiple libraries were found for "TimeLib.h"
 Used: C:\Users\tuant\Documents\Arduino\libraries\Time
Multiple libraries were found for "EEPROM.h"
 Used: C:\Users\tuant\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0\libraries\EEPROM
Multiple libraries were found for "SoftwareSerial.h"
 Used: C:\Users\tuant\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0\libraries\SoftwareSerial
Multiple libraries were found for "FastLED.h"
 Used: C:\Users\tuant\Documents\Arduino\libraries\FastLED
exit status 1
no matching function for call to 'SoftwareSerial::SoftwareSerial(const uint8_t&, const uint8_t&)'

bill-orange, 2019-11-09:
@tihoangyeudau  I have exactly the same failure.  You are not alone.  I reverted to an earlier board manager version.

devyte, 2019-11-09:
CC @dok-net

dok-net, 2019-11-09:
Intentionally so, and for some time now, in previous releases IIRC. It aligns to the HW serial API.Won't change.

bill-orange, 2019-11-09:
@devyte @dok-net I really think this change needs to be reconsidered.  Board Manager version 2.5.2 worked fine.  For those of use that don't follow the update work blow-by-blow this is a new change.

I ran a few test compilations and it appears this change breaks, TinyGPS, all Adafruit GPS examples, all SparkFun GPS examples and all SDS011 sensor examples.  Most likely the underlying libraries are broken.

This will create chaos.

dok-net, 2019-11-09:
For a different POV:

Software serial can handle the diagnostics to USB quite well, whereas using the HW UART for the real application is in almost any scenario the superior option. I can’t even begin to imagine why all those libraries never considered

```
Serial.swap();
```

Now, if you’re faced with this situation, it’s only fair that I give you the simple change that needs to be done to the sources:

Before, somewhere in the sketch, provided that RX and TX are defines for the RX and TX pin numbers respectively:

```
SoftwareSerial swSer1(RX, TX); // non-inverted physical protocol, buffer size 256
swSer1.begin(115200);
```

Now, for quite awhile:
```
SoftwareSerial swSer1;
swSer1.begin(115200, RX, TX);
```

Where for the HW UART things would be:
```
Serial.swap();
Serial.begin(115200);
```

I hope you see the beauty of it – it makes it simpler to switch to HW UART, and it’s a bit easier on the eye. Plus, switching bitrate at runtime should not involve the constructor!

bill-orange, 2019-11-09:
Makes sense. Earlier in the day, I changed the SDS011 library in a similar fashion so I could compile. Adafruit and Sparkfun are still likely to be hopping mad. This will be a lot of work for them and a lot of confused customers. 

One other thing. I am still getting an error on the first line of compile ‘fatal:not a git repository (or any of the parent directories) :git’.

dok-net, 2019-11-10:
EspSoftwareSerial 6.0.0 has backward compatibility now for the common ctor/begin() setup. If default arguments are overridden, please adapt to the new API.

devyte, 2019-11-10:
Addressed in #6734.   
There are some limitations in the backwards compatibility. I understand this is the best that is reasonable.

closed 2019-11-10 by devyte

stevenveenma, 2019-11-10:
@dok-net hooked my issue to this thread. I updated to ESP8266 Arduino release 2.6.0 and try to get my sketch based on this https://github.com/neographikal/P1-Meter-ESP8266-MQTT/blob/master/P1Meter/P1Meter.ino to work.

I replaced 
```
SoftwareSerial mySerial(SERIAL_RX, -1, true, MAXLINELENGTH); // (RX, TX. inverted, buffer)
```
by

```
SoftwareSerial swSer1; 
swSer1.begin(115200, SERIAL_RX, -1);
```

Now I get the errors: 

fatal: Not a git repository (or any of the parent directories): .git
p1_mqtt2:54:1: error: 'swSer1' does not name a type

 swSer1.begin(115200, SERIAL_RX, -1);

 ^

C:\Users\sveen\Documents\Arduino\p1_mqtt2\p1_mqtt2.ino: In function 'void setup()':

p1_mqtt2:71:22: error: no matching function for call to 'SoftwareSerial::begin(int)'

   swSer1.begin(115200);

bill-orange, 2019-11-10:
@stevenveenma  You were close!  I made a few more changes and got the example to compile.  You can ignore the 'fatal: not a git whatever' error.  It is cosmetic.

Here's the relevant sections:

```
//Infrastructure stuff
const int MAXLINELENGTH = 64;
char telegram[MAXLINELENGTH];
SoftwareSerial mySerial;
constexpr SoftwareSerialConfig swSerialConfig = SWSERIAL_8N1;
constexpr int IUTBITRATE = 115200;
WiFiClient espClient;
PubSubClient client(espClient);

void setup()
{
  Serial.begin(115200);
  mySerial.begin (IUTBITRATE,SERIAL_RX, -1, swSerialConfig,true, MAXLINELENGTH); 
  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.waitForConnectResult() != WL_CONNECTED)
  {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }
  // mySerial.begin(115200); removed
  ArduinoOTA.setHostname(hostName);

```

dok-net, 2019-11-10:
@bill-orange Would you like to report this (solved by you) issue along with the involved version numbers and your fix to the P1Meter repository? Even if the author doesn't implement your fix, everyone in need might find it there, but they won't be looking here. Thanks.

bill-orange, 2019-11-10:
Sure, probably will not happen till tomorrow, but will do.

stevenveenma, 2019-11-11:
> @stevenveenma You were close! I made a few more changes and got the example to compile. You can ignore the 'fatal: not a git whatever' error. It is cosmetic.
>
> Here's the relevant sections:
>
> ``` [...]

I just tested the new sketch with your guidance and it works!. I now get a telegram every ten seconds. Content seems to be right. I will check if it stays stable on the long run. Thank you very much for your help, highly appreciated!

I don't know what @dok-net  meant with the P1meter repository, but I think other communities need to know this as well, I raised the issue because on the Tweakers.net forum the softwareserial was considered as not suitable for the P1 meter. I will react on that myself. Besided the various sketches on github that make use of the old softwareserial should be corrected because people like me pick the wrong sketch and are disappointed.

dok-net, 2019-11-12:
This is kind of going backward now :-)
I've commited compatibility changes to EspSoftwareSerial that should make upgrading easy for the simple use cases, where inversion and buffer sizes don't matter.
This is contained now in ESP8266 Arduino core master, I expect a 2.6.1 bug-fix release relatively soon, then it will be in the Arduino BSP manager available in packaged format as well.

```
SoftwareSerial mySerial(SERIAL_RX, -1);

mySerial.begin(115200, SWSERIAL_8N1);
```

Caveat, this comes at the price that the syntax you are using became obsolete and won't compile yet again:

Wrong in EspSoftwareSerial 6.0.0 and later:
```
SoftwareSerial mySerial;

mySerial.begin (IUTBITRATE, SERIAL_RX, -1, swSerialConfig,true, MAXLINELENGTH);
```
Correct in EspSoftwareSerial 6.0.0 and above:
```
SoftwareSerial mySerial;

mySerial.begin(IUTBITRATE, swSerialConfig, SERIAL_RX, -1, true, MAXLINELENGTH);
```
Take notice how SoftwareSerialConfig and the pin assignment get swapped.

bill-orange, 2019-11-12:
After loading EspSoftwareSerial 6.1.1 through the library manager,revising the P1Meter sketch as an above, I now get:

```
P1Meter:50:16: error: no matching function for call to 'SoftwareSerial::SoftwareSerial()'

 SoftwareSerial mySerial;

                ^

C:\Users\William\Downloads\P1-Meter-ESP8266-MQTT-master\P1-Meter-ESP8266-MQTT-master\P1Meter\P1Meter.ino:50:16: note: candidates are:

In file included from C:\Users\William\Downloads\P1-Meter-ESP8266-MQTT-master\P1-Meter-ESP8266-MQTT-master\P1Meter\P1Meter.ino:4:0:

C:\Users\William\AppData\Local\Arduino15\packages\Heltec-esp8266\hardware\esp8266\0.0.1\libraries\SoftwareSerial/SoftwareSerial.h:37:3: note: SoftwareSerial::SoftwareSerial(int, int, bool, unsigned int)

   SoftwareSerial(int receivePin, int transmitPin, bool inverse_logic = false, unsigned int buffSize = 64);

   ^

C:\Users\William\AppData\Local\Arduino15\packages\Heltec-esp8266\hardware\esp8266\0.0.1\libraries\SoftwareSerial/SoftwareSerial.h:37:3: note:   candidate expects 4 arguments, 0 provided

C:\Users\William\AppData\Local\Arduino15\packages\Heltec-esp8266\hardware\esp8266\0.0.1\libraries\SoftwareSerial/SoftwareSerial.h:35:7: note: constexpr SoftwareSerial::SoftwareSerial(const SoftwareSerial&)

 class SoftwareSerial : public Stream {

       ^

C:\Users\William\AppData\Local\Arduino15\packages\Heltec-esp8266\hardware\esp8266\0.0.1\libraries\SoftwareSerial/SoftwareSerial.h:35:7: note:   candidate expects 1 argument, 0 provided

P1Meter:51:11: error: 'SoftwareSerialConfig' does not name a type

 constexpr SoftwareSerialConfig swSerialConfig = SWSERIAL_8N1;

           ^

C:\Users\William\Downloads\P1-Meter-ESP8266-MQTT-master\P1-Meter-ESP8266-MQTT-master\P1Meter\P1Meter.ino: In function 'void setup()':

P1Meter:59:30: error: 'swSerialConfig' was not declared in this scope

   mySerial.begin (IUTBITRATE,swSerialConfig,SERIAL_RX, -1,true, MAXLINELENGTH); // (RX, TX. inverted, buffer)

                              ^

Multiple libraries were found for "SoftwareSerial.h"
 Used: C:\Users\William\AppData\Local\Arduino15\packages\Heltec-esp8266\hardware\esp8266\0.0.1\libraries\SoftwareSerial
 Not used: C:\Users\William\Documents\Arduino\libraries\EspSoftwareSerial
exit status 1
no matching function for call to 'SoftwareSerial::SoftwareSerial()'

```

stevenveenma, 2019-11-13:
I cheered too early. Initially the sketch I use sends the messages every ten seconds. But this stops sometimes for longer periods. I took me some time to find the reason. The sketch compares actual and preceding results of the telegrams and don't send a mqtt message if this the result on any of the readings is negative. I can check this by the mqqt log. In case of not receiving a normal message the log showed a null values on one of the telegram readings. Apparantly something is going wrong the the receiving of the telegram or the translation. It will be difficult for me to find out  the root-cause. I am considering to process the messages anyway and filter the erroneous value in Python.

bill-orange, 2019-11-13:
Have you tried setting the following to true and looking for corruption in your data?

```
const bool outputOnSerial = false;
const bool outputMqttLog = false;
```

You could also try increasing the buffer size to 128.

`const int MAXLINELENGTH = 64;`

I have found that the ESP32 generally are more robust in their handling of serial data.  You could try an ESP32.  You would only need to change a half a dozen lines of code to use ESP32 hardware serial.  Of course, if the data coming out of your meter is corrupt none of this will matter!

P.S.  I would move this conversation here. https://github.com/neographikal/P1-Meter-ESP8266-MQTT/issues  where it would be more "on topic"

stevenveenma, 2019-11-13:
Yes, the logging, that was exactly what I did. I found some null values in the mqttlog. The serial is difficult to interpret. I changed the maxlinelength to 128 and its running again, lets see if things will change.

bill-orange, 2019-11-13:
If that doesn't work.  You can try scattering `yield();` around.  Since the ESP8266 has only one CPU, WiFi connection maintenance can get in the way of serial communication.  A couple of likely spots to put the `yield();` are before 

`if (mySerial.available())`

and

`while (mySerial.available())`

You can try taking the` yield();` before` if (decodeTelegram(len + 1))` out and see if things get worse.  I went through all this with my GPS sketches.

macron0, 2020-08-06:
Hi guys, 

I must be overlooking something but how is this fixed?
I tried modding the original ino from jantenhove but it refuses to compile. 
So I tried only the code of Bill-orange removing the esp client stuff and adding #define SERIAL_RX  2 but it fails with : 

invalid conversion from 'int' to 'SoftwareSerialConfig' [-fpermissive]

Should I switch to espSoftwareSerial or something?

I am using arduino ide 1.8.10  and  Generic esp8266 2.6.3 

Below what I try to compile:

`#include <SoftwareSerial.h>
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <ESP8266HTTPClient.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include "CRC16.h"

//Infrastructure stuff
#define SERIAL_RX 2
const int MAXLINELENGTH = 64;
char telegram[MAXLINELENGTH];
SoftwareSerial mySerial;
constexpr SoftwareSerialConfig swSerialConfig = SWSERIAL_8N1;
constexpr int IUTBITRATE = 115200;

void setup()
{
  Serial.begin(115200);
  mySerial.begin (IUTBITRATE,SERIAL_RX, -1, swSerialConfig,true, MAXLINELENGTH); 
  Serial.println("Booting");

while (WiFi.waitForConnectResult() != WL_CONNECTED)
  {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
  }
voide loop
  {
  }
  `
Thx for any suggestions.

bill-orange, 2020-08-06:
Maybe SoftWareSerial syntax has changed a bit.  Try this.

```
#include <SoftwareSerial.h>
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <ESP8266HTTPClient.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
//#include "CRC16.h"

//Infrastructure stuff
#define SERIAL_RX 2
const int MAXLINELENGTH = 64;
char telegram[MAXLINELENGTH];
SoftwareSerial mySerial;
constexpr SoftwareSerialConfig swSerialConfig = SWSERIAL_8N1;
constexpr int IUTBITRATE = 115200;

void setup()
{
Serial.begin(115200);
mySerial.begin (IUTBITRATE, swSerialConfig, SERIAL_RX,true, MAXLINELENGTH);
Serial.println("Booting");

while (WiFi.waitForConnectResult() != WL_CONNECTED)

Serial.println("Connection Failed! Rebooting...");
delay(5000);
}
void loop()
{
}
```

macron0, 2020-08-06:
That  fast! 
Thank you very much that worked.

magicjimmy, 2021-02-08:
can you give the full new code for ---- jantenhove/P1-Meter-ESP8266
updated
thanks

> 
> 
> That was fast!
> Thank you very much that worked.
> 
> Regards
> 
> Hans

can you give the full new code for ---- jantenhove/P1-Meter-ESP8266
updated
thanks

magicjimmy, 2021-02-08:
> Maybe SoftWareSerial syntax has changed a bit. Try this.
>
> ```
> #include <SoftwareSerial.h>
> #include <ESP8266WiFi.h>
> #include <ESP8266mDNS.h>
> #include <ESP8266HTTPClient.h>
> #include <WiFiUdp.h> [...]

can you give the full new code for ---- jantenhove/P1-Meter-ESP8266
updated
thanks