esp8266/Arduino #6742: Panic: BSSL stack overflow

kyeett, 2019-11-09:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				Adafruit Huzzah w/ NodeMCU
- Core Version:			2.6.0 (through Arduino IDE)
- Development Env:		Arduino IDE
- Operating System:		MacOS

### Settings in IDE

- Module:  			Adafruit Feather HUzzAH ESP8266
- Flash Mode:			usb
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		ck
- Flash Frequency:	40Mhz
- CPU Frequency:		160MHz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

Hi!

I'm trying to connect to google scripts from my Adafruit Feather Huzzah (ESP8266). 

It works maybe 2/3 it works, but in 1/3 of the cases, I get an `abort` from the BearSSL library, and the controller restarts. If I enable HTTP_CLIENT+SSL debug, I can see the error `FATAL ERROR: BSSL stack overflow`. 

I have tried the following, without success:
- Changing the  stackSize in `StackThunk.cpp`
- I have tried turning on "Basic SSL ciphers"
- Upgrading the board to 2.6.0

If anyone can help, I am grateful!

See code and debug messages below (my code uses https://github.com/electronicsguy/ESP8266/tree/master/HTTPSRedirect)

```cpp

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include "HTTPSRedirect.h"
#include <StackThunk.h>

#ifndef STASSID
...
...
#endif

const char* ssid = STASSID;
const char* password = STAPSK;

const int httpsPort = 443;

// Write to Google Spreadsheet
String url = String("/macros/s/MY_ID_HERE/exec?DeviceID=Device1&Value=1");

const char* host = "script.google.com";

HTTPSRedirect* client = nullptr;

void setup() {
[... 138 lines cut ...]
3ffffe60:  00000074 00000005 8f3815bc 00000000  
3ffffe70:  00000001 00006547 3fff73ec 4021dd9e  
3ffffe80:  3fff78ec 3ffefc7c 3fff73ec 00000001  
3ffffe90:  00000001 00000001 3ffef904 402038d0  
3ffffea0:  3ffefc7c 00000000 3ffef904 40203ab3  
3ffffeb0:  40104f19 3ffeea74 3ffeea74 00000008  
3ffffec0:  00000000 00000000 00000000 402068f7  
3ffffed0:  00000000 3ffefb84 3ffef904 402026a0  
3ffffee0:  000001bb 00000d50 3ffef904 00000000  
3ffffef0:  000001bb 3ffef904 3ffefc7c 00000000  
3fffff00:  000001bb 3ffef904 3ffefc7c 40203bd1  
3fffff10:  40208488 01d33ad8 40208488 01d33ad8  
3fffff20:  3ffefa08 0000012e 3ffef904 40204f41  
3fffff30:  3ffee94c 3ffefa04 3ffef904 4020500d  
3fffff40:  3ffe8788 3ffeea74 00001388 3ffee94c  
3fffff50:  3ffee948 3ffeea74 00001388 3ffee94c  
3fffff60:  3ffee948 3ffe84c0 00001388 40205032  
3fffff70:  3ffee948 3ffe84c0 00001388 4020117d  
3fffff80:  40208488 1100a8c0 feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3ffeea18  
3fffffa0:  3fffdad0 00000000 3ffee9d8 40206460  
3fffffb0:  feefeffe feefeffe 3ffe84e8 40100b75  
<<<stack<<<

```

earlephilhower, 2019-11-11:
It's definitely possible the BSSL stack may be going past the end of bounds on this connection.  The size was empirically determined, and you might be hitting a corner case.

However, the fact that you see if with only Basic SSL (Tools->SSL->Basic SSL (lower mem)) makes me wonder if it's something else.  The reason is the elliptic curves are the ones that take the largest amount of 2nd stack, and they're disabled in the basic mode.  Basically, when that mode is enabled it's been shown that the BSSL stack can actually be shrunk quite a lot because the RSA work is easier on the memory (we don't shrink it, though...probably worth a PR sometime in the future).

The HTTPSRedirect library you're using might be the key.  The last update was 3 years ago and BearSSL only came into the core after that.  Possibly it's engaging multiple SSL connections to handle the redirect w/o closing and destroying the first connection and hitting OOM, and overwriting something because it's ignoring an error condition.

In that case the BSSL stack canary, which is checked after every call, could be corrupted not by the stack but by a memory access from the main code...

Can you try it w/o the redirect library involved?  Your code may not work, but I'd like to see if the negotiations succeed all the time...

kyeett, 2019-11-11:
Good idea. I will try it out as soon as I have my Arduino at hand.

kyeett, 2019-11-16:
Hi!

I did as you suggested and used the client directly (using the example `BasicHttpsClient` in the Arduino IDE), and it seems to be working! 

Thanks for your support. I will close this issue.

/Magnus

closed 2019-11-16 by kyeett

macaca75815, 2019-12-06:
Hello!!kyeett,I use the Esp8266 and having the same problem for google javascrip.
Can you help me,share your code.., I try BasicHttpsClient but still confuse,Thank you very much.