esp8266/Arduino #6745: Compilation with ver. 2.6 results in error

bill-orange, 2019-11-10:
Subject; Compiler error 

- Hardware:				Huzzah Feather
- Core Version:			latest 2.6
- Development Env:		Arduino IDE 1.8.9
- Operating System:		Windows 10

### Settings in IDE

- Module:  			Huzzah Feather
- Flash Mode:			unknown
- Flash Size: 		        4MB
- lwip Variant:                   v2 Lower Memory
- Reset Method:		unknown
- Flash Frequency:	        80Mhz
- CPU Frequency:		unknown
- Upload Using:		SERIAL
- Upload Speed:              115200

Problem Description: Compilation results in an error labeled 'fatal'  However, compilation appears to be successful.

Detailed problem description:

Compiler Error: fatal: not a git repository (or any of the parent directories): .git

Sketch: Simple Blink.ino will produce the error.

Additional information:

This error was mentioned in posts by myself and others, in passing.  I felt that it should get its own issue so that it does not get overlooked.  Since compilation does occur, this is somewhat cosmetic but it would be confusing to new users, especially since it says 'fatal'.  I believe this has something to do with how the change was released through github.

mcspr, 2019-11-10:
Hey,

Can you please try removing this line?
https://github.com/esp8266/Arduino/blob/dde76d1b5da4a99c109a8a4a29dae693fef316b4/platform.txt#L95

I think this line was supposed to be removed by the release script, but the pattern is different now:
https://github.com/esp8266/Arduino/blob/dde76d1b5da4a99c109a8a4a29dae693fef316b4/package/build_boards_manager_package.sh#L97
9629874038aae217f0ea2d0587db703c5e8abdae <

d-a-v, 2019-11-10:
@mcspr That is right : https://github.com/esp8266/Arduino/commit/8b5433977edbe9749d3d7cdfc6a2fb7b19944095

d-a-v, 2019-11-10:
@bill-orange 
You may wish to try [my alpha release channel](https://d-a-v.github.io) to check if your issue is fixed.

PR #6747 (2019-11-10): (fix) remove makecorever.py call in released version
may fix #6745
closed 2019-11-10 by d-a-v