esp8266/Arduino #6749: After upgrade to 2.6.0: not a git repository (or any parent up to mount point /

jbrepogmailcom, 2019-11-10:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				Wemos D1 mini
- Core Version:			2.6.0
- Development Env:		[Arduino IDE 1.8.9
- Operating System:		Ubuntu

### Settings in IDE

- Module:  			Wemos D1 mini r2
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

getting following error after upgrade in board manager from 2.5.2 to 2.6.0, Error happens in sudo as well:

Arduino: 1.8.9 (Linux), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

Build options changed, rebuilding all
fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
In file included from /home/janbenes/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/tools/sdk/libc/xtensa-lx106-elf/include/sys/ctype.h:7:0,
                 from /home/janbenes/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/tools/sdk/libc/xtensa-lx106-elf/include/ctype.h:5,
                 from /home/janbenes/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/cores/esp8266/WString.h:28,
                 from /home/janbenes/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/cores/esp8266/IPAddress.h:24,
                 from /home/janbenes/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/libraries/ESP8266WiFi/src/ESP8266WiFi.h:31,
                 from /home/janbenes/Dropbox/Apps-180116/Arduino/toshiba_ac-19111001/toshiba_ac-19111001.ino:5:
/home/janbenes/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:25:130: error: 'const char HTTP_HEAD []' redeclared as different kind of symbol
   #define PROGMEM __attribute__((section( "\".irom.text." __FILE__ "." __STRINGIZE(__LINE__) "."  __STRINGIZE(__COUNTER__) "\"")))
                                                                                                                                  ^
/home/janbenes/Dropbox/Apps-180116/Arduino/libraries/WiFiManager/WiFiManager.h:25:24: note: in expansion of macro 'PROGMEM'
 const char HTTP_HEAD[] PROGMEM            = "<!DOCTYPE html><html lang=\"en\"><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/><title>{v}</title>";
                        ^
In file included from /home/janbenes/Dropbox/Apps-180116/Arduino/libraries/WiFiManager/WiFiManager.h:17:0,
                 from /home/janbenes/Dropbox/Apps-180116/Arduino/toshiba_ac-19111001/toshiba_ac-19111001.ino:10:
/home/janbenes/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:33:39: error: previous declaration of 'HTTPMethod HTTP_HEAD'
 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
                                       ^
exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

After running git init I get

Arduino: 1.8.9 (Linux), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

fatal: Needed a single revision
In file included from /home/janbenes/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/tools/sdk/libc/xtensa-lx106-elf/include/sys/ctype.h:7:0,
                 from /home/janbenes/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/tools/sdk/libc/xtensa-lx106-elf/include/ctype.h:5,
                 from /home/janbenes/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/cores/esp8266/WString.h:28,
                 from /home/janbenes/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/cores/esp8266/IPAddress.h:24,
                 from /home/janbenes/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/libraries/ESP8266WiFi/src/ESP8266WiFi.h:31,
                 from /home/janbenes/Dropbox/Apps-180116/Arduino/toshiba_ac-19111001/toshiba_ac-19111001.ino:5:
/home/janbenes/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:25:130: error: 'const char HTTP_HEAD []' redeclared as different kind of symbol
   #define PROGMEM __attribute__((section( "\".irom.text." __FILE__ "." __STRINGIZE(__LINE__) "."  __STRINGIZE(__COUNTER__) "\"")))
                                                                                                                                  ^
/home/janbenes/Dropbox/Apps-180116/Arduino/libraries/WiFiManager/WiFiManager.h:25:24: note: in expansion of macro 'PROGMEM'
 const char HTTP_HEAD[] PROGMEM            = "<!DOCTYPE html><html lang=\"en\"><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/><title>{v}</title>";
                        ^
In file included from /home/janbenes/Dropbox/Apps-180116/Arduino/libraries/WiFiManager/WiFiManager.h:17:0,
                 from /home/janbenes/Dropbox/Apps-180116/Arduino/toshiba_ac-19111001/toshiba_ac-19111001.ino:10:
/home/janbenes/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:33:39: error: previous declaration of 'HTTPMethod HTTP_HEAD'
 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
                                       ^
exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

d-a-v, 2019-11-10:
duplicate of #6745 . There is an alpha pre-release to test

devyte, 2019-11-10:
Closings due to duplicate.

closed 2019-11-10 by devyte